MIT-SPARK/Kimera-VIO-ROS #5: permission denied when git cloneing

fengziyue, 2019-10-19:
when I excute the git clone command:"git clone git@github.com:MIT-SPARK/Kimera-VIO-ROS.git", it shows this error message:

~/catkin_ws/src$ git clone git@github.com:MIT-SPARK/Kimera-VIO-ROS.git
Cloning into 'Kimera-VIO-ROS'...
Warning: Permanently added the RSA host key for IP address '140.82.114.4' to the list of known hosts.
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

fengziyue, 2019-10-19:
I will use https method to clone, sorry for that

closed 2019-10-19 by fengziyue

marcusabate, 2019-10-19:
@fengziyue do you have ssh keys for your github.com account on your machine? If not, try:
```bash
git clone https://github.com/MIT-SPARK/Kimera-VIO-ROS.git
```