MIT-SPARK/Kimera-VIO-ROS #8: catkin build errors

secondsensei, 2019-10-21:
**Description:**
When running the `catkin build` command during [KimeraVIO ROS wrapper installation](https://github.com/MIT-SPARK/Kimera-VIO-ROS#b-kimeravio-ros-wrapper-installation), some errors popped up. Not a 100% sure what they mean. Can you provide solution? Thanks.

**Console output:**
```
`Warnings << dbow2_catkin:make /home/user/catkin_ws/logs/dbow2_catkin/build.make.001.log`
`make[3]: warning: jobserver unavailable: using -j1. Add '+' to parent make rule.`

`Errors << KimeraVIO:make /home/user/catkin_ws/logs/KimeraVIO/build.make.001.log`
`/usr/bin/ld: cannot find -lDBoW2`
`collect2: error: ld returned 1 exit status`
`make[2]: *** [libKimeraVIO.so] Error 1`
`make[1]: *** [CMakeFiles/KimeraVIO.dir/all] Error 2`
`make: *** [all] Error 2`

`Failed << KimeraVIO:make [Exited with code 2]`

`Abandoned <<< kimera_ros [Unrelated job failed]`

`[build] Summary: 11 of 12 packages succeeded.`
`[build] Warnings: 1 packages succeeded with warnings.`
`[build] Abandoned: 1 packages were abandoned.`
`[build] Failed: 1 packages failed.`
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: 3.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04.3 LTS
* Did you change the source code? (yes / no): no

[image]

msadowski, 2019-10-22:
I got the same issue  when following readme. To fix this you need to install DBoW2:

```
git clone https://github.com/dorian3d/DBoW2.git
cd DBoW2
mkdir build && cd build && cmake .. && make && sudo make install
```

After this your package should build!

ToniRV, 2019-10-22:
We will try to make DBoW2 optional to avoid this error unless you want to enable loop closures.
Thanks for raising the issue though!

closed 2019-10-22 by ToniRV

marcusabate, 2019-10-22:
@secondsensei, did you use the rosinstall file to clone and build the catkin wrappers for all the dependencies? If you do the following:
```bash
# Install dependencies from rosinstall file using wstool
wstool init # Use unless wstool is already initialized

# For ssh:
wstool merge kimera_ros/install/kimera_ros_ssh.rosinstall
# For https
# wstool merge kimera_ros/install/kimera_ros_https.rosinstall

# Finally, download and update repos:
wstool update
```
before calling `catkin build` then it should automatically build `DBoW2` for you. Please let us know if that didn't work!

msadowski, 2019-10-23:
@marcusabate I tried the instructions you pasted here and run them with one modification. Instead of `wstool merge kimera_ros/install/kimera_ros_ssh.rosinstall` I had to use `wstool Kimera-VIO-ROS/install/kimera_ros_ssh.rosinstall` because the path was not matching. 

When I run it it seemed like DBoW2 was not installed (In my workspace I could see dbow2_catkin though).

nkwangyh, 2019-11-08:
It seems the reason is the path of libdbow2 is not specified in cmakelist.txt. Just replacing  "DBoW2" in the target_link_libraries command with "${DBoW2_LIBRARIES}" will fix the issue.
Specifically, in Kimera-VIO/CMakeLists.txt:
```
target_link_libraries(KimeraVIO
  PRIVATE
    opengv
    #DBoW2
    ${DBoW2_LIBRARIES}
    KimeraRPGO
  PUBLIC
    Boost::boost
    ${OpenCV_LIBRARIES} # TODO: modernize
    gflags::gflags
    glog::glog
    gtsam
    gtsam_unstable
)
```