MIT-SPARK/Kimera-VIO-ROS #12: GTSAM was built against a different version of Eigen

hardesh, 2019-10-26:
**Description:**
I built gtsam with and without the default eigen. Still getting the error

**Console output:**
```
In file included from /home/harshal/vio_ws/devel/include/gtsam/base/Matrix.h:26:0,
                 from /home/harshal/vio_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/harshal/vio_ws/devel/include/gtsam/base/Lie.h:25,
                 from /home/harshal/vio_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/harshal/vio_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/harshal/vio_ws/devel/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/harshal/vio_ws/src/Kimera-VIO/include/kimera-vio/StereoFrame.h:35,
                 from /home/harshal/vio_ws/src/Kimera-VIO/include/kimera-vio/StereoVisionFrontEnd.h:25,
                 from /home/harshal/vio_ws/src/Kimera-VIO/src/StereoVisionFrontEnd.cpp:16:
/home/harshal/vio_ws/devel/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
```
Similar error is thrown for other library files.

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 16.04 with ROS kinetic
* Did you change the source code? (yes / no):  no

hardesh, 2019-10-26:
'''
if(GTSAM_USE_SYSTEM_EIGEN)
	find_package(Eigen3 REQUIRED)

	# Use generic Eigen include paths e.g. <Eigen/Core>
	set(GTSAM_EIGEN_INCLUDE_FOR_INSTALL "${EIGEN3_INCLUDE_DIR}")

	# check if MKL is also enabled - can have one or the other, but not both!
	# Note: Eigen >= v3.2.5 includes our patches
	if(EIGEN_USE_MKL_ALL AND (EIGEN3_VERSION VERSION_LESS 3.2.5))
	  message(FATAL_ERROR "MKL requires at least Eigen 3.2.5, and your system appears to have an older version. Disable GTSAM_USE_SYSTEM_EIGEN to use GTSAM's copy of Eigen, or disable GTSAM_WITH_EIGEN_MKL")
	endif()

	# Check for Eigen version which doesn't work with MKL
	# See http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1527 for details.
	if(EIGEN_USE_MKL_ALL AND (EIGEN3_VERSION VERSION_EQUAL 3.3.4))
		message(FATAL_ERROR "MKL does not work with Eigen 3.3.4 because of a bug in Eigen. See http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1527. Disable GTSAM_USE_SYSTEM_EIGEN to use GTSAM's copy of Eigen, disable GTSAM_WITH_EIGEN_MKL, or upgrade/patch your installation of Eigen.")
	endif()

	# The actual include directory (for BUILD cmake target interface):
	set(GTSAM_EIGEN_INCLUDE_FOR_BUILD "${EIGEN3_INCLUDE_DIR}")
else()
	# Use bundled Eigen include path.
	# Clear any variables set by FindEigen3
	if(EIGEN3_INCLUDE_DIR)
		set(EIGEN3_INCLUDE_DIR NOTFOUND CACHE STRING "" FORCE)
	endif()

	# set full path to be used by external projects
	# this will be added to GTSAM_INCLUDE_DIR by gtsam_extra.cmake.in
	set(GTSAM_EIGEN_INCLUDE_FOR_INSTALL "include/gtsam/3rdparty/Eigen/")

	# The actual include directory (for BUILD cmake target interface):
	set(GTSAM_EIGEN_INCLUDE_FOR_BUILD "${CMAKE_SOURCE_DIR}/gtsam/3rdparty/Eigen/")

	message("used bundled eigen")   # added this line

endif()
'''
I added message in the CMake to check whether the gtsam was getting built with the proper version and I was able to see the message output on the terminal window.

jbdaniel18, 2019-10-29:
Any update on this?

hardesh, 2019-10-30:
I built GTSAM separately from source. I stopped getting this issue. But I am not able to build KimeraVIO package. My laptop crashes everytime I try to build the package. I am still not sure why this is happening.

ToniRV, 2019-11-07:
@jbdaniel18 what Ubuntu are you using?
It seems this issue is related to Ubuntu 16.04 only, just to check.

jbdaniel18, 2019-11-07:
I am using Ubuntu 16.04

hardesh, 2019-11-10:
I am also using Ubuntu 16.04

acciosisu, 2019-11-11:
Hi, I am able to build KimeraRPGO but Kimera VIO fails. I am using Ubuntu 16.04 as well.

673402618, 2019-11-14:
Hi, I am using Ubuntu 16.04. I found the problem is that the package `pcl_ros` relies on `Eigen 3.2.92` in `/usr/include`. I replace it with `Eigen 3.3.7` and build successfully. But it may cause other problems.
I test it on EuRoC dataset. It can work, but there is a red on terminal `Attempt to register LCD/PGO callback, but no LoopClosureDetector member is active in pipeline.`

ToniRV, 2019-11-14:
Thanks @673402618 ! The error message can be safely ignored. It is just that the Loop Closure detector is not activated.

roelofvandijk, 2019-11-26:
Same issue, fixed by upgrading system eigen in /user/include/eigen3 to Eigen 3.3.7. Thanks @673402618.

mzahana, 2019-12-19:
@roelofvandijk can you please describe how you updated Eigen? Is it through source or apt-get?

mzahana, 2019-12-19:
I was able to fix the problem by setting `GTSAM_USE_SYSTEM_EIGEN=ON`:
```
catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=ON
```

hardesh, 2020-01-03:
Switched to Ubuntu 18.04 and ROS melodic. I did not come across this issue after that.

closed 2020-01-03 by hardesh

ToniRV, 2020-02-18:
Check also #43

DuyNhatNguyenIT, 2023-08-08:
```
cmake .. -DGTSAM_USE_SYSTEM_EIGEN=ON
make
```

tianyilim, 2024-10-29:
I had to do this in a Ubuntu 20.04 Docker container (ros_noetic base), following the steps in the [Kimera-VIO-ROS readme](https://github.com/MIT-SPARK/Kimera-VIO-ROS?tab=readme-ov-file#b-kimeravio-ros-wrapper-installation)

PR #34 (2020-01-16): Update installation instructions
Add `-DGTSAM_USE_SYSTEM_EIGEN=ON` to CMake flags for Ubuntu 16.04 (see issue #12).

Resolves #12