MIT-SPARK/Kimera-VIO-ROS #13: CMake Error :  This workspace contains non-catkin packages

qqadssp, 2019-11-13:
CMake Error :  This workspace contains non-catkin packages in it, and catkin cannot build

gtsam (plain cmake)
KimeraRPGO (plain cmake)
KimeraVIO (plain cmake)

how to fix it?

ToniRV, 2019-11-13:
These packages are indeed *not* catkin packages, but they all have a package.xml file that indicates to catkin that the package should be built using plain cmake.
Can you give us more info on :
- OS
- ROS version
- catkin version
- cmake version
- build command (`catkin_make`, `catkin build`?)

ToniRV, 2019-12-28:
@qqadssp closing, but feel free to re-open if you couldn't fix the issue.

closed 2019-12-28 by ToniRV

ains-postech, 2021-02-23:
ubuntu 18.04
ros melodic
catkin version lastest
cmake lastest
build command : catkin_make

I tried with catkin_make but it gives an error. Is there a solution?

Base path: /home/cho/catkin_ws
Source space: /home/cho/catkin_ws/src
Build space: /home/cho/catkin_ws/build
Devel space: /home/cho/catkin_ws/devel
Install space: /home/cho/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/cho/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/cho/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/cho/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/cho/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/cho/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 28 packages in topological order:
-- ~~  - catkin_simple
-- ~~  - cmake_external_project_catkin
-- ~~  - dbow2_catkin
-- ~~  - drone_mavros
-- ~~  - gflags_catkin
-- ~~  - glog_catkin
-- ~~  - gtsam (plain cmake)
-- ~~  - kimera_rpgo (plain cmake)
-- ~~  - opencv3_catkin
-- ~~  - opengv_catkin
-- ~~  - kimera_vio (plain cmake)
-- ~~  - avoidance_msg_generator
-- ~~  - drone_gazebo
-- ~~  - disparity_image_proc
-- ~~  - kimera_rviz_markers
-- ~~  - orb_slam2_ros
-- ~~  - orb_slam3_ros_wrapper
-- ~~  - pose_graph_tools
-- ~~  - image_undistort
-- ~~  - avoidance_apm_nv
-- ~~  - local_planner_apm_nv
-- ~~  - mesh_rviz_plugins
-- ~~  - kimera_vio_ros
-- ~~  - vision_to_mavros
-- ~~  - zed_interfaces
-- ~~  - zed_nodelets
-- ~~  - zed_ros (metapackage)
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/cho/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/cho/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:3064: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

[image]

pranav-nt, 2021-07-20:
Managed to reproduce this issue with Ubuntu 18.04, ROS Melodic, catkin version 0.6.1, cmake version 3.20.0
The issue only occurs with the command 'catkin_make'. With 'catkin build', the build process goes through.
If the issue persists with 'catkin build', run 'catkin clean' in the root directory just before the build command.