MIT-SPARK/Kimera-VIO-ROS #14: Error: The current devel space layout, `linked`,is incompatible with the configured layout, `merged`.

hamsterasu, 2019-11-15:
**Description:**
Hi, I am unable to run `catkin build`. This is what I did:

**Command:**
```
sudo apt-get install -y --no-install-recommends apt-utils
sudo apt-get install -y \
      cmake build-essential unzip pkg-config autoconf \
      libboost-all-dev \
      libjpeg-dev libpng-dev libtiff-dev \
      libvtk5-dev, libgtk2.0-dev  \
      libatlas-base-dev gfortran \
      libparmetis-dev \
      python-wstool python-catkin-tools \
sudo apt-get install libtbb-dev
mkdir -p ~/kimera_ws/src
cd ~/kimera_ws
catkin init
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin config --merge-devel
echo 'source ~/kimera_ws/devel/setup.bash' >> ~/.bashrc
cd ~/catkin_ws/src
git clone git@github.com:MIT-SPARK/Kimera-VIO-ROS.git
wstool init
wstool merge Kimera-VIO-ROS/install/kimera_ros_ssh.rosinstall
wstool update
catkin build
```

**Console output:**
```
Error: The current devel space layout, `linked`,is incompatible with the configured layout, `merged`.
```

I am running on Ubuntu 16.04 with ROS Kinetic installed.

ToniRV, 2019-11-15:
Do a catkin clean:
‘catkin clean -y’

hamsterasu, 2019-11-15:
That did not solve my problem. I realised that my workspace was inside another workspace which was my home folder. I solved my problem by removing the .catkin_tools folder in my home directory.

closed 2019-11-15 by hamsterasu