MIT-SPARK/Kimera-VIO-ROS #17: Consider adding c++14 requirement to installation instructions?

mfe7, 2019-11-22:
**Description:**
On Ubuntu 16.04, I got a bunch of errors that went away when i added a flag to use C++14.

**Command:**
```
catkin build
```

**Console output:**
```
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:209:30: error: ‘to_string’ is not a member of ‘std’
       std::string robot_id = std::to_string(keyedPose.first);
                              ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:209:55: error: request for member ‘first’ in ‘keyedPose’, which is of non-class type ‘const int’
       std::string robot_id = std::to_string(keyedPose.first);
                                                       ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:210:32: error: request for member ‘second’ in ‘keyedPose’, which is of non-class type ‘const int’
       MakeMenuMarker(keyedPose.second, robot_id);
                                ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:212:9: error: ‘interactive_mrkr_srvr_’ was not declared in this scope
     if (interactive_mrkr_srvr_ != nullptr) {
         ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:212:35: error: ‘nullptr’ was not declared in this scope
     if (interactive_mrkr_srvr_ != nullptr) {
                                   ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:233:22: error: ISO C++ forbids declaration of ‘keyedPose’ with no type [-fpermissive]
     for (const auto &keyedPose : keyed_poses_) {
                      ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:233:34: warning: range-based ‘for’ loops only available with -std=c++11 or -std=gnu++11
     for (const auto &keyedPose : keyed_poses_) {
                                  ^
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:233:34: error: ‘keyed_poses_’ was not declared in this scope
/mnt/ubuntu_extra_ssd2/code/kimera_vio_ws/src/pose_graph_tools/src/visualizer.cpp:234:55: error: request for member ‘first’ in ‘keyedPose’, which is of non-class type ‘const int’
       m.points.push_back(getPositionFromKey(keyedPose.first));
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: master? on 11/22/19?
* GTSAM version used: ?
* OpenGV version used: ?
* OpenCV version used: ?
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

I switched the line in the installation instructions from:
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
```
to:
```
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=14
```
and the errors disappeared. I don't know what the implications of this are for the rest of the system but it now compiles, so potentially worth adding to instructions.

ToniRV, 2019-11-25:
Thanks Michael! I think it is a problem with [pose_graph_tools](https://github.com/MIT-SPARK/pose_graph_tools). I'll report this there, but I'll update this issue.

ToniRV, 2019-12-06:
Another reason might be the usage of pcl, which requires c++14 (check pcl_catkin).

ToniRV, 2020-02-09:
Closing as I think the issue is solved here: https://github.com/MIT-SPARK/pose_graph_tools/commit/305c09f8700930afbfcbf68e0cd10246d6245fd8

closed 2020-02-09 by ToniRV