MIT-SPARK/Kimera-VIO-ROS #18: SPEEDUP ODOMETRY

logindiamant, 2019-11-25:
Dear colleagues, please tell how to speedup odometry?
MH_04_difficult.bag dataset, i7-3820 CPU @ 3.60GHz, /kimera_ros/odometry is ~5Hz

ToniRV, 2019-11-25:
Hi @logindiamant, can you copy-paste the log output that you get when running Kimera-VIO?
Something similar to:
```
Module: MesherModule - frequency: 62.5 Hz. (16 ms).
Keyframe 1554 with: 591 smart measurements
Module: VioFrontEnd - frequency: 20.8333 Hz. (48 ms).
------------------- Processing frame k = 1555--------------------
Module: VioFrontEnd - frequency: 100 Hz. (10 ms).
------------------- Processing frame k = 1556--------------------
Module: VioFrontEnd - frequency: 55.5556 Hz. (18 ms).
------------------- Processing frame k = 1557--------------------
Module: VioFrontEnd - frequency: 125 Hz. (8 ms).
------------------- Processing frame k = 1558--------------------
Module: VioBackEnd - frequency: 21.2766 Hz. (47 ms).
```

So that we can see what module is slowing things down for you.

logindiamant, 2019-11-26:
I1126 11:05:28.116629  3820 StereoVisionFrontEnd.cpp:101] ------------------- Processing frame k = 572--------------------
W1126 11:05:28.116714  3820 StereoVisionFrontEnd.cpp:125] Current IMU Preintegration frequency: 185185 Hz. (54 us).
I1126 11:05:28.229601  3820 StereoVisionFrontEnd.cpp:163] Keyframe 572 with: 400 smart measurements
W1126 11:05:28.229730  3820 StereoVisionFrontEnd.cpp:80] Current Stereo FrontEnd (keyframe) frequency: 8.84956 Hz. (113 ms).
I1126 11:05:28.229760  3820 StereoVisionFrontEnd.cpp:101] ------------------- Processing frame k = 573--------------------
W1126 11:05:28.229811  3820 StereoVisionFrontEnd.cpp:125] Current IMU Preintegration frequency: 263158 Hz. (38 us).
W1126 11:05:28.237658  3820 StereoVisionFrontEnd.cpp:80] Current Stereo FrontEnd frequency: 142.857 Hz. (7 ms).
I1126 11:05:28.237727  3820 StereoVisionFrontEnd.cpp:101] ------------------- Processing frame k = 574--------------------
W1126 11:05:28.237790  3820 StereoVisionFrontEnd.cpp:125] Current IMU Preintegration frequency: 196078 Hz. (51 us).
W1126 11:05:28.248302  3820 StereoVisionFrontEnd.cpp:80] Current Stereo FrontEnd frequency: 100 Hz. (10 ms).
I1126 11:05:28.273361  3820 StereoVisionFrontEnd.cpp:101] ------------------- Processing frame k = 575--------------------
W1126 11:05:28.273438  3820 StereoVisionFrontEnd.cpp:125] Current IMU Preintegration frequency: 222222 Hz. (45 us).
I1126 11:05:28.287458  3829 VioBackEnd.cpp:205] Backend: Update IMU Bias.
W1126 11:05:28.287755  3829 VioBackEnd.cpp:177] Current Backend frequency: 17.5439 Hz. (57 ms).
W1126 11:05:28.289456  3820 StereoVisionFrontEnd.cpp:80] Current Stereo FrontEnd frequency: 62.5 Hz. (16 ms).
W1126 11:05:28.292342  3830 Mesher.cpp:202] Current Mesher frequency: 333.333 Hz. (3 ms).

[image]
[image]

ToniRV, 2019-11-26:
The frontend is clearly the one to blame here...
How did you compile/install OpenCV and OpenGV?
Make sure both are in Release mode (actually, make sure everything is installed in Release mode).

logindiamant, 2019-11-26:
OpenCV steps
https://linuxize.com/post/how-to-install-opencv-on-ubuntu-18-04
...
cmake -D CMAKE_BUILD_TYPE=RELEASE \
    -D CMAKE_INSTALL_PREFIX=/usr/local \
    -D INSTALL_C_EXAMPLES=ON \
    -D INSTALL_PYTHON_EXAMPLES=ON \
    -D OPENCV_GENERATE_PKGCONFIG=ON \
    -D OPENCV_EXTRA_MODULES_PATH=~/home/gleb/opencv/opencv_contrib-3.4.8/modules \
    -D BUILD_EXAMPLES=ON ..

[image]
[image]
[image]

logindiamant, 2019-11-26:
Thanks for you answer, @ToniRV
What is the typically frequency can I expect in /kimera_ros/odometry with properly configured OpenCV/GV?

ToniRV, 2019-12-02:
You should get around at least `10Hz` odometry messages.

I would try to run Kimera without ROS to see if it works at that speed (sometimes linking with kimera-VIO-ROS with Kimera-VIO would slow the pipeline if Kimera is built as a shared library instead of static... still not confirmed though).

Also, make sure you delete the files you have in your `.cmake` folder in your home folder pointing to potentially old GTSAM/OpenCV/OpenGV/Kimera-X repos, as these may link Kimera to old builds you may have (perhaps a debug build?).

Hope it helps.

ToniRV, 2019-12-28:
@logindiamant we made substantial changes to the ROS wrapper to parallelize the IMU/image callbacks. That might help improving odometry output.

closed 2019-12-28 by ToniRV