MIT-SPARK/Kimera-VIO-ROS #19: GIT_TAG in dbow2_catkin and opengv_catkin

roelofvandijk, 2019-11-26:
**Description:**
`catkin build` was unable to correctly clone DBoW2 and opengv after a fresh clone.

**Command:**
```
catkin build
```

**Console output:**
```
---------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/ibotics/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/ibotics/temp
---------------------------------------------------------------------------
Build Space:        [exists] /home/ibotics/temp/build
Devel Space:        [exists] /home/ibotics/temp/devel
Install Space:      [unused] /home/ibotics/temp/install
Log Space:         [missing] /home/ibotics/temp/logs
Source Space:       [exists] /home/ibotics/temp/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
---------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
---------------------------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                                                                                                                                                
[build] Updating package table.                                                                                                                                                                            
Starting  >>> catkin_simple                                                                                                                                                                                
Starting  >>> gtsam                                                                                                                                                                                        
Finished  <<< catkin_simple                                [ 1.3 seconds ]                                                                                                                                 
Starting  >>> cmake_external_project_catkin                                                                                                                                                                
Starting  >>> dbow2_catkin                                                                                                                                                                                 
Starting  >>> kimera_rviz_markers                                                                                                                                                                          
Starting  >>> mesh_rviz_plugins                                                                                                                                                                            
Starting  >>> opencv3_catkin                                                                                                                                                                               
Starting  >>> opengv_catkin                                                                                                                                                                                
Starting  >>> pose_graph_tools                                                                                                                                                                             
___________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << gtsam:cmake /home/ibotics/temp/logs/gtsam/build.cmake.000.log                                                                                                                                
Ignoring Boost restriction on optional lvalue assignment from rvalues
cd /home/ibotics/temp/build/gtsam; catkin build --get-env gtsam | catkin env -si  /usr/bin/cmake /home/ibotics/temp/src/gtsam --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/ibotics/temp/devel -DCMAKE_BUILD_TYPE=Release; cd -
...........................................................................................................................................................................................................
Finished  <<< cmake_external_project_catkin                [ 2.3 seconds ]                                                                                                                                 
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << dbow2_catkin:make /home/ibotics/temp/logs/dbow2_catkin/build.make.000.log                                                                                                                    
Cloning into 'origin'...
CMake Error at /home/ibotics/temp/build/dbow2_catkin/dbow2_src-prefix/tmp/dbow2_src-gitclone.cmake:49 (message):
  Failed to checkout tag: 'master'

make[2]: *** [dbow2_src-prefix/src/dbow2_src-stamp/dbow2_src-download] Error 1
make[1]: *** [CMakeFiles/dbow2_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/ibotics/temp/build/dbow2_catkin; catkin build --get-env dbow2_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << dbow2_catkin:make                            [ Exited with code 2 ]                                                                                                                          
Failed    <<< dbow2_catkin                                 [ 4.1 seconds ]                                                                                                                                 
Abandoned <<< KimeraRPGO                                   [ Unrelated job failed ]                                                                                                                        
Abandoned <<< KimeraVIO                                    [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_ros                                   [ Unrelated job failed ]                                                                                                                        
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << opengv_catkin:make /home/ibotics/temp/logs/opengv_catkin/build.make.000.log                                                                                                                  
Cloning into 'origin'...
CMake Error at /home/ibotics/temp/build/opengv_catkin/opengv_src-prefix/tmp/opengv_src-gitclone.cmake:49 (message):
  Failed to checkout tag: 'master'

make[2]: *** [opengv_src-prefix/src/opengv_src-stamp/opengv_src-download] Error 1
make[1]: *** [CMakeFiles/opengv_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/ibotics/temp/build/opengv_catkin; catkin build --get-env opengv_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << opengv_catkin:make                           [ Exited with code 2 ]                                                                                                                          
Failed    <<< opengv_catkin                                [ 4.8 seconds ]    
```

Please give also the following information:
* SparkVio branch, tag or commit used VIO-Ros 6930beced3296239e4c9126ae8bfcdd7c0a505d9
* GTSAM version used: ?
* OpenGV version used: ?
* OpenCV version used: 3.3.1-dev
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

roelofvandijk, 2019-11-26:
Fixed it using by using the ([correct](https://cmake.org/cmake/help/latest/module/ExternalProject.html) and specific) tag `GIT_TAG origin/master` and removing the line ` GIT_PROGRESS 1` in both CMakeLists.txt files of opengv_catkin and dbow2_catkin.

**Example CMakeLists.txt for dbow2_catkin:**

```
cmake_minimum_required(VERSION 2.8.3)
project(dbow2_catkin)

find_package(catkin_simple REQUIRED)
catkin_simple()

include(ExternalProject)

file(MAKE_DIRECTORY ${CATKIN_DEVEL_PREFIX}/include)

ExternalProject_Add(dbow2_src
  GIT_REPOSITORY https://github.com/dorian3d/DBoW2.git
  GIT_TAG origin/master
  GIT_PROGRESS 1
  UPDATE_COMMAND ""
  #PATCH_COMMAND ""
  CONFIGURE_COMMAND cmake -DCMAKE_BUILD_TYPE=Release
                          -DCMAKE_INSTALL_PREFIX:PATH=${CATKIN_DEVEL_PREFIX}
                          -DBUILD_Demo=Off ../dbow2_src
  BUILD_COMMAND make
  INSTALL_COMMAND make install
)

install(DIRECTORY ${CATKIN_DEVEL_PREFIX}/include/DBoW2
        DESTINATION ${CATKIN_GLOBAL_INCLUDE_DESTINATION}
        FILES_MATCHING PATTERN "*.h")

install(DIRECTORY ${CATKIN_DEVEL_PREFIX}/lib/
        DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
        FILES_MATCHING PATTERN "libDBoW2*")

cs_export(INCLUDE_DIRS ${CATKIN_DEVEL_PREFIX}/include
          LIBRARIES DBoW2)
```

`

ToniRV, 2019-11-26:
Thanks! Feel free to open PRs in the corresponding repos, otw Ill update it myself asap

mzahana, 2019-12-19:
@roelofvandijk  Thanks! This solved the same issue I had. Would be nice to updated in the CMakeLists.txt

closed 2020-03-14 by ToniRV