MIT-SPARK/Kimera-VIO-ROS #20: gtsam compilation issue related to boost float128

roelofvandijk, 2019-11-26:
**Description:**

This is a curious issue.
`catkin build` was unable to build gtsam (16.04, ROS Kinetic).
The errors are along the vein of "error: unable to find numeric literal operator ‘operator""Q’".

Seems related to a c++11 and c++14 issue in [boost::multiprecision::float128](https://stackoverflow.com/questions/25365160/boostmultiprecisionfloat128-and-c11). 

Changes to the original code:

- First I used the c++14 fix from this [issue](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/17).
- Then I fixed the CMakeLists.txt according to my reply to [this issue](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/19).

**Command:**
```
`catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release -DCMAKE_CXX_STANDARD=14` 
`catkin build`
```

**Console output:**
```
Errors     << gtsam:make /home/ibotics/temp/logs/gtsam/build.make.000.log                                                                                                                                  
In file included from /usr/include/boost/config.hpp:61:0,
                 from /usr/include/boost/optional/optional.hpp:24,
                 from /usr/include/boost/optional.hpp:15,
                 from /home/ibotics/temp/src/gtsam/gtsam/base/OptionalJacobian.h:25,
                 from /home/ibotics/temp/src/gtsam/gtsam/base/Matrix.h:25,
                 from /home/ibotics/temp/src/gtsam/gtsam/base/Manifold.h:22,
                 from /home/ibotics/temp/src/gtsam/gtsam/base/Lie.h:25,
                 from /home/ibotics/temp/src/gtsam/gtsam/base/VectorSpace.h:11,
                 from /home/ibotics/temp/src/gtsam/gtsam/geometry/Point2.h:20,
                 from /home/ibotics/temp/src/gtsam/gtsam/geometry/Unit3.h:23,
                 from /home/ibotics/temp/src/gtsam/gtsam/geometry/Rot3.h:25,
                 from /home/ibotics/temp/src/gtsam/gtsam/geometry/Rot3M.cpp:25:
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:277:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(half, 5.000000000000000000000000000000000000e-01, "5.00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:277:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_third<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:278:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(third, 3.333333333333333333333333333333333333e-01, "3.33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:278:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_twothirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:279:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(twothirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:279:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_two_thirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:280:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(two_thirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:280:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_three_quarters<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:281:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(three_quarters, 7.500000000000000000000000000000000000e-01, "7.50000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:281:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_two<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:282:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_two, 1.414213562373095048801688724209698078e+00, "1.41421356237309504880168872420969807856967187537694807317667973799073247846210703885038753432764157273501384623e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:282:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_three<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:283:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_three, 1.732050807568877293527446341505872366e+00, "1.73205080756887729352744634150587236694280525381038062805580697945193301690880003708114618675724857567562614142e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:283:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half_root_two<T>::get(const mpl_::int_<5>&)’:
[... 288 lines cut ...]
/usr/include/boost/math/constants/constants.hpp:343:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_two_div_pi<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:344:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_two_div_pi, 7.978845608028653558798921198687637369e-01, "7.97884560802865355879892119868763736951717262329869315331851659341315851798603677002504667814613872860605117725e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:344:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
make[2]: *** [gtsam/CMakeFiles/gtsam.dir/geometry/Rot3M.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [gtsam/CMakeFiles/gtsam.dir/all] Error 2
make: *** [all] Error 2
cd /home/ibotics/temp/build/gtsam; catkin build --get-env gtsam | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << gtsam:make                                   [ Exited with code 2 ]                                                                                                                          
Failed    <<< gtsam                                        [ 1 minute and 43.4 seconds ]                                                                                                                   
Abandoned <<< KimeraRPGO                                   [ Unrelated job failed ]                                                                                                                        
[... cut ...]                                                                                                              
Abandoned <<< kimera_ros                                   [ Unrelated job failed ]      
```
Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used: 4.0.2
* OpenGV version used: 
* OpenCV version used: 
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no):  no, then yes

roelofvandijk, 2019-11-26:
Fixed the float128 issue by adding this line to the CMakeList.txt of gtsam, as per [this issue](https://github.com/UCL/STIR/issues/209):

`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DBOOST_MATH_DISABLE_FLOAT128")`

ToniRV, 2019-12-02:
Thanks @roelofvandijk !

closed 2019-12-02 by ToniRV