MIT-SPARK/Kimera-VIO-ROS #21: DBoW2 not installing correctly

roelofvandijk, 2019-11-26:
**Description:**
Building KimeraVIO fails because it cannot find the DBoW2 supplied with this repository. A workaround I tested is cloning DBoW2 manually, making it, and make installing it.

Given the differences in the CMakeFiles.txt files of DBoW2 and opengb I assume the BUILD COMMAND and the INSTALL COMMAND of DBoW2 should be changed.

from:
``` 
  BUILD_COMMAND make
  INSTALL_COMMAND make install
```
into:

``` 
  BUILD_COMMAND cd ../dbow2_src && make -j 8
  INSTALL_COMMAND cd ../dbow2_src && make install
```

**Command:**
```
catkin build
```

**Console output:**
```
/usr/bin/ld: cannot find -lDBoW2
collect2: error: ld returned 1 exit status
make[2]: *** [libKimeraVIO.so] Error 1
make[1]: *** [CMakeFiles/KimeraVIO.dir/all] Error 2
make: *** [all] Error 2
cd /home/ibotics/kimera-ws/build/KimeraVIO; catkin build --get-env KimeraVIO | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
............................................................................................................................................................................................................
Failed     << KimeraVIO:make                               [ Exited with code 2 ]                                                                                                                           
Failed    <<< KimeraVIO                                    [ 1 minute and 59.6 seconds ]                                                                                                                    
Abandoned <<< kimera_ros                                   [ Unrelated job failed ]
```

**Additional files:**
Please attach all the files needed to reproduce the error.

    SparkVio branch, tag or commit used VIO-Ros 6930bec
    GTSAM version used: ?
    OpenGV version used: ?
    OpenCV version used: 3.3.1-dev
    Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
    Did you change the source code? (yes / no): yes. see issues [19](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/19) and [20](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/20).

ToniRV, 2019-12-02:
Pinging @marcusabate, he is the maintainer of the dbow2_catkin repo

ana-GT, 2019-12-02:
@roelofvandijk  We ran into the same issue and fixed it by changing line 97:
From: DBoW2
To: ${DBoW2_LIBS}

You could also use ${DBoW2_LIBRARIES}. These CMake variables have the whole path to the library.  Compilation worked fine after that.

marcusabate, 2019-12-02:
@roelofvandijk I am unable to replicate this problem, but I have updated [dbow2_catkin](https://github.com/marcusabate/dbow2_catkin) with your changes since they follow the more standard format.

Can you confirm that this solves the problem?

We will soon push an update to the Kimera-VIO `CMakeLists.txt` that adds an interface library variable similar to ana-GT's suggestion.

closed 2020-03-14 by ToniRV