MIT-SPARK/Kimera-VIO-ROS #24: how run on live from a Realsense camera?

FPSychotic, 2019-12-24:
I have a realsense camera, but I cannot find how run it on live slam, only I can find ros bag based .launch files.
Other question is can Kimera use Imu data from a Pix 4 or t265 and the stereo from d435? what name should have the IMU msg and what kind of coordinate system (quaternions or other)

thanks

violetteavi, 2020-03-14:
Yes, you can run Kimera on live slam! Below is a picture of it performing live SLAM on the infrared stereo + IMU of a d435i. 

For now, you need a system like the D435i that provides synchronized stereo images and imu data. You can specify what topics you are publishing the data on in the launch file, and camera calibration parameters in the VIO params folder. We'll release a launch file and calibration files at some point in the near future. In the meantime, feel free to consult [this doc](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md) for how to do it yourself!

[image]

violetteavi, 2020-04-14:
Working parameters and launch file have not yet been merged into master, but are available in these pull requests:
[Kimera-VIO](feature/centralized_params_with_realsense)
[Kimera-VIO-ROS](https://github.com/MIT-SPARK/Kimera-VIO-ROS/pull/61)
You can check them out with:
```
cd /PATH/TO/Kimera-VIO
git checkout feature/centralized_params_with_realsense
cd /PATH/TO/Kimera-VIO-ROS
git checkout feature/realsense_metric_reconstruction_demo
```
[This guide](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/feature/realsense_metric_reconstruction_demo/docs/hardware_setup.md) has tips for setting up the RealSense drivers/firmware/etc as well as the command line commands to get the RealSense running with the given launch files.

violetteavi, 2020-04-14:
You can pair the stereo images from the d435i with IMU data from another device if you can get the device clocks synchronized. That said, the d435i has a pretty good on board accelerometer and gyroscope-- I would recommend using those.

All messages sent/accepted over ROS are in ROS standard formats. For the IMU data, it follows [this format](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/Imu.html).

shirbarzel, 2020-05-07:
> You can pair the stereo images from the d435i with IMU data from another device if you can get the device clocks synchronized. That said, the d435i has a pretty good on board accelerometer and [...]

Hi,
Have you managed to perform a stable localization/odometry with only the d435i?
I'm running the Kimera-VIO-ROS here with my D435i and I'm getting large localization errors
(Running with the kimera_vio_ros_realsense_IR.launch)

shirbarzel, 2020-05-07:
> You can pair the stereo images from the d435i with IMU data from another device if you can get the device clocks synchronized. That said, the d435i has a pretty good on board accelerometer and [...]

Have you managed to perform stable localization with only the d435i and the Kimera-VIO?

violetteavi, 2020-05-07:
Yep! I got it to do stable localization with kimera_vio_ros_realsense_IR.launch. If you are experiencing unstable localization, it could be a problem with
a) your device
b) your device driver
c) your Kimera parameters
d) Kimera itself
Given that I've gotten it working (resolving some device issues along the way) I suspect it's b) or c).
Can you run Kimera on this sample rosbag collected from my D435i?
https://drive.google.com/open?id=1ipuwygPzKnTu7I5df97dcVih_1oxqpmq
This will allow us to narrow down the possible sources of failure

violetteavi, 2020-05-07:
Also, what is your environment like? The current set of parameters is tuned for indoor environments-- close range, fairly feature dense. They may need additional tuning if you are operating outside.

closed 2020-05-11 by violetteavi