MIT-SPARK/Kimera-VIO-ROS #26: Expected image with MONO8 or BGR8 encoding

clungzta, 2019-12-30:
**Description:**
I am attempting to use Kimera ROS node with my ZED Mini Stereo Camera. However, the node does not support images of bgra8 format.

**Console output:**

```

F1230 20:35:42.739012  1157 RosDataProvider.cpp:111] Check failed: cv_ptr->encoding == sensor_msgs::image_encodings::MONO8 (bgra8 vs. mono8) Expected image with MONO8 or BGR8 encoding

```

Please give also the following information:
* SparkVio branch, tag or commit used: 6608e65e0be71a6ced3ddf53be8419dc4cbbdec0
* OpenCV version used: type `opencv_version`: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

ToniRV, 2019-12-30:
Should be easy to add support for `bgra8`.
I'll try to add it, together with #27

tftangming, 2020-08-07:
@clungzta Hello, Have you success to achieve the live meshing using your zed mini camera? Recently I bought a ZED2 and wang to use it...

hanzheteng, 2021-06-11:
@ToniRV  Have we added support for `bgra8` yet? I run into the same issue using Zed camera.