MIT-SPARK/Kimera-VIO-ROS #27: Use camera_info rather than hardcoded param files.

clungzta, 2019-12-30:
**Is your feature request related to a problem? Please describe.**
I'm trying to use Kimera-VIO-ROS with my own ZED Mini stereo camera (which came factory calibrated). However, the setup process appears tedious and error-prone; it appears that I manually have to transfer all calibration from `camera_info` (intrinsics), `tf` (extrinsics).

**Describe the solution you'd like**
It would be excellent to have the Kimera ROS `StereoFrame` node automatically fetch this information from `camera_info` and the ROS `tf` tree. 

**Describe alternatives you've considered**
I'm about to embark on the manual entry process of transferring all of my calibration values across. I'm concerned that if I make a data entry mistake, I'd have issues troubleshooting the poor calibration.

ToniRV, 2019-12-30:
Thanks @clungzta, we are going to add this feature asap, thank you!

clungzta, 2019-12-30:
Excellent, thank you!

savagecamille, 2020-03-02:
I am also looking for this feature! Any updates on when it will be available ?!?

ToniRV, 2020-03-02:
Oh, I forgot about this one, will get this done by Thursday afternoon (bit of a busy week ahead), thanks for the reminder!

ToniRV, 2020-05-10:
This is merged already.

closed 2020-05-10 by ToniRV