MIT-SPARK/Kimera-VIO-ROS #28: Crushed in "catkin build"

vulturest, 2019-12-31:
I followed your guide and met some problems in my installation,and solved these problem by others solution in issues.But after 13 packages have been installed successfully,my pc crushed when installed kimera-vio-ros (about 80%),cant finsh installtion.Anyone else has the same problem?Looking for your help.
ps:My system is Ubuntu 16.04 lts

XanxusCrypto, 2020-01-03:
Meet the same issue. During the final installation of kimera-vio the system crushed.
My system is aslo Ubuntu 16.04 lts.

ToniRV, 2020-01-03:
When using catkin, you can limit the number of processors it takes to avoid such cases, particularly when running short of compute power: `catkin build --jobs N` where `N` is less than the number of cores your computer has.
I am also noticing long build times with `catkin` which might be caused by over-conservative `#include` directives in Kimera which add extra dependencies.
(The fact that we use `catkin_simple` does not help).

vulturest, 2020-01-05:
Thanks for your response.Now my pc don't crushed but come up with another error.Error logs as follow:
/usr/bin/ld: cannot find -lopencv_dnn
/usr/bin/ld: cannot find -lopencv_bgsegm
/usr/bin/ld: cannot find -lopencv_bioinspired
/usr/bin/ld: cannot find -lopencv_ccalib
/usr/bin/ld: cannot find -lopencv_cvv
/usr/bin/ld: cannot find -lopencv_datasets
/usr/bin/ld: cannot find -lopencv_dpm
/usr/bin/ld: cannot find -lopencv_face
/usr/bin/ld: cannot find -lopencv_fuzzy
/usr/bin/ld: cannot find -lopencv_hdf
/usr/bin/ld: cannot find -lopencv_optflow
/usr/bin/ld: cannot find -lopencv_plot
/usr/bin/ld: cannot find -lopencv_reg
/usr/bin/ld: cannot find -lopencv_stereo
/usr/bin/ld: cannot find -lopencv_structured_light
/usr/bin/ld: cannot find -lopencv_surface_matching
/usr/bin/ld: cannot find -lopencv_text
/usr/bin/ld: cannot find -lopencv_tracking
/usr/bin/ld: cannot find -lopencv_xobjdetect
/usr/bin/ld: cannot find -lopencv_xphoto
collect2: error: ld returned 1 exit status
My opencv version is 3.3.1 which is installed by ROS-kinetic.How can i solve this?Thanks for your help anyway.

ToniRV, 2020-02-06:
@vulturest not sure what happen here :/ Seems like opencv is not linking correctly, perhaps you are missing some dependencies?
Closing for now, but feel free to re-open if it concerns Kimera installation.

closed 2020-02-06 by ToniRV

qmoji, 2020-03-18:
> Thanks for your response.Now my pc don't crushed but come up with another error.Error logs as follow:
> /usr/bin/ld: cannot find -lopencv_dnn
> /usr/bin/ld: cannot find -lopencv_bgsegm
> /usr/bin/ld: [...]

same problem, did you solved it?

qmoji, 2020-03-19:
> same problem, did you solved it?

i solved the problem
`moji@ubuntu-host:~/work/kimera_ws/src/Kimera-VIO-ROS$ git diff .
diff --git a/package.xml b/package.xml
index 331a801..cfcb913 100644
--- a/package.xml
+++ b/package.xml
@@ -15,8 +15,8 @@
 
   
-  <depend>opencv3_catkin</depend>
-  <depend>eigen_catkin</depend>
+  <depend>OpenCV 3.3.1</depend>
+  <depend>Eigen3</depend>
   <depend>kimera_vio</depend>
 
   <depend>roscpp</depend>`

qmoji, 2020-03-19:
> same problem, did you solved it?

solved,
moji@ubuntu-host:~/work/kimera_ws/src/Kimera-VIO-ROS$ git diff .
diff --git a/package.xml b/package.xml
index 331a801..cfcb913 100644
--- a/package.xml
+++ b/package.xml
@@ -15,8 +15,8 @@
 
   
-  <depend>opencv3_catkin</depend>
-  <depend>eigen_catkin</depend>
+  <depend>OpenCV 3.3.1</depend>
+  <depend>Eigen3</depend>
   <depend>kimera_vio</depend>
 
   <depend>roscpp</depend>