MIT-SPARK/Kimera-VIO-ROS #31: Is kimera_vio supposed to be a ROS package?

ruffsl, 2020-01-14:
In one of the core launch files, the `path_to_vocab` argument must be properly set: 
https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/6608e65e0be71a6ced3ddf53be8419dc4cbbdec0/launch/kimera_vio_ros.launch#L36

Given the use of `find` directo to locate the path to the `kimera_vio` package directory, this assumes `kimera_vio` is a ROS package that includes such assets like the vocabulary folder. However, the `kimera_vio` doesn't register itself as a package, thus this find lookup fails to locate it.

marcusabate, 2020-01-14:
Hello @ruffsl, thank you for your interest in Kimera! [Kimera-VIO](https://github.com/MIT-SPARK/Kimera-VIO) (builds as `kimera_vio`) does have a `package.xml` file in the root directory, allowing it to be built as a catkin package. In this case, the `find` command should work as expected.

If you've built `Kimera-VIO` system-wide instead of in your catkin workspace, you'll have to change that launch argument to an absolute path towards your install.

If you are building `Kimera-VIO` in your catkin workspace and the `find` command still isn't locating the package properly, please let us know and paste your full console output here.

closed 2020-02-06 by ToniRV

ruffsl, 2020-02-07:
> If you are building Kimera-VIO in your catkin workspace and the find command still isn't locating the package properly, please let us know and paste your full console output here.

Sorry for the late reply, but if run a container using the image from the Dockerfile in https://github.com/MIT-SPARK/Kimera-VIO-ROS/pull/32 , then try to use `rospkg find kimera_vio`, it should return `[rospack] Error: package 'kimera_vio' not found
`.

You could also pull the image I uploaded to dockerhub about the time I opened this issue:

```
$ docker run -it --rm ruffsl/kimera:ros bash
root@8d6cdea7e61f:/opt/ros_ws# ls ../overlay_ws/install/
COLCON_IGNORE             _local_setup_util_sh.py  kimera_vio        local_setup.ps1  local_setup.zsh  setup.ps1  setup.zsh
_local_setup_util_ps1.py  kimera_rpgo              local_setup.bash  local_setup.sh   setup.bash       setup.sh
root@8d6cdea7e61f:/opt/ros_ws# ls ../overlay_ws/install/kimera_vio/
include  lib  share
root@8d6cdea7e61f:/opt/ros_ws# rospack list | grep kimera
kimera_rviz_markers /opt/ros_ws/install/kimera_rviz_markers/share/kimera_rviz_markers
kimera_vio_ros /opt/ros_ws/install/kimera_vio_ros/share/kimera_vio_ros
root@8d6cdea7e61f:/opt/ros_ws# rospack find kimera_vio
[rospack] Error: package 'kimera_vio' not found
```