MIT-SPARK/Kimera-VIO-ROS #33: Build fails on Ubuntu 16.04 + ROS Kinetic due to missing dependencies

m-pilia, 2020-01-16:
**Description:**
The installation instructions fail on a fresh installation of Ubuntu 16.04 + ROS Kinetic (official `ros-kinetic-desktop-full` package following the [upstream ROS installation instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu)).

Apparently `mesh_rviz_plugins` depends on the following packages that are not part of a default install (replace `kinetic` with relevant name for other ROS versions):
* `ros-kinetic-image-geometry`
* `ros-kinetic-pcl-ros`
* `ros-kinetic-cv-bridge`

**Command:**

All the installation commands in [the current version of README.md](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/3e5bece281/README.md), up to
```
catkin build
```

**Console output:**
```
Errors     << mesh_rviz_plugins:cmake /home/martino/catkin_ws/logs/mesh_rviz_plugins/build.cmake.000.log                                                                                                                                                                          
CMake Error at /home/martino/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findpcl_ros.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pcl_ros", but
  CMake did not find one.

  Could not find a package configuration file provided by "pcl_ros" with any
  of the following names:

    pcl_rosConfig.cmake
    pcl_ros-config.cmake

  Add the installation prefix of "pcl_ros" to CMAKE_PREFIX_PATH or set
  "pcl_ros_DIR" to a directory containing one of the above files.  If
  "pcl_ros" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:5 (catkin_simple)
```

```
Errors     << mesh_rviz_plugins:cmake /home/martino/catkin_ws/logs/mesh_rviz_plugins/build.cmake.001.log                                                                                                                                                                          
CMake Error at /home/martino/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_bridge", but CMake did not find one.

  Could not find a package configuration file provided by "cv_bridge" with
  any of the following names:

    cv_bridgeConfig.cmake
    cv_bridge-config.cmake

  Add the installation prefix of "cv_bridge" to CMAKE_PREFIX_PATH or set
  "cv_bridge_DIR" to a directory containing one of the above files.  If
  "cv_bridge" provides a separate development package or SDK, be sure it has
  been installed.
```

```
Errors     << mesh_rviz_plugins:cmake /home/martino/catkin_ws/logs/mesh_rviz_plugins/build.cmake.002.log                                                                                                                                                                          
CMake Error at /home/martino/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findimage_geometry.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "image_geometry", but CMake did not find one.

  Could not find a package configuration file provided by "image_geometry"
  with any of the following names:

    image_geometryConfig.cmake
    image_geometry-config.cmake

  Add the installation prefix of "image_geometry" to CMAKE_PREFIX_PATH or set
  "image_geometry_DIR" to a directory containing one of the above files.  If
  "image_geometry" provides a separate development package or SDK, be sure it
  has been installed.
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: 3e5bece281
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

ToniRV, 2020-02-06:
Great, updating Readme, thanks!

closed 2020-02-06 by ToniRV

calmelo, 2021-01-27:
Have you solve the problem? I think there is a collision between libvtk5-dev and the three bags. So should we use the libvtk6 as 18.04?