MIT-SPARK/Kimera-VIO-ROS #35: CMake error on Ubuntu 16.04

m-pilia, 2020-01-16:
**Description:**
It seems the syntax used in [Kimera-VIO/cmake/kimera_vioConfig.cmake.in](https://github.com/MIT-SPARK/Kimera-VIO/blob/bceed7203/cmake/kimera_vioConfig.cmake.in) breaks the build on a fresh install of Ubuntu 16.04. Removing the version number from [line 14](https://github.com/MIT-SPARK/Kimera-VIO/blob/bceed7203/cmake/kimera_vioConfig.cmake.in#L14) seems to fix the issue.

**Command:**
```
catkin build
```

**Console output:**
```
Errors     << kimera_vio_ros:cmake /home/martino/catkin_ws/logs/kimera_vio_ros/build.cmake.003.log                                                                                                                                                                                
CMake Error at /usr/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /home/martino/catkin_ws/devel/lib/cmake/kimera_vio/kimera_vioConfig.cmake:14 (find_dependency)
  /home/martino/catkin_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:40 (find_package)
  CMakeLists.txt:16 (catkin_simple)
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: 6608e65e0b
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

sam1992xjh, 2020-02-04:
**Description:**
I get the same problem. But it can't help when I remove the version number from line 14 .
I upgrade Cmake to 3.8.2,but it's not work.[https://github.com/pabloariasal/modern-cmake-sample/issues/5]

**Additional files:**
Please attach all the files needed to reproduce the error.
Please give also the following information:
-   SparkVio branch, tag or commit used: master
-    GTSAM version used: master
-    OpenGV version used:master
-    OpenCV version used: type required vision
-    Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
- Did you change the source code? (yes / no): no

sam1992xjh, 2020-02-06:
I have the same problem.
 Removing the version number from line 14 seems to fix the issue.
like this `find_dependency(Glog REQUIRED)`

**Console output:**
Finished  <<< kimera_vio                                   [ 4 minutes and 57.0 seconds ]                                                                              
Starting  >>> kimera_vio_ros                                                                                                                                           
_________________________________________________________________________________
Errors     << kimera_vio_ros:cmake /home/xjh/catkin_kimera/logs/kimera_vio_ros/build.cmake.000.log                                                                     
CMake Error at /usr/local/share/cmake-3.5/Modules/CMakeFindDependencyMacro.cmake:45 (message):
  Invalid arguments to find_dependency
Call Stack (most recent call first):
  /home/xjh/catkin_kimera/devel/lib/cmake/kimera_vio/kimera_vioConfig.cmake:14 (find_dependency)
  /home/xjh/catkin_kimera/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:40 (find_package)
  CMakeLists.txt:16 (catkin_simple)

ToniRV, 2020-02-06:
Thank you! I'll push the fix soon.

closed 2020-03-11 by ToniRV