MIT-SPARK/Kimera-VIO-ROS #37: Kimera-VIO node crashes if loop closure detection is enabled

m-pilia, 2020-01-17:
**Description:**
When running Kimera-VIO-ROS in real-time using a Mynt Eye S camera as data source, the Kimera node crashes at startup if the loop closure detector is enabled. If the LCD is disabled, the crash does not happen.

**Command:**
```
roslaunch kimera_vio_ros kimera_ros_mymynt.launch use_lcd:=true
```

**Console output:**
```
[ INFO] [1579263518.245934011]: Starting KimeraVIO wrapper for online
[ INFO] [1579263518.261440349]: >>>>>>> Started data subscribers <<<<<<<<
I0117 14:18:38.291894 11760 RegularVioBackEnd.cpp:113] Using Regular VIO backend.
I0117 14:18:38.292043 11760 LoopClosureDetector.cpp:98] LoopClosureDetector:: Loading vocabulary from /home/martino/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml
F0117 14:18:38.414885 11881 RosOnlineDataProvider.cpp:156] Check failed: imu_single_callback_ Did you forget to register the IMU callback?
*** Check failure stack trace: ***
    @     0x7f5d24e4643d  google::LogMessage::Fail()
    @     0x7f5d24e48253  google::LogMessage::SendToLog()
    @     0x7f5d24e45fcb  google::LogMessage::Flush()
    @     0x7f5d24e48c3e  google::LogMessageFatal::~LogMessageFatal()
    @           0x47d06f  VIO::RosOnlineDataProvider::callbackIMU()
    @           0x48553b  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x49e2ee  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f5d242b0d8d  ros::SubscriptionQueue::call()
    @     0x7f5d24256838  ros::CallbackQueue::callOneCB()
    @     0x7f5d24257074  ros::CallbackQueue::callOne()
    @     0x7f5d242b4265  ros::AsyncSpinnerImpl::threadFunc()
    @     0x7f5d1bb295d5  (unknown)
    @     0x7f5d234e86ba  start_thread
    @     0x7f5d218a141d  clone
```
<details>
<summary>Full output (click to expand)</summary>

```
$ roslaunch kimera_vio_ros kimera_ros_mymynt.launch use_lcd:=true
WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/martino/.ros/log/f6f3aedc-3919-11ea-8226-a4c3f096aa6e/roslaunch-martino-ThinkPad-T480-11743.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://martino-ThinkPad-T480:37317/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros/left_cam_frame_id: left_cam
 * /kimera_vio_ros/kimera_vio_ros/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros/online_run: True
 * /kimera_vio_ros/kimera_vio_ros/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros/use_lcd: True
 * /kimera_vio_ros/kimera_vio_ros/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros (kimera_vio_ros/kimera_vio_ros)
    posegraph_viewer (pose_graph_tools/visualizer_node)

[... 211 lines cut ...]
    @     0x7f5d242b0d8d  ros::SubscriptionQueue::call()
    @     0x7f5d24256838  ros::CallbackQueue::callOneCB()
    @     0x7f5d24257074  ros::CallbackQueue::callOne()
    @     0x7f5d242b4265  ros::AsyncSpinnerImpl::threadFunc()
    @     0x7f5d1bb295d5  (unknown)
    @     0x7f5d234e86ba  start_thread
    @     0x7f5d218a141d  clone
[kimera_vio_ros/kimera_vio_ros-1] process has died [pid 11760, exit code -6, cmd /home/martino/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros --left_cam_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/LeftCameraParams.yaml   --right_cam_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/RightCameraParams.yaml   --imu_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/ImuParams.yaml   --backend_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/regularVioParameters.yaml   --frontend_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/trackerParameters.yaml   --use_lcd=true   --lcd_params_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/LCDParameters.yaml   --vocabulary_path=/home/martino/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml   --flagfile=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/flags/Mesher.flags   --flagfile=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/flags/VioBackEnd.flags   --flagfile=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/flags/RegularVioBackEnd.flags   --flagfile=/home/martino/catkin_ws/src/Kimera-VIO-ROS/param/MyMynt/flags/Visualizer3D.flags   --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --backend_type=1 --log_output=false --output_path=/home/martino/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --visualize=false --parallel_run=True left_cam:=/mynteye/left/image_raw right_cam:=/mynteye/right/image_raw imu:=/mynteye/imu/data_raw reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img:=debug_mesh_img time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros __log:=/home/martino/.ros/log/f6f3aedc-3919-11ea-8226-a4c3f096aa6e/kimera_vio_ros-kimera_vio_ros-1.log].
log file: /home/martino/.ros/log/f6f3aedc-3919-11ea-8226-a4c3f096aa6e/kimera_vio_ros-kimera_vio_ros-1*.log
```

</details>

**Additional files:**
Please attach all the files needed to reproduce the error.

* [kimera_ros_mymynt.launch](https://github.com/MIT-SPARK/Kimera-VIO-ROS/files/4076757/kimera_ros_mymynt.launch.txt) (identical to [the example MyntEye launch file](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/27d64d4e0/launch/kimera_ros_mynteye.launch) except for the name of the parameter folder)

Please give also the following information:
* SparkVio branch, tag or commit used: bceed7203f824c1bcffc0c195a186d13c5ea2b22
* GTSAM version used:  4.0.0 (342f30d148fae84c92ff71705c9e50e0a3683bda)
* OpenGV version used: 306a54e6c6b94e2048f820cdf77ef5281d4b48ad
* OpenCV version used: type `opencv_version`: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

ToniRV, 2020-02-06:
It seems like setting `use_lcd=True` skips the registration of the IMU callback.
@marcusabate  any idea?

WouterJansen, 2020-07-24:
Hi, @marcusabate @ToniRV , I know this is an older issue but currently experiencing this one due to me using older versions of Kimera VIO/ROS due to the issues persistent here:https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/90.

I'm guessing this issue was solved with the heavy rewrites of some of the components but was this issue discovered? Can I solve it somehow on my end on this older version,

morsingher, 2021-06-24:
Hi @ToniRV @marcusabate @m-pilia, did any of you solved this issue? A very similar things happens when running the default example as:

`roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch rosbag_path:="/home/c-morsingher/Downloads/V1_01_easy.bag" use_lcd:=true`

It works fine for a while, but then it crashes with the following error:

`[kimera_vio_ros/kimera_vio_ros_node-2] process has died [pid 1285605, exit code -9, cmd /home/c-morsingher/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=true --vocabulary_path=/home/c-morsingher/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/c-morsingher/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/c-morsingher/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackend.flags --flagfile=/home/c-morsingher/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackend.flags --flagfile=/home/c-morsingher/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/c-morsingher/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/c-morsingher/.ros/log/738a43aa-d4d4-11eb-9477-fdf8646b1d04/kimera_vio_ros-kimera_vio_ros_node-2.log].
log file: /home/c-morsingher/.ros/log/738a43aa-d4d4-11eb-9477-fdf8646b1d04/kimera_vio_ros-kimera_vio_ros_node-2*.log
`

If I run the same command without the `use_lcd:=true` flag, everything works just fine. Any ideas?

brytsknguyen, 2023-04-10:
Hi,

I also encounter this problem and it appears to be something wrong with the opengv or eigen. Has anybody found the solution?

[image]