MIT-SPARK/Kimera-VIO-ROS #39: Porting to ROS2

ruffsl, 2020-02-06:
I'd like to ask if there would be any interest in porting the Kimera ROS wrapper to ROS2. I'm currently in progress of developing the port, and am at a stage where I'd like to reach out to collaborate further. I have a compiling and running prototype modeled after the existing ROS1 wrapper, but included a number of performance improvement to take advantage on newer paradigms and features in ROS2. Areas I'd like to extend include:

* composable nodes so both kimera and sensor nodes can be co-located in the same process
  * https://index.ros.org/doc/ros2/Tutorials/Composition
* use of zero copy transport or loaned messages when composing nodes is not applicable
  * https://github.com/ros2/rmw_iceoryx
* use of multithreaded executors for reducing message callback latency
  * https://index.ros.org/doc/ros2/Tutorials/Intra-Process-Communication
* SROS2 Security, etc...

References:
* https://index.ros.org/doc/ros2/
* http://design.ros2.org/

ToniRV, 2020-02-06:
Hi @ruffsl, I should read more about ROS2 as I've only heard good things about it so far but I am not fully aware of its potential.
I am definitely willing to help on this effort since it will help making Kimera run even faster :+1: 
I'll have a look at the references you mention and the code changes you propose. 
Thank you very much!

ruffsl, 2020-02-07:
Awsome! I'm just now calibrating a realsense D435i, and I believe the example realsense config in the package is missing sample camera yaml files. This perhaps explains why non of the output callbacks are being triggered, though input fill callbacks are being called. I'll also update the .repos file for the draft PR so the Dockerfile builds, as I've moved the branch names around before opening the draft.

ToniRV, 2020-02-09:
I'm preparing a new push with fixed output callbacks.

ruffsl, 2020-02-10:
> I'm preparing a new push with fixed output callbacks.

Is this pointed at upstream in kimera_vio or to the draft PR pointed at kimera_vio_ros?

ToniRV, 2020-02-10:
Upstream in kimera_vio.

ruffsl, 2020-05-12:
Current progress can be tracked at my PR over at our new repo: https://github.com/MIT-SPARK/Kimera-VIO-ROS2/pull/5

closed 2020-05-12 by ruffsl