MIT-SPARK/Kimera-VIO-ROS #41: cannot find -lglog

amburkoff, 2020-02-10:
I do everything according to the instructions after the catkin build I get an error:
Errors     << kimera_vio:make /home/amburkoff/cb_ws/logs/kimera_vio/build.make.004.log              
/usr/bin/ld: cannot find -lglog
collect2: error: ld returned 1 exit status
make[2]: *** [libkimera_vio.so] Error 1
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [all] Error 2

I use ubuntu 18.04 and ros melodic

ToniRV, 2020-02-10:
Do you have glog_catkin in your workspace?

amburkoff, 2020-02-11:
Yes, i have glog_catkin in my workspace.

This command on the command line helped.
sudo apt-get install libgoogle-glog-dev libgflags-dev

Now it works, thanks.

closed 2020-02-11 by amburkoff