MIT-SPARK/Kimera-VIO-ROS #43: compile kimera-vio-ros:Error: GTSAM was built against a different version of Eigen

ACFFF, 2020-02-18:
**Description:**
Error: GTSAM was built against a different version of Eigen

**Command:**
```
# replace this line with the command(s) you used
```
catkin build kimera_vio_ros
**Console output:**
```
Errors     << kimera_vio_ros:make /home/data/projects/KimeraVIO_ws/logs/kimera_vio_ros/build.make.011.log                                                                                                  
In file included from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Matrix.h:26:0,
                 from /home/data/projects/KimeraVIO_ws/devel/include/kimera-vio/common/vio_types.h:8,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosDataProviderInterface.h:28,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosBagDataProvider.h:22,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/src/RosBagDataProvider.cpp:8:
/home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/kimera_vio_ros.dir/src/RosBagDataProvider.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Matrix.h:26:0,
                 from /home/data/projects/KimeraVIO_ws/devel/include/kimera-vio/common/vio_types.h:8,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosDataProviderInterface.h:28,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosOnlineDataProvider.h:18,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/src/RosOnlineDataProvider.cpp:8:
/home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/kimera_vio_ros.dir/src/RosOnlineDataProvider.cpp.o] Error 1
In file included from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Matrix.h:26:0,
                 from /home/data/projects/KimeraVIO_ws/devel/include/kimera-vio/common/vio_types.h:8,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosDataProviderInterface.h:28,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/RosOnlineDataProvider.h:18,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/src/RosDataProvider.cpp:11:
/home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/kimera_vio_ros.dir/src/RosDataProvider.cpp.o] Error 1
In file included from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Matrix.h:26:0,
                 from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Manifold.h:22,
                 from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Lie.h:25,
                 from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/VectorSpace.h:11,
                 from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/geometry/Point2.h:20,
                 from /home/data/projects/KimeraVIO_ws/devel/include/gtsam/geometry/Cal3_S2.h:24,
                 from /home/data/projects/KimeraVIO_ws/devel/include/kimera-vio/backend/VioBackEnd-definitions.h:23,
                 from /home/data/projects/KimeraVIO_ws/devel/include/kimera-vio/pipeline/Pipeline.h:25,
                 from /home/data/projects/KimeraVIO_ws/src/Kimera-VIO-ROS/src/KimeraVioRos.cpp:16:
/home/data/projects/KimeraVIO_ws/devel/include/gtsam/base/Vector.h:72:1: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
 static_assert(
 ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/kimera_vio_ros.dir/src/KimeraVioRos.cpp.o] Error 1
make[1]: *** [CMakeFiles/kimera_vio_ros.dir/all] Error 2
make: *** [all] Error 2
# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

ACFFF, 2020-02-18:
I used Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=ON. my Eigen version is 3.3.7

ACFFF, 2020-02-18:
maybe i have found the problem,Kimera-vio-ros uses the pcl that comes with ros. The eigen used by ros to compile and install pcl is different from the eigen version of the system used by gtsam.One method is to use ubuntu18.04, but Upgrading the system is more troublesome. When I used it, I directly commented out all the code about pcl in RosDataProvider.cpp and RosDataProviderInterface.h、package.xml, and recompiled successfully.The disadvantage is that it cannot publish point cloud data and mesh3D data.

ToniRV, 2020-02-18:
@ACFFF thanks for finding the cause of this outstanding issue! There should be a way to install GTSAM using Ros eigen though which would be nicer than commenting out pcl in the ROS wrapper.

ToniRV, 2020-03-11:
Hmm, I am also seeing this error when just compiling kimera-vio... I will need to investigate this further.

ToniRV, 2020-03-12:
The solution I ended up using is:
1. Install gtsam from source, as detailed here: 
2. Remove gtsam from the catkin workspace (or catkin ignore it: `touch CATKIN_IGNORE` inside the gtsam repo).
3. `catkin build` as usual

commit 936f1448 (2020-03-13): Use eigen_catkin to fix #43
closed 2020-03-13 by ToniRV