MIT-SPARK/Kimera-VIO-ROS #48: install   kimera-VIO-ROS1  dev_tools/linter'  failed

resplendent-star, 2020-03-07:
**Description:**
install   dev_tools/linter'  failed

**Command:**
```wstool update

```

**Console output:**
```FIRST  OUTPUT:

WARNING [vcstools] Command failed: 'git submodule update --init --recursive'
 run at: '/home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO'
 errcode: 1:
submoudle 'dev_tools/linter'（https://github.com/ToniRV/linter.git）isnot register for path'dev_tools/linter' 
cloning to '/home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO/dev_tools/linter'...
fatal: unable to access 'https://github.com/ToniRV/linter.git/': Failed to receive SOCKS4 connect request ack.
fatal: cant clone 'https://github.com/ToniRV/linter.git'
to submodule path '/home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO/dev_tools/linter'
clone 'dev_tools/linter' failed。plan to retry
cloning '/home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO/dev_tools/linter'...
fatal: unable to access 'https://github.com/ToniRV/linter.git/': Failed to receive SOCKS4 connect request ack.
fatal:cant clone 'https://github.com/ToniRV/linter.git' to  '/home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO/dev_tools/linter'
secondly try to clone 'dev_tools/linter' failed，out
[/vcstools]
[Kimera-VIO-ROS] Updating /home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO-ROS
[Kimera-VIO-ROS] Done.

SECOND OUTPUT 2:
```Exception caught during install: Error processing 'Kimera-VIO' : [Kimera-VIO] Checkout of git@github.com:MIT-SPARK/Kimera-VIO.git version master into /home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO failed.

ERROR in config: Error processing 'Kimera-VIO' : [Kimera-VIO] Checkout of git@github.com:MIT-SPARK/Kimera-VIO.git version master into /home/lzw/resplendent_code/kimera_ros1_ws/src/Kimera-VIO failed.

* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu18.04
* Did you change the source code? (yes / no): no change

ToniRV, 2020-03-11:
I haven't tested the linter in a ROS workspace, but it seems to me that there might be a problem with `ssh` vs `https` when git cloning.

It looks like for some reason git is rejecting your connection. Have you tried these steps?

https://help.github.com/articles/error-permission-denied-publickey/

resplendent-star, 2020-03-11:
thanks you  very much！when  I  let  my  PC  connect  to  the Internet provided  by  my mobile phone ，it success！ nevertheless ，thanks a  lot------------------ 原始邮件 ------------------

 
 
https://help.github.com/articles/error-permission-denied-publickey/
 
&mdash;

closed 2020-03-11 by ToniRV