MIT-SPARK/Kimera-VIO-ROS #49: Data synchronization

zxrzju, 2020-03-10:
There is a warning
> No IMU data from last frame timestamp: 1559195795711012276 to timestamp: 1559195795811099519
But measurements between the two timestamps has been added in function `fillImuQueue`.
Here is the test dataset:
[https://drive.google.com/open?id=1b9bjyJyqGGUyao-tWvSFo2yul6U5M9Nu](url)

Thank you

ToniRV, 2020-03-11:
We are about to release a fix for this issue, thanks!

ToniRV, 2020-03-11:
@zxrzju does it work now?

zxrzju, 2020-03-11:
There are some compliation errors. 
1.  > Kimera-VIO/src/mesh/MesherFactory.cpp:18:12: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_bool(serialize_mesh_2d3d, false, "Serialize 2D/3D Mesh to file.");
Kimera-VIO/src/mesh/MesherFactory.cpp:27:54: error: ‘FLAGS_serialize_mesh_2d3d’ was not declared in this scope
       return VIO::make_unique<Mesher>(mesher_params, FLAGS_serialize_mesh_2d3d);

I fixed it by replacing the file `MesherFactory.cpp` with old version.

2.  > Kimera-VIO-ROS/include/kimera_vio_ros/RosDataProviderInterface.h:33:54: fatal error: kimera-vio/frontend/VisionFrontEndParams.h: No such file or directory

I found that it should be `VioFrontEndParams.h`

3. > Kimera-VIO-ROS/src/RosDataProviderInterface.cpp:172:47: error: ‘struct VIO::FrontendOutput’ has no member named ‘feature_tracks_’
         cv_bridge::CvImage(h, "bgr8", output->feature_tracks_).toImageMsg());

I couldn't fix it.

Thank you

ToniRV, 2020-03-14:
These should be solved now! Let me know if it works.

closed 2020-05-10 by ToniRV