MIT-SPARK/Kimera-VIO-ROS #51: Rosbag error

alandszhang, 2020-03-13:
**Description:**
Thank you for your impressive work. I successfully run the Kimera-VIO-ROS, but it will crash because of the testing rosbag V1_01_easy.bag.

**Command:**
```
rosbag play --clock ~/Downloads/V1_01_easy.bag
```

**Console output:**
```

W0313 03:01:14.222038 21868 DataProviderModule.cpp:92] Waiting for IMU data...
F0313 03:01:14.765467 21868 QueueSynchronizer.h:111] Check failed: timestamp == payload_timestamp (1403715286162143000 vs. 1403715286212143000) Syncing queue data_provider_right_frame_queue in module Data Provider failed;
 Could not retrieve exact timestamp requested: 
 - Requested timestamp: 1403715286162143000
 - Actual timestamp:    1403715286212143000
*** Check failure stack trace: ***
    @     0x7f0f2015fb0d  google::LogMessage::Fail()
    @     0x7f0f201619b1  google::LogMessage::SendToLog()
    @     0x7f0f2015f63d  google::LogMessage::Flush()
    @     0x7f0f20162389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f0f1eccfb46  VIO::SimpleQueueSynchronizer<>::syncQueue()
    @     0x7f0f1ecc9cdb  VIO::DataProviderModule::getInputPacket()
    @     0x7f0f1ecd445c  VIO::PipelineModule<>::spin()
    @     0x55fb3f165087  (unknown)
    @     0x55fb3f1649c3  (unknown)
    @     0x55fb3f164ab9  (unknown)
    @     0x7f0f1f76b827  __pthread_once_slow
    @     0x55fb3f167096  (unknown)
    @     0x7f0f1e2d76ef  (unknown)
    @     0x7f0f1f7636db  start_thread
    @     0x7f0f1dd3288f  clone
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 21798, exit code -6, cmd /home/alan/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --left_cam_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/LeftCameraParams.yaml --right_cam_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/RightCameraParams.yaml --imu_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/ImuParams.yaml --backend_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/regularVioParameters.yaml --frontend_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/trackerParameters.yaml --use_lcd=false --lcd_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/LCDParameters.yaml --vocabulary_path=/home/alan/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackEnd.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackEnd.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --backend_type=1 --log_output=false --output_path=/home/alan/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --visualize=false --parallel_run=true left_cam:=/cam0/image_raw right_cam:=/cam1/image_raw imu:=/imu0 reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img:=debug_mesh_img feature_tracks:=feature_tracks time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/alan/.ros/log/4d684728-64f8-11ea-b042-080027127613/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/alan/.ros/log/4d684728-64f8-11ea-b042-080027127613/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: a5109c
* GTSAM version used:
* OpenGV version used:
* OpenCV version used:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

closed 2020-03-14 by alandszhang

violetteavi, 2020-04-30:
@alandszhang What was the issue here? And how did you fix it?

alandszhang, 2020-05-01:
It seems that the rosbag is out of date. You can try the latest version.

alandszhang, 2020-05-01:
> @alandszhang What was the issue here? And how did you fix it?

It seems that the rosbag is out of date. You can try the latest version.