MIT-SPARK/Kimera-VIO-ROS #52: Node crashes when I set log_output to true.

alandszhang, 2020-03-17:
**Description:**
When I set log_output to true in kimera_vio_ros_euroc.launch, the Kimera-VIO-ROS node crashed. 

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
```

**Console output:**
```

[ INFO] [1584437080.786460888]: Starting KimeraVIO wrapper for online
I0317 05:24:40.786512 25640 RosOnlineDataProvider.cpp:38] Requested initialization from ground-truth. Initializing ground-truth odometry one-shot subscriber.
W0317 05:24:40.789090 25640 RosOnlineDataProvider.cpp:46] Waiting for ground-truth pose to initialize VIO on ros topic: /kimera_vio_ros/gt_odom
W0317 05:24:40.950736 25640 RosOnlineDataProvider.cpp:61] Missing ground-truth pose while waiting for 3s.Enabling autoInitialize and continuing without ground-truth pose.
[ INFO] [1584437080.974957777, 1403715363.473703805]: >>>>>>> Started data subscribers <<<<<<<<
I0317 05:24:40.978472 25640 Logger.cpp:57] Opening output file: output_frontend_stats.csv
F0317 05:24:40.978513 25640 Logger.h:38] Check failed: output_file->is_open() Cannot open file: /home/alan/catkin_ws/src/Kimera-VIO-ROS/output_logs/Euroc/output_frontend_stats.csv
*** Check failure stack trace: ***
    @     0x7fdfde541b0d  google::LogMessage::Fail()
    @     0x7fdfde5439b1  google::LogMessage::SendToLog()
    @     0x7fdfde54163d  google::LogMessage::Flush()
    @     0x7fdfde544389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fdfdd244c8d  VIO::OfstreamWrapper::openLogFile()
    @     0x7fdfdd244ffd  VIO::OfstreamWrapper::OfstreamWrapper()
    @     0x7fdfdd2474b8  VIO::FrontendLogger::FrontendLogger()
    @     0x7fdfdd11c1a9  VIO::StereoVisionFrontEnd::StereoVisionFrontEnd()
    @     0x7fdfdd125b17  VIO::VisionFrontEndFactory::createFrontend()
    @     0x7fdfdd211b11  VIO::Pipeline::Pipeline()
    @     0x561e6946e749  (unknown)
    @     0x7fdfdc014b97  __libc_start_main
    @     0x561e6946f60a  (unknown)
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 25640, exit code -6, cmd /home/alan/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --left_cam_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/LeftCameraParams.yaml --right_cam_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/RightCameraParams.yaml --imu_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/ImuParams.yaml --backend_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/regularVioParameters.yaml --frontend_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/trackerParameters.yaml --use_lcd=true --lcd_params_path=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/LCDParameters.yaml --vocabulary_path=/home/alan/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackEnd.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackEnd.flags --flagfile=/home/alan/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --backend_type=1 --log_output=true --output_path=/home/alan/catkin_ws/src/Kimera-VIO-ROS/output_logs/Euroc --visualize=false --parallel_run=true left_cam:=/cam0/image_raw right_cam:=/cam1/image_raw imu:=/imu0 reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img:=debug_mesh_img feature_tracks:=feature_tracks time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/alan/.ros/log/64b5c944-682e-11ea-aac1-0800270fda36/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/alan/.ros/log/64b5c944-682e-11ea-aac1-0800270fda36/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: a5109c7157482bb1203918512ef6593e226507f7
* GTSAM version used: 4.0.2
* OpenGV version used: ddee36ad8d9bf25a26b2eed11c1ba5cef23006bf
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

ToniRV, 2020-03-18:
Hi @alandszhang, thanks for reporting, the issue is in a simple renaming from EuRoC to Euroc in launch files and param files.
I will be soon pushing a fix.
Thanks.

alandszhang, 2020-03-18:
Thanks for your quick response @ToniRV .

closed 2020-05-10 by ToniRV