MIT-SPARK/Kimera-VIO-ROS #53: Catkin Build Error

WouterJansen, 2020-03-18:
**Description:**
Building the workspace crashes due to issue I believe between kimera-vio and gtsam. I follow the tutorial.

**Command:**
```
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/
catkin init
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin config --merge-devel
echo 'source ~/catkin_ws/devel/setup.bash' >> ~/.bashrc
cd ~/catkin_ws/src
git clone https://github.com/MIT-SPARK/Kimera-VIO-ROS.git
wstool init 
wstool merge Kimera-VIO-ROS/install/kimera_vio_ros_https.rosinstall
wstool update
catkin build
```

**Console output:**
```
...omitted many warnings...
/home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::BackendLoggerFixture_logBackendOutput_Test::TestBody()’:
/home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:137:31: error: no matching function for call to ‘gtsam::Rot3::Random(boost::random::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/username/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:29:
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(std::mt19937&)
     static Rot3 Random(std::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note:   no known conversion for argument 1 from ‘boost::random::mt19937 {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘std::mt19937& {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:140:31: error: no matching function for call to ‘gtsam::Rot3::Random(boost::random::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/username/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:29:
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(std::mt19937&)
     static Rot3 Random(std::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note:   no known conversion for argument 1 from ‘boost::random::mt19937 {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘std::mt19937& {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::FrontendLoggerFixture_logFrontendRansac_Test::TestBody()’:
/home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:331:31: error: no matching function for call to ‘gtsam::Rot3::Random(boost::random::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /usr/local/include/gtsam/geometry/Pose3.h:24:0,
                 from /usr/local/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /usr/local/include/gtsam/geometry/CameraSet.h:22,
                 from /usr/local/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /usr/local/include/gtsam/slam/SmartFactorBase.h:25,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /usr/local/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/username/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/username/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:29:
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(std::mt19937&)
     static Rot3 Random(std::mt19937 & rng);
[... 63 lines cut ...]
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(std::mt19937&)
     static Rot3 Random(std::mt19937 & rng);
                 ^~~~~~
/usr/local/include/gtsam/geometry/Rot3.h:139:17: note:   no known conversion for argument 1 from ‘boost::random::mt19937 {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘std::mt19937& {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
libkimera_vio.so: undefined reference to `gtsam::Scatter::Scatter(gtsam::GaussianFactorGraph const&)'
libkimera_vio.so: undefined reference to `gtsam::GaussianFactorGraph::hessian() const'
libkimera_vio.so: undefined reference to `gtsam::HessianFactor::HessianFactor(gtsam::GaussianFactorGraph const&, gtsam::Scatter const&)'
libkimera_vio.so: undefined reference to `gtsam::fpEqual(double, double, double)'
libkimera_vio.so: undefined reference to `gtsam::GaussianFactorGraph::optimize(boost::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&) const'
libkimera_vio.so: undefined reference to `gtsam::SO<3>::Expmap(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, gtsam::OptionalJacobian<3, 3>)'
libkimera_vio.so: undefined reference to `gtsam::Marginals::Marginals(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::Marginals::Factorization)'
libkimera_vio.so: undefined reference to `gtsam::EliminateableFactorGraph<gtsam::GaussianFactorGraph>::eliminateMultifrontal(boost::optional<gtsam::Ordering::OrderingType>, boost::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&, boost::optional<gtsam::VariableIndex const&>) const'
collect2: error: ld returned 1 exit status
make[2]: *** [stereoVIOEuroc] Error 1
make[1]: *** [CMakeFiles/stereoVIOEuroc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
```

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

WouterJansen, 2020-03-18:
I also tried extending the cmake arguments to `-DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=OFF -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON` but same result. 

my catkin settings look like 
```
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/username/catkin_ws
-----------------------------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/username/catkin_ws/build
Devel Space:        [exists] /home/username/catkin_ws/devel
Install Space:      [unused] /home/username/catkin_ws/install
Log Space:          [exists] /home/username/catkin_ws/logs
Source Space:       [exists] /home/username/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-----------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
```

ToniRV, 2020-03-18:
Hi @WouterJansen,

This happens because GTSAM got an update that breaks compatibility.
It should be as easy as changing `boost::random::mt19937 rng_;` to `std::mt19937 rng_;` in `tests/testLogger.cpp`.

I will be pushing a fix this week.

S-LR, 2020-04-06:
Hi there, 
during catkin build i get another error, also in die testLogger.cpp 
/home/slr/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());

do you have an idea how to fix that ? 

Best regards
S-LR

ToniRV, 2020-04-06:
@S-LR did you update gtsam?

ujasmandavia, 2020-04-06:
@ToniRV  I tried using updated GTSAM earlier where I found it less stable then the current version and I'm finding the same issue in testLogger.cpp currently for melodic while earlier when I used it Kinetic it worked fine. I will try updating GTSAM and see if it is building, though scared of updating GTSAM.  And to mention I'm also trying to make the VIO repo compatible with opencv 4 where they are trying to change library command slightly

wangq0688, 2020-04-10:
@ToniRV 

I update my gtsam,But I have the same problem with @S-LR 
`/home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/bot/catkin_bot/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/bot/catkin_bot/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&}’
/home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::FrontendLoggerFixture_logFrontendRansac_Test::TestBody()’:
/home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:328:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/bot/catkin_bot/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/bot/catkin_bot/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, 18ul, 1812433253ul>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615u, 11ul, 4294967295u, 7ul, 2636928640u, 15ul, 4022730752u, 18ul, 1812433253u>&}’
/home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:330:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/bot/catkin_bot/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/bot/catkin_bot/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/bot/catkin_bot/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/bot/catkin_bot/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/bot/catkin_bot/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^
/home/bot/catkin_bot/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32ul, 624ul, 397ul, 31ul, 2567483615ul, 11ul, 4294967295ul, 7ul, 2636928640ul, 15ul, 4022730752ul, [... cut ...]
[ 78%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testParallelPlaneRegularBasicFactor.cpp.o
[ 80%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testParallelPlaneRegularTangentSpaceFactor.cpp.o
[ 81%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testPointPlaneFactor.cpp.o
CMakeFiles/testKimeraVIO.dir/build.make:374: recipe for target 'CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o' failed
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
`

AndreaEirale, 2020-04-10:
Hi @ToniRV!
I'm experiencing the same identical error of @S-LR and @wangq0688 

/home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::BackendLoggerFixture_logBackendOutput_Test::TestBody()’:
/home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/andrea/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::FrontendLoggerFixture_logFrontendRansac_Test::TestBody()’:
/home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:328:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/andrea/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:330:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/andrea/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/andrea/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/andrea/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/andrea/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
[... cut ...]
/home/andrea/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2

I have the last version of GTSAM but this doesn't seem to help...

S-LR, 2020-04-12:
Does someone have any solution for this ?

Thanks alot :)

wangq0688, 2020-04-17:
@S-LR  i update GTSAM and 
-DGTSAM_USE_SYSTEM_EIGEN=ON 

solved the problem

AndreaEirale, 2020-04-19:
I continue to have this problem also after the update of GTSAM. 
Does someone have a solution?

ToniRV, 2020-04-19:
@S-LR if you can't solve this with the gtsam update (which is weird, I would really make sure the new gtsam version is not only compiled but installed (either using `catkin build gtsam` or `make install` if you are building gtsam outside your catkin package)).

Then the second best option is just to fix that particular issue, it should be as easy as converting all instances of `std::mt19937` in Kimera to `boost::mt19937`, that should solve most of the errors, if new ones appear just post them here, there should be a minor amount of new errors or none.

S-LR, 2020-04-19:
@ToniRV 
Thanks alot for your help, thats very friendly , I will try this out the next days :)

Best regards and have a nice day 
SLR

SamDSchofield, 2020-04-26:
Using the develop branch of gtsam instead of master fixed this for me.

closed 2020-05-06 by WouterJansen