MIT-SPARK/Kimera-VIO-ROS #55: run kimera-vio using mynteye camera online

jguan2984-1, 2020-03-24:
**Description:**

Have you tried to run the kimera-vio-ros with mynteye-d camera?
I tried to run it online with mynteye camera, but always get the same error at the very beginning:

F0324 08:37:29.031905 15002 QueueSynchronizer.h:110] Check failed: timestamp == payload_timestamp (1585010246980287075 vs. 1585010247013477087) Syncing queue data_provider_right_frame_queue in module Data Provider failed;
 Could not retrieve exact timestamp requested: 
 - Requested timestamp: 1585010246980287075
 - Actual timestamp:    1585010247013477087

[image]

Do you know how to fix it?

violetteavi, 2020-04-30:
I'm also getting this error ([issue](https://github.com/MIT-SPARK/Kimera-VIO/issues/58)), will let you know if I find anything.

violetteavi, 2020-04-30:
Can you make a short rosbag (1-2sec) that gives you the error, and upload your launch files?

sunrentao, 2020-05-01:
@violetteavi 
> Can you make a short rosbag (1-2sec) that gives you the error, and upload your launch files?

Hello, I have the same problem, I use realsense T265

This is the kimera-vio-ros launch file

`<launch>
  <arg name="dataset_name" value="RealSenseIR"/>
  <arg name="online" default="True" />

  <arg name="log_output" default="false"/>
  <arg name="log_output_path"
       default="$(find kimera_vio_ros)/output_logs/"
       if="$(arg log_output)"/>
  <arg name="log_gt_data" default="false"/>
  <arg name="gt_topic" default=""/>
  
  <arg name="should_use_sim_time" default="false"/>

  
  

  
  <arg name="world_frame_id"     value="world"/>
  <arg name="base_link_frame_id" value="base_link"/>
  <arg name="map_frame_id"       value="map"/>
  <arg name="left_cam_frame_id"  value="left_cam"/>
  <arg name="right_cam_frame_id" value="right_cam"/>

  
  <arg name="left_cam_topic"  value="/camera/fisheye1/image_raw"/>
  <arg name="right_cam_topic" value="/camera/fisheye2/image_raw"/>
  <arg name="imu_topic"       value="/camera/imu"/>

  
  <include file="$(find kimera_vio_ros)/launch/kimera_vio_ros.launch"
           pass_all_args="true"/>

 
 <node pkg="tf" type="static_transform_publisher" name="velo_link_broadcaster"
   args="0 0 0 0 0 0 1 base_link velo_link 100"/>

</launch>`

This is the realsense_camera launch file
`<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default="t265"/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="848"/> 
  <arg name="fisheye_height"      default="800"/>
  <arg name="enable_fisheye1"     default="true"/>
  <arg name="enable_fisheye2"     default="true"/>

  <arg name="fisheye_fps"         default="30"/>

  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="62"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_sync"           default="true"/>

  <arg name="linear_accel_cov"      default="0.01"/>
  <arg name="initial_reset"         default="false"/>
  <arg name="unite_imu_method"      default="copy"/>

  <arg name="publish_odom_tf"     default="true"/>
  <arg name="topic_odom_in"             default=""/>
  <arg name="calib_odom_file"           default="/home/rentao/wheel/master-wheel/src/realsense-ros/realsense2_camera/launch/calibration_odometry.json"/>
  
  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_sync"              value="$(arg enable_sync)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye1"          value="$(arg enable_fisheye1)"/>
      <arg name="enable_fisheye2"          value="$(arg enable_fisheye2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>

      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="topic_odom_in"         value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"       value="$(arg calib_odom_file)"/>
    </include>
  </group>
  <rosparam>
      /camera/tracking_module/enable_map_preservation: true
  </rosparam>
</launch>`

Please update about this issue, Thank you.

violetteavi, 2020-05-15:
@sunrentao It looks like you are trying to use the RealSenseIR parameters. These parameters are taken from a calibration of a RealSense d435i, and so will probably not fit your T265. We'll be releasing a more in-depth look at how to transfer calibration parameters from a Kalibr imu camchain file to Kimera's param format. Watch [this issue](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/73) for more details.

As far as the QueueSynchronizer issue goes, I have some suspicions as to what is going on. I'm going to make a fresh issue addressing them.

violetteavi, 2020-05-16:
See #77