MIT-SPARK/Kimera-VIO-ROS #58: RosBagDataProvider doesn't shut itself down

violetteavi, 2020-04-09:
I found two infinite loops when using RosBagDataProvider in accordance with the DataProviderInterface expectations. Neither of them interfere with normal roslaunch usage, but I am documenting them here for future reference.

**Parallel**
A single call to RosbagDataProvider's `spin()` method loads the rosbag's contents into the pipeline's input queues, then polls and publishes the pipeline's output. There is no active mechanism currently in place to shut either down, so they both spin until the user kills the node manually. This not a huge issue as they process all the data before this infinite loop. A more elegant solution would end the infinite loop once all the data has been read and processed.

**Sequential**
It appears that neither RosbagDataProvider nor RosOnlineDataProvider have sequential modes. Their `spin()` methods spin until the ros node is terminated, so the typical sequential structure (below) deadlocks at the first call.
```
while (dataset_parser_->spin() && vio_pipeline_->spin()) { };
```

Neither of these are urgent. Sequential mode is unnecessary for ros nodes. The data is processed before deadlocking in the parallel case. However, it would be nice to refactor these classes to match the API expectations.

ToniRV, 2020-04-09:
Sequential is not needed until it is needed :laughing: 
Let's try to keep the idea that if something runs in parallel, it should be able to run in sequential mode.
I agree I have overlooked that on the ROS side of things a bit, but we should strive to make both parallel and sequential working without deadlocks ofc.

ToniRV, 2020-04-09:
Sequential is not needed until it is needed for debugging :laughing: 
Let's try to keep the idea that if something runs in parallel, it should be able to run in sequential mode.
I agree I have overlooked that on the ROS side of things a bit, but we should strive to make both parallel and sequential working without deadlocks ofc.

violetteavi, 2020-05-06:
Update: Sequential issue for the RosBagDataProvider is addressed in [this PR](https://github.com/MIT-SPARK/Kimera-VIO-ROS/pull/66). Online data provider is not sequential yet.

closed 2020-05-10 by ToniRV