MIT-SPARK/Kimera-VIO-ROS #60: Smart Factor Error on RealSense

violetteavi, 2020-04-11:
**Description:**
I'm running kimera_vio_ros on the Intel RealSense. After running reasonably well for ten seconds or so, it throws the error below. Repeating yields the same error but with a different landmark ID: 822, 5288, 23, ...

Any ideas why this is happening?

**Command:**
```
roscore
roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation
roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz
```
**Console Output:**
```
F0410 20:37:34.120515  7066 RegularVioBackEnd.cpp:558] Check failed: old_smart_factors_it != old_smart_factors->end() Landmark with id: 32090 not found in old_smart_factors_
*** Check failure stack trace: ***
    @     0x7f6f4e760b0d  google::LogMessage::Fail()
    @     0x7f6f4e7629b1  google::LogMessage::SendToLog()
    @     0x7f6f4e76063d  google::LogMessage::Flush()
    @     0x7f6f4e763389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f6f47060b6a  VIO::RegularVioBackEnd::updateExistingSmartFactor()
    @     0x7f6f47061155  VIO::RegularVioBackEnd::updateLandmarkInGraph()
    @     0x7f6f47061d23  VIO::RegularVioBackEnd::addLandmarksToGraph()
    @     0x7f6f47068a7c  VIO::RegularVioBackEnd::addVisualInertialStateAndOptimize()
    @     0x7f6f470b88b6  VIO::VioBackEnd::addVisualInertialStateAndOptimize()
    @     0x7f6f470be40f  VIO::VioBackEnd::spinOnce()
    @     0x7f6f47146c2e  VIO::VioBackEndModule::spinOnce()
    @     0x7f6f47147cd1  VIO::PipelineModule<>::spin()
    @     0x7f6f4d39c6df  (unknown)
    @     0x7f6f4cabe6db  start_thread
    @     0x7f6f4cdf788f  clone
```

**Additional files:**
[image]

Please give also the following information:
* SparkVio branch, tag or commit used: [feature/centralized_params_with_realsense](https://github.com/MIT-SPARK/Kimera-VIO/tree/feature/centralized_params_with_realsense)
* VioRos branch used: [feature/realsense_metric_reconstruction_demo](https://github.com/MIT-SPARK/Kimera-VIO-ROS/tree/feature/realsense_metric_reconstruction_demo)
* GTSAM version used: 
* OpenGV version used:
* OpenCV version used: 
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 18.04
* Did you change the source code? (yes / no): No, code is the same as in master on both branches, but the branches have different launch/parameter files

ToniRV, 2020-04-11:
What are the horizon (BackendParams) and max_feature_age (FrontendParams) parameters values? These must be proportional.

violetteavi, 2020-04-11:
`horizon: 6` and `maxFeatureAge: 25`
I didn't know what went where in the new format everything went, so I just copied the Euroc parameters for the front and back end. Euroc seems to be working fine though-- I'll see if I can reconstruct the proper values from the old format :)

violetteavi, 2020-04-11:
Reset to `horizon: 5` and `maxFeatureAge: 15` and it works!
[image]

ToniRV, 2020-04-11:
Are the Euroc params wrong??

violetteavi, 2020-04-11:
Kimera is working on Euroc with `rosbag play --clock`. Not sure why it works for Kimera but not for RealSense. The settings I changed from Euroc are [here](https://github.com/MIT-SPARK/Kimera-VIO/commit/fcf3a34e50e314a46023f4a326e20aa3c70bf1ec), not including the camera settings :)

violetteavi, 2020-04-11:
Btw I am getting some strange errors when I cover the camera-- `Indeterminant linear system detected`, as well as `No keypoint in reference frame!` -> `cv::Exception`. It works when I run it normally though-- I'll post an issue on Monday.

closed 2020-04-11 by violetteavi