MIT-SPARK/Kimera-VIO-ROS #63: opengv_catkin and gtsam causes catkin to fail

HMellor, 2020-04-17:
**Description:**
Cant build catkin workspace, see console output.

**Command:**
```
catkin build
```
**Console output:**

```
__________________________________________________________________________________________________________________
Errors     << opengv_catkin:make /home/user/catkin_ws/logs/opengv_catkin/build.make.000.log                 
Cloning into 'opengv_src'...
Already on 'master'
Submodule 'python/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'python/pybind11'
Cloning into '/home/user/catkin_ws/build/opengv_catkin/opengv_src-prefix/src/opengv_src/python/pybind11'...
Submodule 'tools/clang' (https://github.com/wjakob/clang-cindex-python3) registered for path 'python/pybind11/tools/clang'
Cloning into '/home/user/catkin_ws/build/opengv_catkin/opengv_src-prefix/src/opengv_src/python/pybind11/tools/clang'...
make[3]: warning: -jN forced in submake: disabling jobserver mode.
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[5]: *** [CMakeFiles/opengv.dir/src/absolute_pose/modules/main.cpp.o] Error 4
make[5]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[5]: *** [CMakeFiles/opengv.dir/src/relative_pose/modules/main.cpp.o] Error 4
make[4]: *** [CMakeFiles/opengv.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [opengv_src-prefix/src/opengv_src-stamp/opengv_src-build] Error 2
make[1]: *** [CMakeFiles/opengv_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/catkin_ws/build/opengv_catkin; catkin build --get-env opengv_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Failed     << opengv_catkin:make                           [ Exited with code 2 ]                                 
Failed    <<< opengv_catkin                                [ 8 minutes and 11.7 seconds ]                         
Abandoned <<< glog_catkin                                  [ Unrelated job failed ]                               
Abandoned <<< KimeraRPGO                                   [ Unrelated job failed ]                               
Abandoned <<< kimera_vio                                   [ Unrelated job failed ]                               
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                               
__________________________________________________________________________________________________________________
Errors     << gtsam:make /home/user/catkin_ws/logs/gtsam/build.make.000.log                                 
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [examples/CMakeFiles/ImuFactorExample2.dir/ImuFactorExample2.cpp.o] Error 4
make[1]: *** [examples/CMakeFiles/ImuFactorExample2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [examples/CMakeFiles/Pose2SLAMExample_graph.dir/Pose2SLAMExample_graph.cpp.o] Error 4
make[1]: *** [examples/CMakeFiles/Pose2SLAMExample_graph.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/catkin_ws/build/gtsam; catkin build --get-env gtsam | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Failed     << gtsam:make                                   [ Exited with code 2 ]                                 
Failed    <<< gtsam                                        [ 31 minutes and 27.9 seconds ]
```

Please give also the following information:
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04.4 LTS
* Did you change the source code? (yes / no): no

HMellor, 2020-04-21:
Closing this issue because I figured out it was a memory issue. To fix it you have to [increase the size of your swapfile](https://itsfoss.com/create-swap-file-linux/) (or get more RAM).

closed 2020-04-21 by HMellor

violetteavi, 2020-05-07:
Alternatively, you can limit the number of concurrent jobs that catkin launches with
`catkin build -j 4`
Thanks for raising the issue!