MIT-SPARK/Kimera-VIO-ROS #67: Large localization errors when running with only D435i

shirbarzel, 2020-05-07:
**Description:**

1. Running the Kimera-VIO-ROS with D435i
2. Normal movement around the 2 room and back to the starting point on the table.
3. Large drift in terms of localization when getting back to the starting point

**Command:**
1. roscore
2.  roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
3. rosbag play ~/d435i_recs/my_bag.bag
4. rviz

The bag was recorded using:
1. roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation
2. rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0
3. rosbag record -a

**Console output:**
Suppose to be x: 0 , y:0 , z:0
World to base_link when getting back to starting point:
  header: 
      seq: 0
      stamp: 
        secs: 1588772794
        nsecs: 545706272
      frame_id: "world"
    child_frame_id: "base_link"
    transform: 
      translation: 
        x: -0.0482988878966
        y: -0.982817650416
        z: 1.06684057522

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used: master branch
* OpenGV version used: master branch
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  ubuntu 18.04
* Did you change the source code? (yes / no): only testlogger.cpp in order to allign with gtsam

violetteavi, 2020-05-07:
Just saw that you made a new issue-- I've replied to your original question [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/24)
Can you check to see if Kimera is working properly using [this rosbag](https://drive.google.com/open?id=1ipuwygPzKnTu7I5df97dcVih_1oxqpmq)?

shirbarzel, 2020-05-10:
> Just saw that you made a new issue-- I've replied to your original question [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/24)
> Can you check to see if Kimera is working properly using [this rosbag](https://drive.google.com/open?id=1ipuwygPzKnTu7I5df97dcVih_1oxqpmq)?

Yes, it's working properly, but its a short bag. For example in my place(indoor environment), I'm recording a bag file which includes:
1. Starting from a point at my table.
2. Walking to the next room(5m away).
3. Going back to the starting point at my table.
Excepted result: world to base link tf: x=0, y=0, z=0
But I'm getting:
frame_id: "world"
    child_frame_id: "base_link"
    transform: 
      translation: 
        x: -1.25345615507
        y: -0.320596752833
        z: 0.992033035094

Thanks