MIT-SPARK/Kimera-VIO-ROS #68: kimera_vio_ros_euroc.launch is not launching as in example after clear instalation Ubuntu 18

fatrybl, 2020-05-07:
**Description:**
I just installed ubuntu 18 on my virtual machine, installed ROS Melodic following official instructions, 
then wanted to install Kimera-VIO-ROS, but during installation openVG was not installed (i dont know why). Then i tried to launch this:
but got this:

**Command:**
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch  (as in tutorial)
```

**Console output:**
```
(base) mark@ubuntu:~/catkin_ws/src$ roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch 
WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/mark/.ros/log/0921792e-9089-11ea-8c0e-000c29f1cd79/roslaunch-ubuntu-62289.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://ubuntu:36185/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/mark/catkin...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
ERROR: cannot launch node of type [kimera_vio_ros/kimera_vio_ros_node]: Cannot locate node of type [kimera_vio_ros_node] in package [kimera_vio_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [pose_graph_tools/visualizer_node]: Cannot locate node of type [visualizer_node] in package [pose_graph_tools]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18
* Did you change the source code? (yes / no): no

violetteavi, 2020-05-07:
The roslaunch error message is happening because the installation of kimera_vio_ros failed, and so roslaunch cannot find the kimera_vio_ros node.

Currently, compiling Kimera with catkin takes an enormous amount of memory. It asks for more memory than your machine has to offer, and so throws an internal compiler error. This can be fixed by limiting the number of concurrent jobs catkin can run using `catkin build kimera_vio_ros -j 4`.

violetteavi, 2020-05-07:
After you fix the internal compiler error, you may also have some issues with compiling dependencies. 
1. We are currently using gtsam's `develop` branch. I think we have it so that it automatically downloads the `develop` branch, but it would be good to check with
```
cd ~/catkin_ws/src/gtsam
git checkout develop
```
2. Sometimes, opengv and DBoW2 don't build before kimera_vio. You may need to compile them manually with `catkin build opengv_catkin dbow2_catkin`

violetteavi, 2020-05-12:
@fatrybl Did this fix your issue?

fatrybl, 2020-05-13:
Thank you for replies, I`ve tried to build with catkin build kimera_vio_ros -j 4, and got this:
https://sun9-36.userapi.com/c858020/v858020252/1f0f22/LSnE1rhziR8.jpg
Also i`ve checked this:
cd ~/catkin_ws/src/gtsam
git checkout develop
And yes, iam on develop branch.
I tried to install opengv with this : catkin build opengv_catkin dbow2_catkin and it helped.
So now seems like i have only 2 packages not properly installed

violetteavi, 2020-05-13:
Internal compiler errors are from excess memory usage (see my previous comment), and can be fixed by limiting catkin to four concurrent jobs
```
catkin build kimera_vio_ros -j 4
```

fatrybl, 2020-05-14:
> 
> 
> Internal compiler errors are from excess memory usage (see my previous comment), and can be fixed by limiting catkin to four concurrent jobs
> 
> ```
> catkin build kimera_vio_ros -j 4
> ```

Yes, I always used this command "catkin build kimera_vio_ros -j 4" to build. But the problem still exists. And i have like 8GB ram on my virtual machine. isnt it enough ?

violetteavi, 2020-05-14:
Hmmm, catkin build may use more than 8GB of RAM. Can you try 
```
catkin build kimera_vio_ros -j 1
```

fatrybl, 2020-05-15:
> Hmmm, catkin build may use more than 8GB of RAM. Can you try
> 
> ```
> catkin build kimera_vio_ros -j 1
> ```

Thank you for help. I uncreased RAM of my virtual machine up to 12GB and with "catkin build kimera_vio_ros -j 4" managed to build it

closed 2020-05-15 by violetteavi

salaheddinek, 2021-09-01:
I have Ubuntu 18, and I have the same problem:
`
ERROR: cannot launch node of type [kimera_vio_ros/kimera_vio_ros_node]: Cannot locate node of type [kimera_vio_ros_node] in package [kimera_vio_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
`
`ERROR: cannot launch node of type [pose_graph_tools/visualizer_node]: Cannot locate node of type [visualizer_node] in package [pose_graph_tools]. Make sure file exists in package path and permission is set to executable (chmod +x)
`

As proposed above, i tried running the command `catkin build kimera_vio_ros -j 4` but it did not help (btw i have 16GB RAM + swap memory).

I also verified that all my python files are executable.

HongLouyemeng, 2021-09-06:
> I have Ubuntu 18, and I have the same problem:
> `ERROR: cannot launch node of type [kimera_vio_ros/kimera_vio_ros_node]: Cannot locate node of type [kimera_vio_ros_node] in package [kimera_vio_ros]. [...]
me too.Have you solve this problem?