MIT-SPARK/Kimera-VIO-ROS #69: RAM usage keeps accumlating

shirbarzel, 2020-05-10:
**Description:**
RAM memory usage keeps accumlating in a rate of about 20MB per second

Running the Kimera-VIO-ROS with D435i
Normal movement around the 2 room and back to the starting point on the table.

**Command:**
```
roscore
roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
rosbag play ~/d435i_recs/my_bag.bag
rviz

```

**Console output:**

```
-----------                        #	Hz	(avg     +- std    )	[min,max]
Data Provider [ms]            	      0	
IMU Preintegration Timing [us]	   1745	11.5143	(133.979 +- 14.3228)	[86,2735]
Mesher [ms]                   	    284	1.87226	(21.0669 +- 5.96944)	[3,40]
VioBackEnd [ms]               	    284	1.87236	(49.3134 +- 13.4856)	[2,139]
VioFrontEnd [ms]              	   1745	11.4843	(56.1456 +- 119.395)	[5,410]
I0510 15:20:52.095813 24526 KimeraVioRos.cpp:99] Statistics

```

**Additional files:**

[image]

Please give also the following information:
SparkVio branch, tag or commit used
GTSAM version used: master branch
OpenGV version used: master branch
OpenCV version used: 3.4.2
Operating system and version (e.g. Ubuntu 16.04 or Windows 10): ubuntu 18.04
Did you change the source code? (yes / no): No, only testlogger.cpp in order to allign with gtsam

ToniRV, 2020-05-10:
Hi @shirbarzel,

Thanks for the heads-up.
Just to be sure, are you using the OpenCV visualizer or Rviz?

I was also seeing a memory leak a while ago and we figured out it was OpenCV's 3D viz module that was not freeing memory appropriately.

shirbarzel, 2020-05-10:
> Hi @shirbarzel,
> 
> Thanks for the heads-up.
> Just to be sure, are you using the OpenCV visualizer or Rviz?
> 
> I was also seeing a memory leak a while ago and we figured out it was OpenCV's 3D viz module that was not freeing memory appropriately.

Hi @ToniRV, thanks for the replay,

Actually the high RAM usage I mentioned was without running Rviz in the background:

[image]

shirbarzel, 2020-05-10:
> Hi @ToniRV, thanks for the replay,
>
> Actually the high RAM usage I mentioned was without running Rviz in the background:
>
> [image]

Update:
I'm not sure, but I think the problem was with that line in the roslaunch file:

  

Is it just viz node? (The line I commented)

ToniRV, 2020-05-10:
So you also have the --visualize flag set to 0 or false?
Meaning, you are not seeing any kind of visualization (OpenCV)?

shirbarzel, 2020-05-10:
Yes, the default value of the visualize flag in the launch file is set to false(src/Kimera-VIO-ROS/launch/kimera_vio_ros.launch)
  <arg name="visualize"    default="false" />

violetteavi, 2020-05-11:
+1. Running on some simulated data (60Hz, 1024x768), my kimera_vio_ros_node consistently accumulated ~2GB of RAM in 12 seconds. I am using RViz to visualize the feature tracks and transforms. When I shut down RViz and run Kimera, the load drops from 2.5GB -> 1.7GB. Seems like a memory leak on the order of image size every frame.

violetteavi, 2020-05-11:
I think I found the reason why my use case is leaking so horribly-- Kimera [throttles its output to 30Hz](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/71), so half of my frames are getting effectively leaked as the queue backs up.

shirbarzel, 2020-05-12:
> I think I found the reason why my use case is leaking so horribly-- Kimera [throttles its output to 30Hz](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/71), so half of my frames are getting effectively leaked as the queue backs up.
> 
> Changing the publishing rate to 120Hz kept the RAM usage to a manageable 400MB throughout

Hi @violetteavi, thanks for the replay
I've changed the param in line 363(Kimera-VIO-ROS/src/RosOnlineDataProvider.cpp)
to : 
`  ros::Rate rate(120);`

But still got huge memory usage accumulating:
[image]

violetteavi, 2020-05-12:
My next guess is that the processing speed on your device is not quite fast enough to process the D435i, leading to a backlog of image frames in the input queues. Are you experiencing any time delay between your input images and Kimera's output? This is pretty easy to check by viewing the RealSense image alongside the debug image `/kimera_vio_ros/feature_tracks/image_raw`.

violetteavi, 2020-05-12:
If the debug image looks like it's behind the RealSense image, that means Kimera isn't keeping up with the image stream. You can a) decrease the RealSense framerate to 15Hz, b) tune the Kimera parameters to decrease quality in favor of speed, or c) use a faster processor. I recommend a), because it is simple. You just need to change d435i's rs_camera.launch and the framerates in the [LeftCameraParams.yaml](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/params/RealSenseIR/LeftCameraParams.yaml#L15) and [RightCameraParams.yaml](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/params/RealSenseIR/RightCameraParams.yaml#L16).

To check to see if this will actually work, you can also artificially double your computer's processing power by running the bag at half speed 
```
rosbag play ~/d435i_recs/my_bag.bag -r 0.5
```

shirbarzel, 2020-05-12:
> My next guess is that the processing speed on your device is not quite fast enough to process the D435i, leading to a backlog of image frames in the input queues. Are you experiencing any time delay [...]

1. I'm not experiencing any time delay between my input and the kimera output, the debug image appears pretty stable.

2. 
BTW: I turned on verbosity flag in the launch file and I got this output,
I saw in the other issue that you mentioned the bold lines, what does it mean?
What can I do in order to fix it?

-----------                        #	Hz	(avg     +- std    )	[min,max]
Data Provider [ms]            	      0	
IMU Preintegration Timing [us]	   4434	30.0094	(27.7161 +- 6.52376)	[20,179]
VioBackEnd [ms]               	    734	4.9707	(1.73978 +- 0.4)	[0,10]
VioFrontEnd [ms]              	   4434	29.9895	(10.5474 +- 15.1792)	[3,80]
I0512 19:26:40.121872 10296 StereoVisionFrontEnd.cpp:83] ------------------- Processing frame k = 4436--------------------
W0512 19:26:40.121872 10229 PipelineModule.h:220] Module: Data Provider - No Input received.
I0512 19:26:40.121922 10296 StereoVisionFrontEnd.cpp:107] Current IMU Preintegration frequency: 344828 Hz. (29 us).
I0512 19:26:40.125460 10296 Tracker.cpp:123] Optical Flow Timing [ms]: 2
W0512 19:26:40.158215 10229 PipelineModule.h:220] Module: Data Provider - No Input received.
I0512 19:26:40.166760 10296 FeatureDetector.cpp:180] Number of points detected : 800
I0512 19:26:40.166776 10296 FeatureDetector.cpp:192] Need n corners: 87
I0512 19:26:40.166806 10296 NonMaximumSuppression.cpp:87] Running RangeTree: 4
I0512 19:26:40.167212 10296 NonMaximumSuppression.cpp:105] Non Maximum Suppression Timing [ms]: 0
I0512 19:26:40.173795 10296 FeatureDetector.cpp:236] Corner Sub Pixel Refinement Timing [ms]: 6
I0512 19:26:40.178676 10296 StereoVisionFrontEnd.cpp:146] Keyframe 4436 with: 791 smart measurements
**I0512 19:26:40.178715 10296 ThreadsafeQueue.h:270] Queue with id: Frame Rate Frontend output ROS is getting full, size: 4426
I0512 19:26:40.178722 10296 ThreadsafeQueue.h:270] Queue with id: Keyframe Rate Frontend output ROS is getting full, size: 732**
I0512 19:26:40.178730 10296 StereoVisionFrontEnd.cpp:83] ------------------- Processing frame k = 4437--------------------
I0512 19:26:40.178771 10296 StereoVisionFrontEnd.cpp:107] Current IMU Preintegration frequency: 285714 Hz. (35 us).
I0512 19:26:40.178915 10298 RegularVioBackEnd.cpp:212] Tracker has a LOW_DISPARITY status.
I0512 19:26:40.180821 10298 VioBackEnd.cpp:1181] Backend: Update IMU Bias.
**I0512 19:26:40.180860 10298 ThreadsafeQueue.h:270] Queue with id: Backend output ROS is getting full, size: 732**
I0512 19:26:40.182718 10296 Tracker.cpp:123] Optical Flow Timing [ms]: 3
**I0512 19:26:40.184615 10296 ThreadsafeQueue.h:270] Queue with id: Frame Rate Frontend output ROS is getting full, size: 4427**
I0512 19:26:40.192183 10296 StereoVisionFrontEnd.cpp:83] ------------------- Processing frame k = 4438--------------------

violetteavi, 2020-05-12:
! Wow! I'm pretty sure `Backend output ROS` is where Kimera-VIO puts keyframes that it has processed. If you have 732 frames in that queue, Kimera is running fast enough but you have a serious publishing backlog. I'm not sure how that would happen except if your publishing rate is super slow-- you did change it to 120Hz, right?

shirbarzel, 2020-05-12:
> ! Wow! I'm pretty sure `Backend output ROS` is where Kimera-VIO puts keyframes that it has processed. If you have 732 frames in that queue, Kimera is running fast enough but you have a serious [...]

Yes

shirbarzel, 2020-05-13:
I tried valgrind and thats the result:

[my_executable_heap_profiling.pdf](https://github.com/MIT-SPARK/Kimera-VIO-ROS/files/4620375/my_executable_heap_profiling.pdf)

violetteavi, 2020-05-13:
This confirms what I suspected-- 90% RAM usage is from the creation of the stereo image packet.
[image]

Pair this with the queue logs you posted, and I think the images are getting processed all the way through and just not published. Fortunately, the latest PR introduced asynchronous publishing from VIO callbacks. Can you pull VIO and VIO-ROS master, rebuild, and try it again?

ToniRV, 2020-05-13:
By the way, touching on the heap profiling, it would be ideal if we managed to get rid of the copy of the images in the `readRosImage` function, but that is currently not possible (I think, maybe it's fixed now), because something in the frontend might be modifying the images or something (you can try to switch off the copying of the images and just use the pointers, but that leads to weird behaviour).

shirbarzel, 2020-05-13:
> This confirms what I suspected-- 90% RAM usage is from the creation of the stereo image packet.
> [image]
>
> Pair this with the queue logs you posted, and I think the images are getting processed all [...]

Thanks!
But after I pulled I got this compilation error:

`Errors     << kimera_vio:make /home/shir/catkin_ws/logs/kimera_vio/build.make.002.log                                                                                                                      
/home/shir/catkin_ws/build/kimera_vio/testKimeraVIO: symbol lookup error: /home/shir/catkin_ws/build/kimera_vio/testKimeraVIO: undefined symbol: _ZN3VIO7TrackerC1ERKNS_14FrontendParamsERKNS_12CameraParamsEPNS_15ThreadsafeQueueISt10unique_ptrINS_16DisplayInputBaseESt14default_deleteIS9_EEEE
CMake Error at /usr/share/cmake-3.10/Modules/GoogleTestAddTests.cmake:39 (message):
  Error running test executable.

    Path: '/home/shir/catkin_ws/build/kimera_vio/testKimeraVIO'
    Result: 127
    Output:
   

`

violetteavi, 2020-05-13:
I guess that works. Most of the weird compile errors I encounter are a result of cached files and are fixed by cleaning and rebuilding.
```
catkin clean kimera_vio kimera_vio_ros
catkin build kimera_vio_ros -j 4
```

violetteavi, 2020-05-13:
If your tests aren't compiling and the pipeline is behaving strangely, it is probably due to artifacts left over from a previous build. Though it takes a while (30 min on my machine), I would recommend cleaning and rebuilding Kimera with the commands above ^

shirbarzel, 2020-05-13:
> > This confirms what I suspected-- 90% RAM usage is from the creation of the stereo image packet.
> > [image]
> > Pair this with the queue logs you posted, and I think the images are getting processed all the way through and just not published. Fortunately, the latest [PR](https://github.com/MIT-SPARK/Kimera-VIO-ROS/pull/72) introduced asynchronous publishing from VIO callbacks. Can you pull VIO and VIO-ROS master, rebuild, and try it again?
> 
> Thanks!
> But after I pulled I got this compilation error:
> 
> `/home/shir/catkin_ws/build/kimera_vio/testKimeraVIO: symbol lookup error: /home/shir/catkin_ws/build/kimera_vio/testKimeraVIO: undefined symbol: _ZN3VIO7TrackerC1ERKNS_14FrontendParamsERKNS_12CameraParamsEPNS_15ThreadsafeQueueISt10unique_ptrINS_16DisplayInputBaseESt14default_deleteIS9_EEEE CMake Error at /usr/share/cmake-3.10/Modules/GoogleTestAddTests.cmake:39 (message): Error running test executable. `
> 
> For now I commented the tests in the cmakelists file
> 
> Update:
> Thanks!! the RAM usage now looks much much better after I pulled!
> 
> But, after a short time I got this error msgs and all the pipeline got stuck:
> 
> `E0513 21:46:36.629834 3909 Mesher.cpp:1668] Keypoints supposedly inside the image: E0513 21:46:36.629870 3909 Mesher.cpp:1670] x: 515.911, y: 392.432 E0513 21:46:36.629881 3909 Mesher.cpp:1670] x: 228, y: 281 E0513 21:46:36.629891 3909 Mesher.cpp:1670] x: 117, y: 116 E0513 21:46:36.629901 3909 Mesher.cpp:1670] x: 152, y: 54 E0513 21:46:36.629907 3909 Mesher.cpp:1670] x: 347.119, y: 360.049 E0513 21:46:36.629914 3909 Mesher.cpp:1670] x: 134, y: 222 E0513 21:46:36.629935 3909 Mesher.cpp:1670] x: 503, y: 211 E0513 21:46:36.629961 3909 Mesher.cpp:1670] x: 280.655, y: 101.233 E0513 21:46:36.629968 3909 Mesher.cpp:1670] x: 127, y: 48 E0513 21:46:36.629992 3909 Mesher.cpp:1670] x: 489, y: 258 E0513 21:46:36.629997 3909 Mesher.cpp:1670] x: 227, y: 8 E0513 21:46:36.630004 3909 Mesher.cpp:1670] x: 165.336, y: 411.89 E0513 21:46:36.630012 3909 Mesher.cpp:1670] x: 408.988, y: 372.457 E0513 21:46:36.630018 3909 Mesher.cpp:1670] x: 492, y: 234 E0513 21:46:36.630026 3909 Mesher.cpp:1670] x: 211.646, y: 281.072 E0513 21:46:36.630033 3909 Mesher.cpp:1670] x: 141, y: 337 E0513 21:46:36.630059 3909 Mesher.cpp:1670] x: 273.718, y: 286.706 E0513 21:46:36.630066 3909 Mesher.cpp:1670] x: 295.961, y: 357.657 E0513 21:46:36.630074 3909 Mesher.cpp:1670] x: 150, y: 331 E0513 21:46:36.630079 3909 Mesher.cpp:1670] x: 285, y: 287 E0513 21:46:36.630086 3909 Mesher.cpp:1670] x: 136, y: 45 E0513 21:46:36.630093 3909 Mesher.cpp:1670] x: 422, y: 247 E0513 21:46:36.630101 3909 Mesher.cpp:1670] x: 235, y: 256 E0513 21:46:36.630111 3909 Mesher.cpp:1670] x: 299, y: 73 E0513 21:46:36.630120 3909 Mesher.cpp:1670] x: 191, y: 279 E0513 21:46:36.630126 3909 Mesher.cpp:1670] x: 473, y: 39 E0513 21:46:36.630133 3909 Mesher.cpp:1670] x: 68.86, y: 311.786 E0513 21:46:36.630141 3909 Mesher.cpp:1670] x: 204.073, y: 75.5732 E0513 21:46:36.630147 3909 Mesher.cpp:1670] x: 120, y: 126 E0513 21:46:36.630154 3909 Mesher.cpp:1670] x: 93, y: 453 E0513 21:46:36.630162 3909 Mesher.cpp:1670] x: 446, y: 249 E0513 21:46:36.630168 3909 Mesher.cpp:1670] x: 69.7736, y: 295.884 E0513 21:46:36.630177 3909 Mesher.cpp:1670] x: 122, y: 96 E0513 21:46:36.630183 3909 Mesher.cpp:1670] x: 147, y: 68 E0513 21:46:36.630189 3909 Mesher.cpp:1670] x: 199, y: 187 E0513 21:46:36.630198 3909 Mesher.cpp:1670] x: 367, y: 273 E0513 21:46:36.630204 3909 Mesher.cpp:1670] x: 450.498, y: 229.4 E0513 21:46:36.630211 3909 Mesher.cpp:1670] x: 122, y: 360 E0513 21:46:36.630220 3909 Mesher.cpp:1670] x: 277, y: 137 E0513 21:46:36.630226 3909 Mesher.cpp:1670] x: 423.181, y: 75.8536 E0513 21:46:36.630234 3909 Mesher.cpp:1670] x: 137, y: 31 E0513 21:46:36.630241 3909 Mesher.cpp:1670] x: 460, y: 102 E0513 21:46:36.630249 3909 Mesher.cpp:1670] x: 68, y: 377 E0513 21:46:36.630255 3909 Mesher.cpp:1670] x: 106, y: 209 E0513 21:46:36.630262 3909 Mesher.cpp:1670] x: 141, y: 246 E0513 21:46:36.630270 3909 Mesher.cpp:1670] x: 157, y: 10 E0513 21:46:36.630275 3909 Mesher.cpp:1670] x: 352, y: 272 E0513 21:46:36.630282 3909 Mesher.cpp:1670] x: 470, y: 150 E0513 21:46:36.630306 3909 Mesher.cpp:1670] x: 485.572, y: 216.766 E0513 21:46:36.630312 3909 Mesher.cpp:1670] x: 474, y: 25 E0513 21:46:36.630318 3909 Mesher.cpp:1670] x: 475, y: 61 E0513 21:46:36.630324 3909 Mesher.cpp:1670] x: 323, y: 74 E0513 21:46:36.630331 3909 Mesher.cpp:1670] x: 472, y: 13 E0513 21:46:36.630336 3909 Mesher.cpp:1670] x: 89, y: 333 E0513 21:46:36.630344 3909 Mesher.cpp:1670] x: 346, y: 6 E0513 21:46:36.630352 3909 Mesher.cpp:1670] x: 280, y: 148 E0513 21:46:36.630357 3909 Mesher.cpp:1670] x: 393.89, y: 275.197 E0513 21:46:36.630364 3909 Mesher.cpp:1670] x: 103.192, y: 329.776 E0513 21:46:36.630373 3909 Mesher.cpp:1670] x: 282, y: 11 E0513 21:46:36.630378 3909 Mesher.cpp:1670] x: 276.983, y: 191.104 E0513 21:46:36.630385 3909 Mesher.cpp:1670] x: 267, y: 268 E0513 21:46:36.630393 3909 Mesher.cpp:1670] x: 348.254, y: 30.795 E0513 21:46:36.630398 3909 Mesher.cpp:1670] x: 275, y: 176 E0513 21:46:36.630405 3909 Mesher.cpp:1670] x: 279, y: 117 E0513 21:46:36.630411 3909 Mesher.cpp:1670] x: 128.687, y: 396.24 E0513 21:46:36.630419 3909 Mesher.cpp:1670] x: 297, y: 135 E0513 21:46:36.630425 3909 Mesher.cpp:1670] x: 281, y: 68 E0513 21:46:36.630432 3909 Mesher.cpp:1670] x: 323, y: 83 E0513 21:46:36.630439 3909 Mesher.cpp:1670] x: 183, y: 65 E0513 21:46:36.630446 3909 Mesher.cpp:1670] x: 376, y: 274 E0513 21:46:3 [... cut ...]  Mesher.cpp:1670] x: 424.218, y: 114.233 E0513 21:46:36.634963 3909 Mesher.cpp:1670] x: 141.459, y: 164.865 E0513 21:46:36.634969 3909 Mesher.cpp:1670] x: 142.588, y: 126.095 E0513 21:46:36.634977 3909 Mesher.cpp:1672] CreateMesh2D: subdiv.insert error (2). A point is outside of the triangulation specified rect... Rectangle size (x, y, height, width): 0, 0, 480, 640 Keypoints to triangulate: 693 Keypoints inside image: 693 `

Thanks, any update regarding the error I got in the Mesher.cpp (above)

violetteavi, 2020-05-13:
Did the Mesher error persist when you cleaned and rebuilt Kimera? If so, please open a separate issue, as it is likely unrelated to the RAM usage :)

ToniRV, 2020-05-13:
Closing as it seems to be solved with latest PR #72

closed 2020-05-13 by ToniRV

shirbarzel, 2020-05-14:
@ToniRV @violetteavi  Please open back this issue, I'm still getting the same RAM usage problems.
The problem is that when I'm setting the 'visualize' flag to false I'm not getting any data of /tf or odometry so it doesn't help me at all