MIT-SPARK/Kimera-VIO-ROS #70: catkin build error

Het369, 2020-05-11:
**Description:**
The catkin build gives error in building the package kimera_vio

**Command:**
```
catkin build
```

**Console output:**
```

Errors     << kimera_vio:make /home/het3/mowito_ws/logs/kimera_vio/build.make.011.log                                                                                                                      
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFDefaultStripSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadTile@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `_TIFFfree@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFGetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFClose@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFGetFieldDefaulted@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFNumberOfDirectories@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFOpen@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadRGBAImage@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `_TIFFmalloc@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFNumberOfTiles@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
make[2]: *** [stereoVIOEuroc] Error 1
make[1]: *** [CMakeFiles/stereoVIOEuroc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFDefaultStripSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadTile@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `_TIFFfree@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFGetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFClose@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFGetFieldDefaulted@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFNumberOfDirectories@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFOpen@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadRGBAImage@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetField@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `_TIFFmalloc@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFReadScanline@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libvtkIOImage-6.2.so.6.2: undefined reference to `TIFFNumberOfTiles@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
make[2]: *** [testKimeraVIO] Error 1
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
cd /home/het3/mowito_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << kimera_vio:make           [ Exited with code 2 ]                                                                                                                                             
Failed    <<< kimera_vio                [ 3.9 seconds ]    

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used: 
* OpenGV version used:
* OpenCV version used: 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04
* Did you change the source code? (yes / no): no

violetteavi, 2020-05-11:
It looks like you don't have `libtiff` installed on your system. `sudo apt-get install libtiff5-dev` should do the trick for 64 bit Ubuntu 16.04 systems.

Het369, 2020-05-12:
@violetteavi 
Thanks for your help 
I had that already installed on my system. I also removed it and installed it again but the same error appears.

violetteavi, 2020-05-12:
Hmmm, interesting. Can you try building opencv by itself, then Kimera by itself?
```
catkin build opencv_catkin
catkin build kimera_vio
```

Het369, 2020-05-12:
This is not working as well, I tried building packages separately
`
catkin build opencv3_catkin
catkin build kimera_vio
`

violetteavi, 2020-05-12:
I dug around on the internet a bit; it seems libtiff conflicts if you install it with `apt-get` and `conda`. Try:
```
conda uninstall libtiff
```
If that doesn't work, I recommend reading through [this thread](https://github.com/colmap/colmap/issues/188), as there are some other ideas there.

Het369, 2020-05-14:
I didn't try that.
I have upgraded my system to Ubuntu 18.04 with ROS Melodic. It's working fine now.  

Thanks for helping.

closed 2020-05-14 by Het369