MIT-SPARK/Kimera-VIO-ROS #74: Build error in Kimera-VIO-ros

realdealneil, 2020-05-14:
**Description:**
I am getting a build error when following the build instructions.  I am building using ros-melodic on Ubuntu 18.04.  I have fulfilled all of the prerequisites mentioned on the build page.  

**Command:**
```
catkin build
```

**Console output:**
```
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::BackendLoggerFixture_logBackendOutput_Test::TestBody()’:
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Pose3.h:24,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: ‘static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)’
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937’ {aka ‘std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>’} to ‘boost::random::mt19937&’ {aka ‘boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&’}
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::FrontendLoggerFixture_logFrontendRansac_Test::TestBody()’:
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:328:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Pose3.h:24,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: ‘static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)’
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937’ {aka ‘std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>’} to ‘boost::random::mt19937&’ {aka ‘boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&’}
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:330:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Pose3.h:24,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: ‘static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)’
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937’ {aka ‘std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>’} to ‘boost::random::mt19937&’ {aka ‘boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&’}
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::LoopClosureDetectorLoggerFixture_logOptimizedTraj_Test::TestBody()’:
/home/subt-neil/devel/kimera/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:410:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/Pose3.h:24,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/subt-neil/devel/kimera/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
[... 57 lines cut ...]

The primary error appears to be this:
```
kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
Kimera-VIO, master branch.  git log shows "commit a913c97d2fbc3310166448776fa60451583c35ff"
* GTSAM version used:
master branch, tag 4.0.2 (342f30d148fae84c92ff71705c9e50e0a3683bda)
* OpenGV version used:
master branch, (306a54e6c6b94e2048f820cdf77ef5281d4b48ad)
* OpenCV version used: type `opencv_version`
3.4.2 (with Cuda)
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
Ubuntu 18.04
* Did you change the source code? (yes / no): 
no

I allowed kimera-VIO-ROS to fetch all the necessary packages.  I have built and installed OpenCV 3.4.2 (with Cuda) to /usr/local.

violetteavi, 2020-05-14:
Kimera needs the `develop` branch of `gtsam`. See [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/68#issuecomment-625412075)

realdealneil, 2020-05-14:
Ok, I'll change that.  Does that mean that the .rosinstall file needs to be changed to check out that branch, then?  I let the rosinstall do everything...

violetteavi, 2020-05-14:
Yes, good point. This [line](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/install/kimera_vio_ros_ssh.rosinstall#L16) needs to be develop. I think we had it that way before-- @ToniRV any reason why we have `master` branch in the launch file?

ToniRV, 2020-05-14:
Ahh should be `develop` yes.

realdealneil, 2020-05-14:
OK, switching my gtsam to develop got things to compile.  I am now having runtime errors when I launch the euroc example.  Here's the output:

```
WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/subt-neil/.ros/log/df26984a-962b-11ea-aab2-7085c2526b2b/roslaunch-subtneil-desktop-21026.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://subtneil-desktop:33489/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/subt-neil/d...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "KimeraRPGO" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [21041]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [21042]
[ INFO] [1589492596.458648609]: Initializing pose graph visualizer
visualizer_node: /home/subt-neil/devel/kimera/kimera_ws/src/pose_graph_tools/src/visualizer.cpp:11: Visualizer::Visualizer(ros::NodeHandle&): Assertion `nh.getParam("frame_id", frame_id_)' failed.
[kimera_vio_ros/posegraph_viewer-2] process has died [pid 21042, exit code -6, cmd /home/subt-neil/devel/kimera/kimera_ws/devel/lib/pose_graph_tools/visualizer_node graph:=pose_graph __name:=posegraph_viewer __log:=/home/subt-neil/.ros/log/df26984a-962b-11ea-aab2-7085c2526b2b/kimera_vio_ros-posegraph_viewer-2.log].
log file: /home/subt-neil/.ros/log/df26984a-962b-11ea-aab2-7085c2526b2b/kimera_vio_ros-posegraph_viewer-2*.log
F0514 15:43:16.770548 21041 YamlParser.h:44] Check failed: file_handle.type() != cv::FileNode::NONE (0 vs. 0) Missing parameter: maxFeaturesPerFrame in file: `ZژU
*** Check failure stack trace: ***
    @     0x7fec8e422c4d  google::LogMessage::Fail()
    @     0x7fec8e424ab3  google::LogMessage::SendToLog()
    @     0x7fec8e4227a8  google::LogMessage::Flush()
    @     0x7fec8e425519  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fec867e2e07  VIO::FeatureDetectorParams::parseYAML()
    @     0x7fec867ce84b  VIO::FrontendParams::parseYAML()
    @     0x7fec868dea41  VIO::VioParams::parseYAML()
    @     0x7fec868dffb1  VIO::VioParams::VioParams()
    @     0x7fec868e01d3  VIO::VioParams::VioParams()
    @     0x7fec8dad9e0d  _ZN9__gnu_cxx13new_allocatorIN3VIO9VioParamsEE9constructIS2_JRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEEvPT_DpOT0_
    @     0x7fec8dad8f34  _ZNSt16allocator_traitsISaIN3VIO9VioParamsEEE9constructIS1_JRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEEvRS2_PT_DpOT0_
    @     0x7fec8dad7686  _ZNSt23_Sp_counted_ptr_inplaceIN3VIO9VioParamsESaIS1_ELN9__gnu_cxx12_Lock_policyE2EEC1IJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEES2_DpOT_
    @     0x7fec8dad5642  _ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EEC2IN3VIO9VioParamsESaIS5_EJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEERPT_St20_Sp_alloc_shared_tagIT0_EDpOT1_
    @     0x7fec8dad3330  _ZNSt12__shared_ptrIN3VIO9VioParamsELN9__gnu_cxx12_Lock_policyE2EEC2ISaIS1_EJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEESt20_Sp_alloc_shared_tagIT_EDpOT0_
    @     0x7fec8dacfe35  _ZNSt10shared_ptrIN3VIO9VioParamsEEC2ISaIS1_EJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEESt20_Sp_alloc_shared_tagIT_EDpOT0_
    @     0x7fec8dacb91d  _ZSt15allocate_sharedIN3VIO9VioParamsESaIS1_EJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEESt10shared_ptrIT_ERKT0_DpOT1_
    @     0x7fec8dac73e0  _ZSt11make_sharedIN3VIO9VioParamsEJRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEESt10shared_ptrIT_EDpOT0_
    @     0x7fec8dabd479  VIO::KimeraVioRos::KimeraVioRos()
    @     0x5598d8dd6ab7  (unknown)
    @     0x7fec8c6c1b97  __libc_start_main
    @     0x5598d8dd66ea  (unknown)
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 21041, exit code -6, cmd /home/subt-neil/devel/kimera/kimera_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/subt-neil/devel/kimera/kimer [... cut ...] a_vio_ros_node-1.log].
log file: /home/subt-neil/.ros/log/df26984a-962b-11ea-aab2-7085c2526b2b/kimera_vio_ros-kimera_vio_ros_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any ideas what this is?  I'm going to try a clean build first and then dig into the subcomponents that are launched to try to understand which node is causing the crash (I think it's kimera-vio-ros-posegraph_viewer)

realdealneil, 2020-05-14:
I tried commenting out the kimera-vio-ros-posegraph_viewer from the launch file, and I still get a fault when launching...here's the critical line (same as above):

```
F0514 15:53:38.670065 21332 YamlParser.h:44] Check failed: file_handle.type() != cv::FileNode::NONE (0 vs. 0) Missing parameter: maxFeaturesPerFrame in file: `5�=�U
```
Looks like it's looking for a bogus file name?  (failing to parse a yaml file?)

gkiri, 2020-05-15:
I faced the same issue .
To bypass the error temporarily, I commented out those test cases that use Rot3::Random(rng_) in
kimera_ws/src/Kimera-VIO/tests/testLogger.cpp

realdealneil, 2020-05-15:
Sorry, I kind of mixed this question with a new one.  I'll close this, but one final comment: If you just update the .rosinstall files to checkout the 'develop' branch of gtsam and you redo the whole wstool update process (where it only checks out gtsam develop branch, never checking out the master branch), the build fails.  I think the git submodules for gtsam are not included in the develop branch...

closed 2020-05-15 by realdealneil