MIT-SPARK/Kimera-VIO-ROS #76: OpenCV problem

zhaozhongch, 2020-05-15:
This question may not be smart.
 I met OpenCV issue when I install the VIO.
```
CMake Error at /home/zhaozhong/test/kimera_ws/src/Kimera-VIO/CMakeLists.txt:36 (find_package):
  Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "3.3.1".

  The following configuration files were considered but not accepted:

    /usr/share/OpenCV/OpenCVConfig.cmake, version: 3.2.0

```
It feels like `opencv3_catkin` does the trick and will download the opencv3.4.2? But to me that package doesn't help. I directly meet the above error when I use `catkin build`.
Should I just download and compile the opencv3.4.2 from the source by myself? I am afraid the OpenCV version will have a conflict.

violetteavi, 2020-05-15:
I haven't seen that particular error before. One thing to try is to build the opencv3_catkin package before building all of it.
```
catkin build opencv3_catkin
catkin build
```
For other common build errors and their solutions, see [here](https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/68)

zhaozhongch, 2020-05-16:
Thanks for the fast reply.
I tried to compile `opencv3_catkin` separately. Then I got the error
```
Errors     << opencv3_catkin:make /home/zhaozhong/test/kimera_vio_ws/logs/opencv3_catkin/build.make.004.log
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: gflags_shared;gflags_nothreads_shared

  Targets not yet defined: gflags_static;gflags_nothreads_static

```
So I tried to build gflags_catkin, glog_catkin... everything in the workspace separately. They all build successfully except opencv3 and kimera_vio and the above error still exists.
Seems I have to try to install opencv3.4.2 from the source by myself. I'll let you know if it works

zhaozhongch, 2020-05-17:
Finally, I decide to move to docker. Inside docker, I install a brand new Ubuntu18+ros melodic with only kimera. The above error doesn't show now but I met another one during compiling `kimera-vio` package
```
/mnt/shared/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function 'virtual void VIO::BackendLoggerFixture_logBackendOutput_Test::TestBody()':
/mnt/shared/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:136:31: error: no matching function for call to 'gtsam::Rot3::Random(std::mt19937&)'
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /mnt/shared/kimera_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /mnt/shared/kimera_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /mnt/shared/kimera_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /mnt/shared/kimera_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /mnt/shared/kimera_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /mnt/shared/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /mnt/shared/kimera_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /mnt/shared/kimera_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /mnt/shared/kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/mnt/shared/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/mnt/shared/kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from 'std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}' to 'boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}'
```
I avoid this by disabling the `test` code during compiling
```
catkin build -DBUILD_TESTS=OFF
```
Finally, build success now. 

Then I tried to run the demo Euroc (Not sure if I should open a new issue because this one is not that related to opencv now as the title).
I run the following three commands as the documentation said in three terminals
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_eur.rviz
rosbag play MH_01_easy.bag
```

[image]
No image is showing. But seems the system is running because the terminal has some output
```
I0516 21:28:11.756561 28361 KimeraVioRos.cpp:114] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Data Provider [ms]                      	    0	
Display [ms]                            	  366	9.90263	{0.0628415 +- 0.00000}	[0,23]
Mesher [ms]                             	  183	4.99951	{0.371585 +- 0.481999}	[0,7]
VioBackEnd [ms]                         	  183	4.99962	{4.05464 +- 1.88720}	[0,17]
VioFrontEnd Frame Rate [ms]             	  551	15.0380	{4.96733 +- 3.10236}	[2,16]
VioFrontEnd Keyframe Rate [ms]          	  183	5.00045	{47.9235 +- 9.28565}	[17,70]
VioFrontEnd [ms]                        	  734	20.0047	{15.7561 +- 16.9402}	[2,70]
Visualizer [ms]                         	  183	4.99950	{0.00000 +- 0.00000}	[0,0]
backend_input_queue Size [#]            	  183	3.02257	{1.00000 +- 0.00000}	[1,1]
data_provider_left_frame_queue Size [#] 	  736	12.1496	{1.00000 +- 0.00000}	[1,1]
data_provider_right_frame_queue Size [#]	  736	12.1496	{1.00000 +- 0.00000}	[1,1]
display_input_queue Size [#]            	  366	6.04465	{1.00000 +- 0.00000}	[1,1]
mesher_backend Size [#]                 	  183	3.02251	{1.00000 +- 0.00000}	[1,1]
mesher_frontend Size [#]                	  734	12.1151	{2.98774 +- 1.22285}	[1,5]
stereo_frontend_input_queue Size [#]    	  734	12.1151	{1.00000 +- 0.00000}	[1,1]
visualizer_backend_queue Size [#]       	  183	3.02251	{1.00000 +- 0.00000}	[1,1]
visualizer_frontend_queue Size [#]      	  734	12.1151	{2.99864 +- 1.22680}	[1,5]
visualizer_mesher_queue Size [#]        	  183	3.02247	{1.00000 +- 0.00000}	[1,1]
```

Anyone meet this problem before??? I'll really appreciate your answer!

zhaozhongch, 2020-05-19:
I might better to open a new question about the current problem and close this one

closed 2020-05-19 by zhaozhongch

zhaozhongch, 2020-05-25:
For the error 
```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: gflags_shared;gflags_nothreads_shared

  Targets not yet defined: gflags_static;gflags_nothreads_static

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-config.cmake:10 (include)
  /usr/local/lib/cmake/Ceres/FindGflags.cmake:327 (find_package)
  /usr/local/lib/cmake/Ceres/CeresConfig.cmake:308 (find_package)
  /home/zhaozhong/test/vio_kimera_ws/build/opencv3_catkin/opencv3_contrib_src/modules/sfm/CMakeLists.txt:8 (find_package)
```
The reason I found is that the opencv tries to find the ceres-solver's (I installed it for other packages before) gflags. I uninstall ceres-solver and the package got compiled