MIT-SPARK/Kimera-VIO-ROS #77: QueueSynchronizer error when Kimera runs slowly

violetteavi, 2020-05-15:
**Description:**
I've occasionally been experiencing QueueSynchronizer errors, and I think I've found an explanation.

Kimera has an implicit processing framerate determined by input resolution, algorithmic parameters, and available processing power. We operate under the assumption that the user will tune the pipeline parameters/input framerate/etc to achieve real time performance. This is reasonable. However, when the input framerate is significantly faster than the processing framerate, QueueSynchronizer fails a timestamp check. I suspect there is a race condition somewhere in the code.

This is an issue because Kimera's implicit framerate can drop dramatically if the processor is running other programs that restrict Kimera's processing power in unpredictable ways, leading to hard-to-debug errors. Fortunately, I've found that speeding up the Euroc rosbag can replicate a processor crunch. Here are the commands I used and the error message I recieved:

**Command:**
```
roscore
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
rosbag play --clock -r 50 PATH/TO/V1_01_easy.bag
```

**Console output:**
```
F0515 18:49:47.979746   739 QueueSynchronizer.h:138] Check failed: timestamp == payload_timestamp (1403715286912143000 vs. 1403715286962143000) Syncing queue data_provider_right_frame_queue in module Data Provider failed;
 Could not retrieve exact timestamp requested: 
 - Requested timestamp: 1403715286912143000
 - Actual timestamp:    1403715286962143000
*** Check failure stack trace: ***
    @     0x7f19603bdb0d  google::LogMessage::Fail()
    @     0x7f19603bf9b1  google::LogMessage::SendToLog()
    @     0x7f19603bd63d  google::LogMessage::Flush()
    @     0x7f19603c0389  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f1958423242  VIO::SimpleQueueSynchronizer<>::syncQueue()
    @     0x7f195841c8d4  VIO::DataProviderModule::getInputPacket()
    @     0x7f195842762d  VIO::PipelineModule<>::spin()
    @     0x7f195fb7d39b  VIO::Pipeline::spin()
    @     0x7f195fb7bea3  _ZNSt17_Function_handlerIFSt10unique_ptrINSt13__future_base12_Result_baseENS2_8_DeleterEEvENS1_12_Task_setterIS0_INS1_7_ResultIbEES3_ENSt6thread8_InvokerISt5tupleIJMN3VIO8PipelineEFbvEPSE_EEEEbEEE9_M_invokeERKSt9_Any_data
    @     0x7f195fb7bf99  std::__future_base::_State_baseV2::_M_do_set()
    @     0x7f195e734827  __pthread_once_slow
    @     0x7f195fb7e7f8  _ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZNSt13__future_base17_Async_state_implINS1_IS2_IJMN3VIO8PipelineEFbvEPS6_EEEEbEC4EOSB_EUlvE_EEEEE6_M_runEv
    @     0x7f195f00a6df  (unknown)
    @     0x7f195e72c6db  start_thread
    @     0x7f195ea6588f  clone
```

Please give also the following information:
* SparkVio branch, tag or commit used: master
* SparkVioRos branch, tag or commit used: master
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): no

violetteavi, 2020-05-16:
Here are some additional insights gained after I played around with it for a bit:
1. If your input framerate is 1-4x the processing framerate, it does not fail checks but lags behind the input stream. Thanks to the asynchronous callbacks from this PR, Kimera processes all of the inputs anyway.
2. If your input framerate is 4-10x the processing framerate, it sometimes throws QueueSynchronizer errors, usually after processing many frames.
3. If your input framerate is >10x the processing framerate, it always throws a QueueSynchronizer error within the first few frames.
4. You can force Kimera to its maximum processing framerate by running your dataset at a rate so fast that Kimera lags behind (the 1-4x range). For the Euroc rosbag on my computer, I used `-r 10`, or 200 fps input framerate. Kimera processed the 3000 Euroc frames in 30 seconds for a processing framerate of 100fps. 

Note that you could do 4) with the RosBagDataProvider, but that cuts out the ros interface time and will give you an artificially fast processing framerate.

violetteavi, 2020-05-16:
For users that are getting a QueueSynchronizer error, I recommend recording a rosbag of sample input about 20 seconds long, then playing it back with `-r 0.5` or `-r 0.1`.

hellovuong, 2020-10-28:
Dear @violetteavi Andrew,  
Currently, I tried to run Kitti datasets on Kimera-VIO and I've also been experiencing QueueSynchronizer errors:
 `QueueSynchronizer.h:138] Check failed: timestamp == payload_timestamp (1317028415425823926 vs. 1317028415425827980)`
I've already played rosbag with -r 0.5 or even slower but It still isn't working. 
Could you spare some time and take a look at my problem. I would be really appreciated that.
Thank you, 
Here is my launch file:[Launchfile](https://drive.google.com/file/d/14KMfzkaG4E9IeoeOFBOYYmIIorcKOY4L/view?usp=sharing)

kcaisfather, 2021-02-04:
Hi @violetteavi,

I was trying to produce dataset which is coming from AirSim Simulator.  QueueSynchronizer error is also occured, ı tried to do all your suggestion to deal with this error but i didnt. Can i send you a bagfile to help me to find where my problem is?  Topic names of our dataset is the same with euroc dataset. 
**terminal**
``
rosbag play --clock -r 0.5 /path/kca_2021-02-01-18-39-14.bag 
``

**Error**

```
F0204 17:14:26.461493 17694 QueueSynchronizer.h:138] Check failed: timestamp == payload_timestamp (1612193957782609940 vs. 1612193957806610346) Syncing queue data_provider_right_frame_queue in module Data Provider failed;
 Could not retrieve exact timestamp requested: 
 - Requested timestamp: 1612193957782609940
 - Actual timestamp:    1612193957806610346
 ```

There is a link to reach my bag files. If you can help me, i will be appreciated to you.
https://drive.google.com/drive/folders/1oXKJmLtNPGqy6PyDNiGgLU5nV98dsksM?usp=sharing