MIT-SPARK/Kimera-VIO-ROS #83: Pose estimation without IMU data

fatrybl, 2020-05-26:
Dear developers,
I`m doing research with the help of Kimera-VIO-ROS project. 
How can i properly switch off IMU data from pose estimation ? So i want to get pose estimations only from image data, is that possible ?
I need it because i`m planning to use multi IMU system and i will implement my own topic that will post IMU data to other Kimera topics for pose_estimation

ToniRV, 2020-05-26:
Hi @fatrybl, the IMU is required right now: it has its own frontend: ImuFrontend (which could be disabled), but it is tightly coupled in the backend (one could avoid adding IMU factors in the backend but to make the pipeline functional you would need to do a major refactoring).

fatrybl, 2020-05-26:
Thank you for the reply,
but I am a bit confused. As far as i understand, i can switch off IMU with its own frontend: ImuFrontend. But it might be hard to make my own IMU pipeline to feed your model with custom IMU data, am i right? 
And what about pose estimation based on only image data? will it be enough if i just switch off IMu with IMUfrontend or should i do anything else to perform my experiment ?

ToniRV, 2020-05-26:
Sorry for the confusion! What I meant is that:
1) You can indeed use your own IMU frontend without problem. So custom IMU data is definitely possible.
2) You **cannot** _only use image data without IMU_, since the backend requires IMU factors. Disabling the IMU frontend (without replacing it by another IMU frontend) will simply not send IMU factors to the backend, and the optimization will just not take place.

fatrybl, 2020-06-12:
Thank you for clear explanation
And is it possible to add wheel odometry data to VIO package for better localization? or it will be absolutely new algorithm and different back-end ?