MIT-SPARK/Kimera-VIO-ROS #85: can not build Kimera VIO

fatrybl, 2020-06-12:
**Description:**
Dear developers,
I have a problem with building. I followed instructions, but get theses errors (links to screeshots are provided below)

I tried to compile with catkin build kimera_vio_ros -j 4
It did not help.
Any ideas how to fix it ?

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```
                 from /home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp: In member function ‘virtual void VIO::LoopClosureDetectorLoggerFixture_logOptimizedTraj_Test::TestBody()’:
/home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:410:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/mark/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mark/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:412:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/mark/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mark/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:414:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’
       gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
                               ^
In file included from /home/mark/catkin_ws/devel/include/gtsam/geometry/Pose3.h:24:0,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CalibratedCamera.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam/geometry/CameraSet.h:22,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/RegularImplicitSchurFactor.h:10,
                 from /home/mark/catkin_ws/devel/include/gtsam/slam/SmartFactorBase.h:25,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionFactor.h:23,
                 from /home/mark/catkin_ws/devel/include/gtsam_unstable/slam/SmartStereoProjectionPoseFactor.h:23,
                 from /home/mark/catkin_ws/src/Kimera-VIO/include/kimera-vio/backend/VioBackEnd-definitions.h:26,
                 from /home/mark/catkin_ws/src/Kimera-VIO/tests/testLogger.cpp:28:
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note: candidate: static gtsam::Rot3 gtsam::Rot3::Random(boost::random::mt19937&)
     static Rot3 Random(boost::mt19937 & rng);
                 ^~~~~~
/home/mark/catkin_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
make[2]: *** [CMakeFiles/testKimeraVIO.dir/tests/testLogger.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
cd /home/mark/catkin_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...................................................................................................
Failed     << kimera_vio:make                 [ Exited with code 2 ]                               
Failed    <<< kimera_vio                      [ 8 minutes and 49.7 seconds ]                       
Abandoned <<< kimera_vio_ros                  [ Unrelated job failed ]                             
[build] Summary: 7 of 9 packages succeeded.                                                        
[build]   Ignored:   7 packages were skipped or are blacklisted.                                   
[build]   Warnings:  5 packages succeeded with warnings.                                           
[build]   Abandoned: 1 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                            
[build] Runtime: 25 minutes and 7.4 seconds total.  

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.
https://sun9-25.userapi.com/c857332/v857332646/143666/Vrt7VfdeIjM.jpg
https://sun9-69.userapi.com/c857332/v857332646/143670/SQCCAy149rQ.jpg

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: 3.2.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04 with last updates
* Did you change the source code? (yes / no):

tftangming, 2020-06-13:
Have you sloved the issue? I also meet the same problem. that is so bad...

fatrybl, 2020-06-13:
> 
> 
> Have you sloved the issue? I also meet the same problem. that is so bad...

Unfortunately, no. Several weeks ago i managed to build the whole project and this problem did not occur. Probably developers have updated some libs and now it comes

tftangming, 2020-06-13:
try to cd the KImera-VIO and git reset to previous version???

zhaozhongch, 2020-06-13:
Before build, try cd to gtsam folder. then
```
git checkout develop
```
then go back to workspace/src
```
catkin build
```
See if it works

fatrybl, 2020-06-13:
> 
> 
> Before build, try cd to gtsam folder. then
> 
> ```
> git checkout develop
> ```
> 
> then go back to workspace/src
> 
> ```
> catkin build
> ```
> 
> See if it works

Yes, i have checked it. Still not working

zhaozhongch, 2020-06-13:
Another thing you can try is to disable the test when compiling
```
catkin build -DBUILD_TESTS=OFF
```

tftangming, 2020-06-14:
I have success to build the gtsam by "git checkout develop; catkin build",;
but when I build the opencv3, something error....
`Starting  >>> glog_catkin                                                                           
Finished  <<< glog_catkin                                  [ 0.2 seconds ]                          
____________________________________________________________________________________________________
Errors     << opencv3_catkin:make /home/tangming/catkin_ws/logs/opencv3_catkin/build.make.008.log   
make[3]: 警告: 子 make 中强制 -jN: 关闭 jobserver 模式。
CMake Warning at cmake/OpenCVFindLAPACK.cmake:29 (message):
  LAPACK(OpenBLAS): CBLAS/LAPACK headers are not found in
  '/usr/include/openblas'
Call Stack (most recent call first):
  cmake/OpenCVFindLAPACK.cmake:97 (ocv_lapack_check)
  CMakeLists.txt:635 (include)

-- Found Ceres compiled with miniglog substitute for glog, beware this will likely cause problems if glog is later linked.
CMake Warning at /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_src/cmake/OpenCVDownload.cmake:190 (message):
  xfeatures2d/boostdesc: Download failed: 28;"Timeout was reached"

  For details please refer to the download log file:

  
  /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_build/CMakeDownloadLog.txt

Call Stack (most recent call first):
  /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_contrib_src/modules/xfeatures2d/cmake/download_boostdesc.cmake:22 (ocv_download)
  /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_contrib_src/modules/xfeatures2d/CMakeLists.txt:8 (download_boost_descriptors)

In file included from /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_src/modules/stitching/include/opencv2/stitching.hpp:49:0,
                 from /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_build/modules/python_bindings_generator/pyopencv_generated_include.h:65,
                 from /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_src/modules/python/src2/cv2.cpp:30:
/home/tangming/catkin_ws/build/opencv3_catkin/opencv3_src/modules/stitching/include/opencv2/stitching/detail/matchers.hpp:52:42: fatal error: opencv2/xfeatures2d/cuda.hpp: 没有那个文件或目录
compilation terminated.
make[5]: *** [modules/python2/CMakeFiles/opencv_python2.dir/__/src2/cv2.cpp.o] Error 1
make[4]: *** [modules/python2/CMakeFiles/opencv_python2.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [opencv3_src-prefix/src/opencv3_src-stamp/opencv3_src-build] Error 2
make[1]: *** [CMakeFiles/opencv3_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/tangming/catkin_ws/build/opencv3_catkin; catkin build --get-env opencv3_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
....................................................................................................
Failed     << opencv3_catkin:make                          [ Exited with code 2 ]                   
Failed    <<< opencv3_catkin                               [ 1 minute and 52.7 seconds ]            
Abandoned <<< kimera_rpgo                                  [ Unrelated job failed ]                 
Abandoned <<< kimera_vio                                   [ Unrelated job failed ]                 
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                 
Finished  <<< gtsam                                        [ 4 minutes and 25.1 seconds ] `

I try to modify the CMakeLists.txt in the /home/tangming/catkin_ws/build/opencv3_catkin/opencv3_src/modules/stitching by adding "INCLUDE_DIRECTORIES("/home/tangming/catkin_ws/build/opencv3_catkin/opencv3_contrib_src/modules/xfeatures2d/include")", but the problem sitll exist.

zhaozhongch, 2020-06-14:
I met similar problem before.
The ceres solver will lead the GLOG to other location.
Try uninstall cares solver. Go to your ceres-solver/build
Then
```
sudo make uninstall
```
Then rebuild Kimera

tftangming, 2020-06-14:
~/opencv_contrib/modules/xfeatures2d/src/boostdesc.cpp:673:20: fatal error: boostdesc_bgm.i: No such file or directory
I still meet some problem.. I have tried to downloads these files such as "
boostdesc_bgm.i
boostdesc_bgm_bi.i
boostdesc_bgm_hd.i
boostdesc_lbgm.i
boostdesc_binboost_064.i
boostdesc_binboost_128.i
boostdesc_binboost_256.i
vgg_generated_120.i
vgg_generated_64.i
vgg_generated_80.i
vgg_generated_48.i
And put them into the folder "/home/tangming/catkin_ws/build/opencv3_catkin/opencv3_build/downloads". What it happend???

fatrybl, 2020-06-14:
> Another thing you can try is to disable the test when compiling
> 
> ```
> catkin build -DBUILD_TESTS=OFF
> ```

Managed to build successfully after deleting everything and repeating whole process once again with develop branch. Thanks much! hope developers will add all changes to origin branch soon

fatrybl, 2020-06-19:
> 
> 
> I met similar problem before.
> The ceres solver will lead the GLOG to other location.
> Try uninstall cares solver. Go to your ceres-solver/build
> Then
> 
> ```
> sudo make uninstall
> ```
> 
> Then rebuild Kimera

I``ve faced the problem with no ability to see triangle mesh in Rviz while launching kimera-VIO-ros on Euroc dataset. So i follow the instruction but in Rviz i see only TFs and raw picture from both camers. Yes I`ve installed special plugins for rviz. So i have Mesh and IMU plugins as add-ons in rviz. Using rostopic echo, i`ve found out that mesh topic does not post anything. 
Any ideas why i happens ?

veronicamuriga, 2020-06-29:
I'm pretty sure step 3 is what worked for me, but what I did in order was:
1. [Uninstall gtsam](https://answers.ros.org/question/233493/how-to-uninstall-gtsam-library-c/) then re-install and[ build from source](https://github.com/borglab/gtsam/blob/develop/INSTALL.md) 
2. Make sure I was working on 'develop' branch (git checkout develop)
3. First build kimera-vio (catkin build kimera_vio) then build kimera_vio_ros (catkin build kimera_vio_ros)

calmelo, 2020-07-02:
/home/wangxiaoyuan/catkin_ws/src/gtsam/gtsam/geometry/Quaternion.h:85:22: error: ‘omega’ was not declared in this scope
     _Scalar theta2 = omega.dot(omega);
How can I solve the problem?

veronicamuriga, 2020-07-02:
@calmelo this seems to be a separate issue from the one in this thread and as such, I would recommend opening a new issue as you may be assisted more promptly then- that being said, the scope errors that I encounter usually happen when I forget to source my devel/setup.bash file of the catkin workspace I'm working in.

calmelo, 2021-01-31:
> ~/opencv_contrib/modules/xfeatures2d/src/boostdesc.cpp:673:20: fatal error: boostdesc_bgm.i: No such file or directory
> I still meet some problem.. I have tried to downloads these files such as " [...]

Have you solve the problem? I meet the problem too.

haoyuxing1998, 2021-06-06:
@zhaozhongch 
I fail to compile gtsam when catkin build, other success, did you meet before?
 src/gtsam/gtsam/3rdparty/Spectra/Util/SimpleRandom.h:32:20: error: ‘Index’ in namespace ‘Eigen’ does not name a type
     typedef Eigen::Index Index;
                    ^
/home/xhy/catkin_ws/src/gtsam/gtsam/3rdparty/Spectra/Util/SimpleRandom.h:75:29: error: ‘Index’ does not name a type
     Vector random_vec(const Index len)

haoyuxing1998, 2021-06-06:
@zhaozhongch
I fail to compile gtsam when catkin build, other success, did you meet before?

> src/gtsam/gtsam/3rdparty/Spectra/Util/SimpleRandom.h:32:20: error: ‘Index’ in namespace ‘Eigen’ does not name a type
typedef Eigen::Index Index;
^
/home/xhy/catkin_ws/src/gtsam/gtsam/3rdparty/Spectra/Util/SimpleRandom.h:75:29: error: ‘Index’ does not name a type
Vector random_vec(const Index len)