MIT-SPARK/Kimera-VIO-ROS #88: Please, update your manual

fatrybl, 2020-06-18:
**Description:**
Dear developers, 
Could you please update some moments in Instruction ?
for example:
1) Download a Euroc rosbag: for example V1_01_easy. - The link to datasets is not valid. i get 403 ERROR permissions denied. So i cant download any dataset from https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets this website. Maybe, website owners have closed or moved their content.
2) roslaunch kimera_vio_ros kimera_vio_ros_kitti.launch
in this case i do not have any .launch file for kitty dataset. I have only
 cam_info_yamlizer.launch            kimera_vio_ros.launch
gt_logger.launch                    kimera_vio_ros_realsense_IR.launch
kimera_vio_ros_euroc.launch         kimera_vio_ros_tesse.launch
. Maybe, It does not exists on develop branch.
But i need launch file for KITTY very much
can you please provide me with it ?

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

closed 2020-06-19 by fatrybl

YiChenCityU, 2020-07-23:
Hi, have you found the kitti launch file?

p-akanksha, 2020-09-27:
I too am unable to locate the kitti launch file (kimera_vio_ros_kitti.launch).

ToniRV, 2020-09-28:
Hi all, sorry for the confusion, but we removed Kitti support because the IMU data is too sparse to be usable.

PR #107 (2020-09-28): Fix/remove kitti support
Fix #88 : deprecate kitti support