MIT-SPARK/Kimera-VIO-ROS #90: No 3d triangle mesh provided

fatrybl, 2020-06-26:
**Description:**
Dear developers, could you please update your tutorial with Euroc dataset in order to launch full pipeline of Kimera. I cant see any triangular mesh in rviz. 3D mesh textured library is installed according to tutorial. But the topics dont publish anything. If you follow your tutorial from scratch, you will find out that is only possible to see raw camera image and tracked features in RVIZ.  But not a mesh.
 And could you also provide more complete tutorial for launching Kimera-VIO-ROS for different datasets. Your project is very interesting, i would like to test it on different datasets but its not well-described how to launch Kimera on different sets. I`m not a pro in ROS but interested in your project very much
Thank you in advance !

**Command:**
```
# replace this line with the command(s) you used
```

**Console output:**
```

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

ToniRV, 2020-06-26:
Hi @fatrybl!
Thank you for your comments.
Sorry I was busy with conferences and deadlines, I’ll try to fix that issue this weekend and update the documentation.
Thanks!

fatrybl, 2020-06-29:
> Hi @fatrybl!
> Thank you for your comments.
> Sorry I was busy with conferences and deadlines, I’ll try to fix that issue this weekend and update the documentation.
> Thanks!

Sounds great, thanks!
And I`ve found out one more issue with [**gt_logger_node.py**](https://sun9-56.userapi.com/c857632/v857632624/213fb9/6VwEkDnb7XY.jpg)
i get this error when set  **<arg name="log_gt_data" default="true"/> to be TRUE** in **kimera_vio_ros_euroc.launch**.
so i tried to modify the code in this way:
```
        with open(self.output_csv_file, mode='w',) as file:
            fieldnames = ['timestamp[ns]', 'x', 'y', 'z', 'qw','qx','qy','qz']
            writer = csv.DictWriter(file, delimiter=",",  fieldnames=fieldnames)
```
but get 
this [errors](https://sun9-36.userapi.com/c857632/v857632624/213fc3/ReST6zln1bg.jpg) while playing dataset
whereas kimera_vio_ros_euroc.launch does not produce any errors
maybe I am doing something incorrect ?
P.s.
V1_01_easy.bag dataset has been used

betaBison, 2020-06-29:
I believe this is a duplicate to #78

fatrybl, 2020-06-29:
> 
> 
> I believe this is a duplicate to #78

Exactly, thank you!

fatrybl, 2020-07-02:
Additionally, I`ve managed to launch mesh reconstruction via Kimera-VIO without ROS. But it switches off  accidentally.
 Just see these 2 videos for clear understanding.
1) https://yadi.sk/i/OVAO5QeXOl25pw
2) https://yadi.sk/i/wyYdIf_s7mtP_g
with V1_01 and MH_05 datasets.
Any ideas?

bwangbb, 2020-09-12:
@fatrybl Hi, have you solved this problem? I got the same issue. Do I need to switch back to old version? Thanks