MIT-SPARK/Kimera-VIO-ROS #93: Kimera_vio_ros_node-1: Process has died in the initailizing-pose-graph-visualizer phase

Haonan-DONG, 2020-07-06:
**Description:**
Thanks for the excellent work. I finally made it on building the source code and enter into the ros phase. But here I got several error that I could not understand.

**Command:**
```
roscore
 roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch online:=false rosbag_path:="../../repo/V1_01_easy.bag"
```

**Console output:**
```
... logging to /home/dong/.ros/log/7351ce7a-bfd7-11ea-98e9-84ef18c5edec/roslaunch-dong-workstation-24700.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dong-workstation:37621/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/ground_truth_odometry_rosbag_topic: 
 * /kimera_vio_ros/kimera_vio_ros_node/imu_rosbag_topic: /imu0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_rosbag_topic: /cam0/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: False
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/dong/projec...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_rosbag_topic: /cam1/image_raw
 * /kimera_vio_ros/kimera_vio_ros_node/rosbag_path: ../../repo/V1_01_...
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [24719]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [24720]
[ INFO] [1594075026.120129748]: Initializing pose graph visualizer
[kimera_vio_ros/kimera_vio_ros_node-1] process has died [pid 24719, exit code -11, cmd /home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/devel/lib/kimera_vio_ros/kimera_vio_ros_node --use_lcd=false --vocabulary_path=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO/vocabulary/ORBvoc.yml --flagfile=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags --flagfile=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO/params/Euroc/flags/VioBackEnd.flags --flagfile=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO/params/Euroc/flags/RegularVioBackEnd.flags --flagfile=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Visualizer3D.flags --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --log_output=false --log_euroc_gt_data=false --output_path=/home/dong/project/cvrs_project/SLAM/slam_application/catkin_ws/src/Kimera-VIO-ROS/output_logs/ --viz_type=0 --visualize=true reinit_flag:=reinit_flag reinit_pose:=reinit_pose odometry:=odometry resiliency:=resiliency imu_bias:=imu_bias optimized_trajectory:=optimized_trajectory pose_graph:=pose_graph mesh:=mesh frontend_stats:=frontend_stats debug_mesh_img/image_raw:=debug_mesh_img/image_raw feature_tracks/image_raw:=feature_tracks/image_raw time_horizon_pointcloud:=time_horizon_pointcloud __name:=kimera_vio_ros_node __log:=/home/dong/.ros/log/7351ce7a-bfd7-11ea-98e9-84ef18c5edec/kimera_vio_ros-kimera_vio_ros_node-1.log].
log file: /home/dong/.ros/log/7351ce7a-bfd7-11ea-98e9-84ef18c5edec/kimera_vio_ros-kimera_vio_ros_node-1*.log

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used
* GTSAM version used: Latest ? 
* OpenGV version used: Latest ? (How to find the version out?)
* OpenCV version used: type `opencv_version` 3.3.1
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 16.04 
* Did you change the source code? (yes / no): No

Since I am new to ros and could anyone give me a hint?
It might be the mis-installation of some of the module.

Thanks a lot.

closed 2020-11-19 by Haonan-DONG

RozDavid, 2020-12-18:
Hey @Haonan-DONG, 

I've been struggling with the same problem. Could you help how you solved your issue? 

Thanks a lot and to @ToniRV for this amazing work!

RozDavid, 2020-12-20:
Hey Guys,

So if anybody ends up here again, I was able to solve our problem. Basically it come down to a really basic and rookie mistake, that my global opencv4 installation got linked with the catkin workspace somehow. 
Also, if you would like to check the backtrace of your ros node, I just realized it is possible to export the core dump from a ROS node and inspect it with gdb. The way to do so:

- `ulimit -c unlimited`
- Add `launch-prefix="xterm -e gdb --args"` to your failing node in the roslaunch file
- Roslaunch your desired failing node, here `roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch`
- And open the core dump with: `gdb devel/lib/kimera_vio_ros/kimera_vio_ros_node <YOUR_HOME_FOLDER>/.ros/core`
- Inspect backtrace with `bt` command

Hope this will help somebody in the future, good luck everybody

X-NEARS, 2022-05-05:
I still want to know how to solve this "exit code -11". 
Can anyone help me.Thanks a lot