MIT-SPARK/Kimera-VIO-ROS #95: CMake Error during catkin build

srujanpanuganti, 2020-07-26:
**Description:**

**Command:**
```
cd kimera_ws
source devel/setup.bash
catkin build
```

**Console output:**
```

Starting  >>> mesh_rviz_plugins                                                                                                                                                                            
Starting  >>> opencv3_catkin                                                                                                                                                                               
Starting  >>> opengv_catkin                                                                                                                                                                                
Starting  >>> pose_graph_tools                                                                                                                                                                             
Finished  <<< cmake_external_project_catkin                [ 0.8 seconds ]                                                                                                                                 
Finished  <<< disparity_image_proc                         [ 2.8 seconds ]                                                                                                                                 
Finished  <<< gflags_catkin                                [ 0.9 seconds ]                                                                                                                                 
Finished  <<< opengv_catkin                                [ 1.3 seconds ]                                                                                                                                 
Finished  <<< dbow2_catkin                                 [ 0.7 seconds ]                                                                                                                                 
Finished  <<< pose_graph_tools                             [ 4.2 seconds ]                                                                                                                                 
Finished  <<< kimera_rviz_markers                          [ 4.7 seconds ]                                                                                                                                 
Finished  <<< image_undistort                              [ 8.3 seconds ]                                                                                                                                 
Finished  <<< mesh_rviz_plugins                            [ 7.9 seconds ]                                                                                                                                 
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << opencv3_catkin:make /home/srujan/kimera_ws/logs/opencv3_catkin/build.make.007.log                                                                                                            
CMake Warning at /home/srujan/kimera_ws/build/opencv3_catkin/opencv3_contrib_src/modules/matlab/CMakeLists.txt:54 (message):
  A required dependency of the matlab module (Python Jinja2 package) was not
  found (installation command: "pip install jinja2").  Disabling Matlab
  bindings...

CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: gflags_shared;gflags_nothreads_shared

  Targets not yet defined: gflags_static;gflags_nothreads_static

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-config.cmake:10 (include)
  /usr/lib/cmake/Ceres/FindGflags.cmake:327 (find_package)
  /usr/lib/cmake/Ceres/CeresConfig.cmake:308 (find_package)
  /home/srujan/kimera_ws/build/opencv3_catkin/opencv3_contrib_src/modules/sfm/CMakeLists.txt:8 (find_package)

make[2]: *** [opencv3_src-prefix/src/opencv3_src-stamp/opencv3_src-configure] Error 1
make[1]: *** [CMakeFiles/opencv3_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/srujan/kimera_ws/build/opencv3_catkin; catkin build --get-env opencv3_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << opencv3_catkin:make                          [ Exited with code 2 ]                                                                                                                          
Failed    <<< opencv3_catkin                               [ 24.2 seconds ]                                                                                                                                
Abandoned <<< glog_catkin                                  [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_rpgo                                  [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_vio                                   [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                                                                                                                        
^C[build] Interrupted by user!                                                                                                                                                                               
[build] Summary: 10 of 16 packages succeeded.                                                                                                                                                              
[build]   Ignored:   None.                                                                                                                                                                                 
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 5 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 33.4 seconds total.
```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used : Don't know what this is
* GTSAM version used: development branch 
* OpenGV version used: Latest avalable
* OpenCV version used: type `3.2.0`
* Operating system and version : Ubuntu 18.04, ROS-melodic 
* Did you change the source code? (yes / no): No

srujanpanuganti, 2020-07-29:
Hi, I kind of fixed building gtsam and glog_catkin, but the opencv3_catkin still doesn't build. Here the `catkin build` output

Console Output

```
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << opencv3_catkin:make /home/srujan/kimera_ws/logs/opencv3_catkin/build.make.011.log                                                                                                            
CMake Warning at /home/srujan/kimera_ws/build/opencv3_catkin/opencv3_contrib_src/modules/matlab/CMakeLists.txt:54 (message):
  A required dependency of the matlab module (Python Jinja2 package) was not
  found (installation command: "pip install jinja2").  Disabling Matlab
  bindings...

CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-targets.cmake:37 (message):
  Some (but not all) targets in this export set were already defined.

  Targets Defined: gflags_shared;gflags_nothreads_shared

  Targets not yet defined: gflags_static;gflags_nothreads_static

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gflags/gflags-config.cmake:10 (include)
  /usr/lib/cmake/Ceres/FindGflags.cmake:327 (find_package)
  /usr/lib/cmake/Ceres/CeresConfig.cmake:308 (find_package)
  /home/srujan/kimera_ws/build/opencv3_catkin/opencv3_contrib_src/modules/sfm/CMakeLists.txt:8 (find_package)

make[2]: *** [opencv3_src-prefix/src/opencv3_src-stamp/opencv3_src-configure] Error 1
make[1]: *** [CMakeFiles/opencv3_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/srujan/kimera_ws/build/opencv3_catkin; catkin build --get-env opencv3_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << opencv3_catkin:make                          [ Exited with code 2 ]                                                                                                                          
Failed    <<< opencv3_catkin                               [ 2.6 seconds ]                                                                                                                                 
Abandoned <<< kimera_rpgo                                  [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_vio                                   [ Unrelated job failed ]                                                                                                                        
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]                                                                                                                        
Finished  <<< gtsam                                        [ 4.1 seconds ]                                                                                                                                 
Finished  <<< glog_catkin                                  [ 0.1 seconds ]                                                                                                                                 
[build] Summary: 12 of 16 packages succeeded.                                                                                                                                                              
[build]   Ignored:   None.                                                                                                                                                                                 
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 3 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 5.2 seconds total.    

```

zinuok, 2020-09-07:
did you solve this problem? I have exactly same problem with you

zinuok, 2020-09-07:
> did you solve this problem? I have exactly same problem with you

I found some solutions. If you uninstall glogs & gflags already installed by apt install command, it works
refer this: https://github.com/alibaba/euler/issues/158#issuecomment-585010067

iamsavva, 2020-10-24:
> > did you solve this problem? I have exactly same problem with you
> 
> I found some solutions. If you uninstall glogs & gflags already installed by apt install command, it works
> refer this: [alibaba/euler#158 (comment)](https://github.com/alibaba/euler/issues/158#issuecomment-585010067)

hey, could you provide the exact lines you were using, please? having the same issue here.

iamsavva, 2020-10-24:
Hey, in my case removing libgflags libraries helped:
`sudo apt-get remove libgflags*`

Found it in [this discussion](https://github.com/ethz-asl/eth_supermegabot/issues/28).