MIT-SPARK/Kimera-VIO-ROS #96: /Kimera-VIO/tests/testLogger.cpp:414:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)

FaboNo, 2020-07-28:
Hi there

First of all thank you for sharing such amazing piece of work!

I am running Ubuntu 18.04 and ROS melodic.

I followed the instructions (actually I took the HTTP way instead of the SSH way) and every thing work well until I tried to compile **kimera-VIO** and **Kimera-VIO-ROS** 

Regarding Kilera-VIO,I had the following error:

```
/home/fabrice/catkin_kimera_ws/devel/include/gtsam/geometry/Rot3.h:124:17: note:   no known conversion for argument 1 from ‘std::mt19937 {aka std::mersenne_twister_engine<long unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>}’ to ‘boost::random::mt19937& {aka boost::random::mersenne_twister_engine<unsigned int, 32, 624, 397, 31, 2567483615, 11, 4294967295, 7, 2636928640, 15, 4022730752, 18, 1812433253>&}’
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO/tests/testLogger.cpp:414:31: error: no matching function for call to ‘gtsam::Rot3::Random(std::mt19937&)’ gtsam::Rot3::Random(rng_), gtsam::Point3::Random());
```
looks like I am the first one to face such error. I commented the testLogger.cpp in the CMakeLists.txt and managed to compile Kimera-VIO successfully.

However with Kimera-VIO-ROS, I have the following error;

```
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp: In function ‘void VIO::utils::msgCamInfoToCameraParams(const CameraInfoConstPtr&, const string&, const string&, VIO::CameraParams*)’:
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:62:45: error: cannot convert ‘const _distortion_model_type {aka const std::__cxx11::basic_string<char>}’ to ‘VIO::DistortionModel’ in assignment
   cam_params->distortion_model_ = cam_info->distortion_model;
                                             ^~~~~~~~~~~~~~~~
In file included from /home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/include/kimera_vio_ros/utils/UtilsRos.h:11:0,
                 from /home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:7:
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:63:39: error: no match for ‘operator==’ (operand types are ‘VIO::DistortionModel’ and ‘const char [7]’)
   CHECK(cam_params->distortion_model_ == "radtan" ||
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:64:39: error: no match for ‘operator==’ (operand types are ‘VIO::DistortionModel’ and ‘const char [18]’)
         cam_params->distortion_model_ == "radial-tangential" ||
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:65:39: error: no match for ‘operator==’ (operand types are ‘VIO::DistortionModel’ and ‘const char [12]’)
         cam_params->distortion_model_ == "equidistant");
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
/home/fabrice/catkin_kimera_ws/src/Kimera-VIO-ROS/src/utils/UtilsRos.cpp:70:56: error: no matching function for call to ‘VIO::CameraParams::convertDistortionVectorToMatrix(const std::vector<double>&, std::vector<double>*)’
       distortion_coeffs, &cam_params->distortion_coeff_);
```
and here I do not know how to handle it?

I have also a question regarding open-cv, as it is include in ROS melodic, why do you we need to recompile it in the catkin workspace? Indeed I got several warning during the compilation due to the existence of two version of opencv.

Thank you for your help

Best regards

FaboNo, 2020-07-28:
Here is an update:

I found some issues in utilsRos.cpp

`cam_params->distortion_model_ = cam_info->distortion_model;`

There is an error because  one wants to assign a string to an enum. However when I looked at In sensor_msgs/distortion__models.h there are only two models: "plump_bob" and "rational_polynomial". It means that nothing is matching "radtan", "radial-tangential" or "equidistant" and no correspondence neither with enum DistortionModel in  src/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h 

As the assigment is done to check, i put the four lines in comments.

I had another error with:
`VIO::CameraParams::convertDistortionVectorToMatrix(distortion_coeffs, &cam_params->distortion_coeff_);`

The method expects a cv::mat as a second argument, so I replaced by:
`VIO::CameraParams::convertDistortionVectorToMatrix(distortion_coeffs, &cam_params->distortion_coeff_mat_);
`
I aw able to compile kimera-vio-ros sussceefully. Let see if it works now:)

robot-mh, 2020-07-29:
I come accross the same issues. @violetteavi @clungzta @m-pilia @gavanderhoorn

gavanderhoorn, 2020-07-29:
@robot-mh: I'm not entirely sure why you @-mentioned me?

khumanhan, 2020-07-30:
Hello @FaboNo , is it work? if not, can you find out another solution? Yes, I'm facing this issue too!

FaboNo, 2020-07-30:
@khumanhan for me kimera-VIO is working at least with the rosbag provided for tests - I am working on plugin my ZED2 for real tests now.
To resume:
- I commented the testLogger.cpp in the CMakeLists.txt
- Regarding utilsRos.cpp:
I put these four lines in comments:
```
/* 
cam_params->distortion_model_ = cam_info->distortion_model;
  CHECK(cam_params->distortion_model_ == "radtan" ||
        cam_params->distortion_model_ == "radial-tangential" ||
        cam_params->distortion_model_ == "equidistant");
*/
```

and replace 
`VIO::CameraParams::convertDistortionVectorToMatrix(distortion_coeffs, &cam_params->distortion_coeff_);`
by
`VIO::CameraParams::convertDistortionVectorToMatrix(distortion_coeffs, &cam_params->distortion_coeff_mat_);`

and I am able to compile successfully

khumanhan, 2020-07-30:
@FaboNo thank you for your reply.
I did as your instruction and successfully compiled too, but I'm not sure about the result of test, the trajectory didn't appear after I run with Euroc data set, but only coordinate systems. You can see the following image:
[image]
Do you know why?

FaboNo, 2020-07-30:
@gavanderhoorn  I had the same result - I did not see the trajectory but actually I was not looking for it...
Regarding the display of the meshes, have a look at this issue: https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/78
Last reply from pherbers provides hints to display the meshes (and try to change rviz background as well it worked for me)

khumanhan, 2020-07-31:
@FaboNo Oh I see, thank you again, I will try it now :)

tftangming, 2020-08-11:
@FaboNo  
I met a problem.if i need to update my kimera-vio?
error: ‘class VIO::CameraParams’ has no member named ‘distortion_coeff_mat_’
       distortion_coeffs, &cam_params->distortion_coeff_mat_);

tftangming, 2020-08-11:
@FaboNo Thanks for your solution about the problems, I success to complie the kimera-vio-ros, and I will download the rosbag for test. And I also have a zed2 too, I want to know if you have success to use the camera for real testing? Thank you again!

FaboNo, 2020-08-11:
@tftangming with the ZED2, actually not yet because the missing calibration data is `imu_time_shift` and to get it, I need to do a calibration and thus generate a rosbag. But I just have a Jetson Nano with me, and it is not powerfull enough to run the ConvNets developed for the ZED2   (stereo and object recognition) and to generate a rosbag without dropping images. Jetson Xavier is really required here! I will let you know when I will make progress
Regards

tftangming, 2020-08-12:
@FaboNo I do a try to create zed2.launch according to the realsense.launch and create my custom params file for zed2. About the data `imu_time_shift` , I set it to 0 for a test, however the program crash because the consecutive imu_datas have same timestamps as follow. But I echo the topic "/zed2/zed_node/imu/data_raw" and find it is normal, So I cant know what it happens.
In addition, I think it can be used for testing online after calibration the zed2 camera and modify its params file.

`F0812 15:15:58.458397 32217 ThreadsafeImuBuffer-inl.h:58] Check failed: timestamp_nanoseconds > last_value.timestamp_ (1597216558478557501 vs. 1597216558478557501) Timestamps not strictly increasing.`

tftangming, 2020-08-13:
@FaboNo Now I have finished the zed2 live meshing using the kimera-vio-ros, and the above problem about timestamps is skipped forcely by myself. But the reconstruction result is not good...I will try to adjust the camera params and imu params.

awenbigprawn, 2021-04-23:
hey, I am suffering the same problem by using the Zed2 dataset, it is the problem of Zed2. Zed2 has too fast 400 Hz imu frequency even higher than the clock, so it is real that some neighbor imu data has the same timestamps. You can check this fact by playing the  rosbag with --pause and then hit "s" to step, meanwhile echo the topic of IMU in another terminal.

soeroesg, 2022-03-16:
The line `CHECK_EQ(distortion_coeffs.size(), 4);` is also problematic because the camera_info can contain more elements. It should say `CHECK_GE(distortion_coeffs.size(), 4);`

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/src/utils/UtilsRos.cpp#L76