MIT-SPARK/Kimera-VIO-ROS #100: [kimera_vio_ros/posegraph_viewer-2] process has died [pid 14460, exit code -6, cmd]

bessszilard, 2020-08-24:
**Description:**

First of all, thank you for this wonderful open-source library. I tried to run the demo file with 
[Euroc](https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets)  / [V1_01_easy](http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/V1_01_easy.bag)

In the simulation, it shows only the point cloud without meshes [1] and not is like in the [demo image](https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/media/Kimera-VIO-ROS_mesh.gif).

My questions:
1. Why do I get that two errors? [2]
2. How can I enable to show the meshes as well?

**Command:**
I used the given commands in the tutorial

1. Run roscore `$ roscore`
2. In another terminal, launch KimeraVIO ROS wrapper: 
```
roslaunch kimera_vio_ros kimera_vio_ros_euroc.launch
```
3. In another terminal, launch rviz for visualization: 
```
rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz
```

I tried to run the 

**Console output:**
I received these 2 errors:

[2] error:
```bash
[kimera_vio_ros/posegraph_viewer-2] process has died [pid 14460, exit code -6, cmd /home/szilard/catkin_ws/devel/lib/pose_graph_tools/visualizer_node graph:=pose_graph __name:=posegraph_viewer __log:=/home/szilard/.ros/log/ec01ca86-e607-11ea-abdf-18dbf240af48/kimera_vio_ros-posegraph_viewer-2.log].
log file: /home/szilard/.ros/log/ec01ca86-e607-11ea-abdf-18dbf240af48/kimera_vio_ros-posegraph_viewer-2*.log
I0824 19:18:54.620903 14459 RosOnlineDataProvider.cpp:56] Requested initialization from ground-truth. Initializing ground-truth odometry one-shot subscriber.
W0824 19:18:54.623056 14459 RosOnlineDataProvider.cpp:65] Waiting for ground-truth pose to initialize VIO on ros topic: /kimera_vio_ros/gt_odom
E0824 19:18:57.721786 14459 RosOnlineDataProvider.cpp:82] Missing ground-truth pose while trying for 3.0008 seconds.
Enabling autoInitialize and continuing without ground-truth pose.
```

The whole console output until the errors:
```

... logging to /home/szilard/.ros/log/ec01ca86-e607-11ea-abdf-18dbf240af48/roslaunch-pop-os-14443.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pop-os:37161/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: cam0
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/szilard/cat...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: cam1
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.7
 * /use_sim_time: True

NODES
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [14459]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [14460]
[ INFO] [1598289520.302719909]: Initializing pose graph visualizer
visualizer_node: /home/szilard/catkin_ws/src/pose_graph_tools/src/visualizer.cpp:11: Visualizer::Visualizer(ros::NodeHandle&): Assertion `nh.getParam("frame_id", frame_id_)' failed.
I0824 19:18:40.463707 14459 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0824 19:18:40.463845 14459 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0824 19:18:40.464617 14459 RosOnlineDataProvider.cpp:45] Waiting for ROS time to be valid... 
(Sim Time is enabled; run rosbag with --clock argument)
[kimera_vio_ros/posegraph_viewer-2] process has died [pid 14460, exit code -6, cmd /home/szilard/catkin_ws/devel/lib/pose_graph_tools/visualizer_node graph:=pose_graph __name:=posegraph_viewer __log:=/home/szilard/.ros/log/ec01ca86-e607-11ea-abdf-18dbf240af48/kimera_vio_ros-posegraph_viewer-2.log].
log file: /home/szilard/.ros/log/ec01ca86-e607-11ea-abdf-18dbf240af48/kimera_vio_ros-posegraph_viewer-2*.log
I0824 19:18:54.620903 14459 RosOnlineDataProvider.cpp:56] Requested initialization from ground-truth. Initializing ground-truth odometry one-shot subscriber.
W0824 19:18:54.623056 14459 RosOnlineDataProvider.cpp:65] Waiting for ground-truth pose to initialize VIO on ros topic: /kimera_vio_ros/gt_odom
E0824 19:18:57.721786 14459 RosOnlineDataProvider.cpp:82] Missing ground-truth pose while trying for 3.0008 seconds.
Enabling autoInitialize and continuing without ground-truth pose.
I0824 19:18:57.735524 14540 Pipeline.h:185] Spinning Kimera-VIO.
I0824 19:18:57.735507 14459 KimeraVioRos.cpp:114] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Display [ms]                            	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
W0824 19:18:58.057679 14540 DataProviderModule.cpp:103] Waiting for IMU data...
I0824 19:18:58.071601 14540 Pipeline.cpp:788] Frontend launched (parallel_run set to 1).
I0824 19:18:58.071628 14540 Pipeline.cpp:601] ------------------- Initialize Pipeline with frame k = 2--------------------
W0824 19:18:58.071638 14540 InitializationFromImu.cpp:26] InitializationFromImu: assumes that the vehicle is stationary and upright along some axis,and gravity vector is along a single axis!
I0824 19:18:58.115443 14540 Pipeline.cpp:310] Before launching threads.
I0824 19:18:58.115618 14540 Pipeline.cpp:817] Backend, mesher and visualizer launched (parallel_run set to 1).
I0824 19:18:58.115628 14540 Pipeline.cpp:312]  launching threads.

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used: `ea63cad3edb4ab8a35509f8417b95e34b29af1ab`
* GTSAM version used: 
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Pop!_OS 18.04 LTS
* Did you change the source code? (yes / no): no

zheng-xiao, 2021-05-21:
Hi,
I have the same issue about "Enabling autoInitialize and continuing without ground-truth pose".
Do you solve it?

bessszilard, 2021-05-21:
Sorry, but I stopped working on this field, so the issue was no longer relevant.

JaeYoungLee79, 2022-01-07:
I have same error.
Some example working well and other is not.
Some computer working with no error but jetson board has happened this.