MIT-SPARK/Kimera-VIO-ROS #104: Errors With GTests When Building Kimera VIO

lpeng8910, 2020-09-08:
**Description:**
I'm getting a build error for kimera vio package. I'm Running ROS Melodic on Ubuntu 18.04. Everything's on the master branch, I've installed the prerequisites from the instructions in the README. Thanks so much in advance for the help. 

**Command:**
```
catkin build
```

**Console output:**
```

In file included from /home/lisa/catkin_ws/src/Kimera-VIO/src/visualizer/OpenCvVisualizer3D.cpp:29:0:
/home/lisa/catkin_ws/src/Kimera-VIO/include/kimera-vio/utils/FilesystemUtils.h: At global scope:
/home/lisa/catkin_ws/src/Kimera-VIO/include/kimera-vio/utils/FilesystemUtils.h:22:13: warning: ‘bool VIO::common::createDirectory(const string&, const string&, bool)’ defined but not used [-Wunused-function]
 static bool createDirectory(const std::string& dir_path,
             ^~~~~~~~~~~~~~~
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /home/lisa/catkin_ws/devel/lib/libDBoW2.so, may conflict with libopencv_core.so.3.4
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /home/lisa/catkin_ws/devel/lib/libDBoW2.so, may conflict with libopencv_core.so.3.4
CMake Error at /usr/share/cmake-3.10/Modules/GoogleTestAddTests.cmake:39 (message):
  Error running test executable.

    Path: '/home/lisa/catkin_ws/build/kimera_vio/testKimeraVIO'
    Result: Segmentation fault
    Output:
      VioPipelineFixture.
        OnlineSequentialStart
        OnlineSequentialShutdown
        OnlineSequentialSpinOnce
        OnlineSequentialSpin
        OnlineParallelStartManualShutdown
        OnlineParallelSpinManualShutdown
        OnlineParallelSpinShutdownWhenFinished
        OfflineSequentialShutdown
        OfflineSequentialSpinOnce
        OfflineSequentialSpin
        OfflineParallelStartManualShutdown
        OfflineParallelSpinManualShutdown
        OfflineParallelSpinShutdownWhenFinished
        OfflineSequentialSpinBackendFailureGracefulShutdown
        OnlineParallelSpinBackendFailureGracefulShutdown
        OnlineParallelSpinRegularBackendFailureGracefulShutdown
      VioParamsFixture.
        defaultConstructorWithoutParsing
        defaultConstructorWithParsing
      TestEurocPlayground.
        DISABLED_basicEurocPlayground
      testCameraParams.
        parseYAML
        equals
        Cal3_S2ToCvmat
        DISABLED_parseKITTICalib
      testCodesignIdeas.
        pixelDisplacementLinearVelocity
      TestDataProviderModule.
        basicSequentialCase
        noImuTest
        manyImuTest
        imageBeforeImuTest
        imageBeforeImuDelayedSpinTest
      testFrame.
        constructor
        ExtractCornersChessboard
        ExtractCornersWhiteBoard
        getNrValidKeypoints
        CalibratePixel
        DISABLED_CalibratePixel
        findLmkIdFromPixel
      testGeneralParallelPlaneRegularBasicFactor.
        ErrorIsZero
        ErrorOtherThanZero
[... 198 lines cut ...]
cd /home/lisa/catkin_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.................................................................................................
Failed     << kimera_vio:make                              [ Exited with code 2 ]                
Failed    <<< kimera_vio                                   [ 3 minutes and 53.2 seconds ]        
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]              
[build] Summary: 15 of 17 packages succeeded.                                                    
[build]   Ignored:   None.                                                                       
[build]   Warnings:  3 packages succeeded with warnings.                                         
[build]   Abandoned: 1 packages were abandoned.                                                  
[build]   Failed:    1 packages failed.                                                          
[build] Runtime: 8 minutes and 39.5 seconds total.   

```
It error is a "Segmentation Fault"? There's a lot of warnings before this error too, although that might be okay. 

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* SparkVio branch, tag or commit used ?
* GTSAM version used: master branch
* OpenGV version used: master branch
* OpenCV version used: type `opencv_version` 3.4.2
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
* Did you change the source code? (yes / no): nope

lpeng8910, 2020-09-10:
I resolved this issue, by wiping my whole catkin workspace and starting over. :/

closed 2020-09-10 by lpeng8910

X-NEARS, 2022-04-18:
hello, I have encountered the same problem. Can you tell me how to do it specifically and what kind of code can be input to solve it