MIT-SPARK/Kimera-VIO-ROS #105: Run live on RealSense D435i stream - launch/camera/rviz settings?

Ritchizh, 2020-09-09:
Thank you for this amazing library!
I am trying to launch Kimera on the stream from RealSense D435i.
Here are my commands, each part in a separate terminal:

1. Launch RealSense camera:
`source ~/ws/RealSense_ws/devel/setup.bash`
`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true`
2. Visualize image stream, disable infrared dots:
`rosrun dynamic_reconfigure dynparam set /camera/stereo_module emitter_enabled 0`
`rosrun image_view image_view image:=/camera/infra1/image_rect_raw`
3. Launch Kimera-VIO ROS wrapper:
`source ~/ws/kimera/devel/setup.bash`
`roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch`
4. Visualize with RVIZ:
`source ~/ws/kimera/devel/setup.bash`
`rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz`

I can see the image stream from _/camera/infra1/image_rect_raw_ topic. The rqt_graph also looks promising:
[image]
it shows that camera is publishing infra1 (and probably infra2) to Kimera algorithm.
However, nothing is visualized in Rviz. My concerns are:
1) Launching the RealSense camera should work with just `roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation` - without enabling infra1,2 topics explicitly, but it doesn't publish them otherwise. **Does this mean I have some incorrect/old version of  _rs_camera.launch_ file? In that case how to check whether my _camera calibration parameters_ are OK?** 
2) **Do I need a specific _kimera_vio_euroc.rviz_ file** to visualize features and mesh generated by Kimera for the live stream?

* The pipeline worked fine for the sample D435i _RealSenseCommonArea.bag_ file provided in one of the issues:
1) `roscore`
2) `roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch`
3) `rosbag play --rate 0.4 ~/data/euroc/RealSenseCommonArea.bag`
4) `rviz -d $(rospack find kimera_vio_ros)/rviz/kimera_vio_euroc.rviz`
For this .bag file I can see texture and features in Rviz.

I would appreciate any advice, thank you. 

===============================================

/rosdistro: melodic
/rosversion: 1.14.6

GTSAM version used: commit 342f30d148fae84c92ff71705c9e50e0a3683bda (HEAD -> master)
Kimera-VIO: commit b9d9ed38fc5d9e64f8e594ac18c10dfb43ea6099 (HEAD -> master)
Kimera-VIO_ROS: commit ea63cad3edb4ab8a35509f8417b95e34b29af1ab (HEAD -> master)
rviz version 1.13.13
compiled against Qt version 5.9.5
compiled against OGRE version 1.9.0 (Ghadamon)
Stereo is NOT SUPPORTED
OpenGl version: 3 (GLSL 1.3).

realsense-ros: commit a0a75383530d688d588d351a6d0ce4f4b60ac6d8 (HEAD, tag: 2.2.16)
RealSense ROS v2.2.16
Running with LibRealSense v2.38.1

Ritchizh, 2020-09-10:
I kind of managed to run Kimera on live stream with the following parameters set:
`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_color:=false enable_depth:=false enable_imu:=true enable_gyro:=true enable_accel:=true`

[image]

- No mesh is visualized.
- The image seems sparse.
- Kimera outputs: 
**5253 Mesher.cpp:1380] Missing landmark information to build 3D Mesh.**

gaotao19890725, 2020-09-25:
Thank  you Ritchizh, 
        The commands you give work, but no mesh is visualized.

ToniRV, 2020-09-25:
For the mesh visualization in Rviz, set to True the flag in `params/Euroc/flags/Mesher.flags`:
`--return_mesh_2d=true`

gaotao19890725, 2020-09-25:
Hi ToniRV：
    Your suggestion confuses me, can you specify which file or which document or which command that I should modify in order to get the mesh visualization. Thank you a million.

gaotao19890725, 2020-09-25:
Hi ToniRV:
     I find it, thank you. We should first find the Mesher.flags document according to ~/catkin_ws/src/Kimera-VIO/params/Euroc/flags/Mesher.flags, then open Mesher.flags and modify the last sentence --return_mesh_2d=false to --return_mesh_2d=true.
    Unfortunately,  after modification no mesh is visualized.
     Thanks again.

ToniRV, 2020-09-25:
@gaotao19890725 that should be it, yes, assuming you are running Kimera in the `Euroc` dataset though.
Otherwise, change the `Mesher.flags` file in the params folder that is pointed to by the `kimera_vio_ros.launch` file that you are using, check:

```
  <arg name="params_folder" value="$(find kimera_vio)/params/$(arg dataset_name)"/>
```
In `~/catkin_ws/src/Kimera-VIO-ROS/launch/kimera_vio_ros.launch`

Ritchizh, 2020-09-29:
@gaotao19890725 Hi!
No mesh is visualized for me too, you can see on the screenshot above there are only features and texture. I think it's a common problem:
https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/78

Setting return_mesh_2d=true doesn't help in my case.
However, it seems, the texture is actually placed on a mesh - so it is there, but is not visualized in form of the green triangles as in the demo.

closed 2020-09-29 by Ritchizh

HM102, 2020-11-12:
Inside rviz under 3d MESH, you can set Polygon mode to WIREFRAME, if you want to see the triangles