MIT-SPARK/Kimera-VIO-ROS #111: source code

guoxiaoyang12, 2020-10-21:
source code: In the Bacckend the regularity fatctors between point and plane may not be added in the optimizaton in open source code now.is this normal?

ToniRV, 2020-10-26:
Hi @guoxiaoyang12 ,
The regularity factors are only added if you use the RegularBackend (see https://github.mit.edu/SPARK/Kimera-VIO/blob/master/params/Euroc/PipelineParams.yaml)
And if you set the backend modality (in https://github.mit.edu/SPARK/Kimera-VIO/blob/master/params/Euroc/BackendParams.yaml) to number 4.

Technically speaking, the RegularBackend can be 4 types of backend:
- Structureless backend (aka the vanilla VioBackend)
- Structureless and projection: it will use two types of factors, structureless, and classic re-projection factors
- Projection only: classic re-projection factor
- Projection and Regularity: uses re-projection factors together with regularity factors {
- Structureless, projection, and regularity factors: it will first use structureless factors to be robust to outliers, then it will detect regularities, and for those landmarks that are expected to be on a regularity, it will transform the structureless factor to multiple projection factors together with a regularity factor with other landmarks.

See https://github.mit.edu/SPARK/Kimera-VIO/blob/f0f1dc27c889dd16e5e0975fd290b02806959ad3/src/backend/RegularVioBackEnd.cpp#L234
for the code that switches between backend modalities inside the RegularVioBackend.

guoxiaoyang12, 2020-10-28:
Thank you for your reply,ToniRV. 
   As shown in the paper,the Mesher thread first calculate the 2D/3D Mesh depends on the current landmarks,secondly,using histograms to get a statistics about the norm of mesh which can calculate the Wall and  floor;
   In the Mesher.cpp ,the spinOnce function call the updateMesh3D function to get the 3D Mesh and update the mesh_3d_...etc variables, but the variable planes_ used  as Regularity factors may not be update .
   In other word ,the function clusterPlanesFromMesh which can extract planes may not be called by spinOnce,so plane_ variable may not be update; 
[image]
[image]
  I hope to hear from you as soon as possible