MIT-SPARK/Kimera-VIO-ROS #114: libkimera_vio.so: undefined reference to `cv::viz

ArghyaChatterjee, 2020-11-02:
**Description:**
Hi, I created a workspace named kimera_ws and followed the steps mentioned in the readme section. Still I get the error.

**Command:**
```
arghya@arghya-Erazer-X7849-MD60379:~/kimera_ws$ catkin build
```

**Console output:**
```
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/arghya/costar_ws/devel:/opt/ros/melodic
Workspace:                   /home/arghya/kimera_ws
--------------------------------------------------------------------------
Build Space:        [exists] /home/arghya/kimera_ws/build
Devel Space:        [exists] /home/arghya/kimera_ws/devel
Install Space:      [unused] /home/arghya/kimera_ws/install
Log Space:          [exists] /home/arghya/kimera_ws/logs
Source Space:       [exists] /home/arghya/kimera_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found '16' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Starting  >>> disparity_image_proc                                             
Starting  >>> gtsam                                                            
Starting  >>> image_undistort                                                  
Finished  <<< disparity_image_proc                         [ 0.2 seconds ]     
Finished  <<< catkin_simple                                [ 0.1 seconds ]     
Starting  >>> cmake_external_project_catkin                                    
Starting  >>> dbow2_catkin                                                     
Starting  >>> gflags_catkin                                                    
Starting  >>> kimera_rviz_markers                                              
Starting  >>> mesh_rviz_plugins                                                
Starting  >>> opencv3_catkin                                                   
Finished  <<< cmake_external_project_catkin                [ 0.1 seconds ]     
Finished  <<< opencv3_catkin                               [ 0.4 seconds ]     
Finished  <<< gflags_catkin                                [ 0.2 seconds ]     
Finished  <<< dbow2_catkin                                 [ 0.2 seconds ]     
Finished  <<< kimera_rviz_markers                          [ 0.6 seconds ]     
Starting  >>> opengv_catkin                                                    
Finished  <<< opengv_catkin                                [ 0.2 seconds ]     
Finished  <<< image_undistort                              [ 0.9 seconds ]     
Starting  >>> pose_graph_tools                                                 
Finished  <<< mesh_rviz_plugins                            [ 0.6 seconds ]     
Finished  <<< pose_graph_tools                             [ 0.3 seconds ]     
Starting  >>> glog_catkin                                                      
Finished  <<< gtsam                                        [ 2.7 seconds ]     
Starting  >>> kimera_rpgo                                                      
Finished  <<< glog_catkin                                  [ 0.1 seconds ]     
Finished  <<< kimera_rpgo                                  [ 0.2 seconds ]     
Starting  >>> kimera_vio                                                       
_______________________________________________________________________________
Errors     << kimera_vio:make /home/arghya/kimera_ws/logs/kimera_vio/build.make.001.log
libkimera_vio.so: undefined reference to `cv::viz::Widget::~Widget()'
[... 116 lines cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [testKimeraVIO] Error 1
make[1]: *** [CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [all] Error 2
cd /home/arghya/kimera_ws/build/kimera_vio; catkin build --get-env kimera_vio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << kimera_vio:make                              [ Exited with code 2 ]
Failed    <<< kimera_vio                                   [ 1.2 seconds ]     
Abandoned <<< kimera_vio_ros                               [ Unrelated job failed ]
[build] Summary: 14 of 16 packages succeeded.                                  
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4.5 seconds total.                                            

```

Please give also the following information:
* SparkVio branch, tag or commit used: SparkVio branch
* GTSAM version used: 4.0 (should have been automatically installed inside the workspace according to readme file)
* OpenGV version used: exact version is not known but followed the readme file during installation.
* OpenCV version used: 3.4.2
* Operating system and version: Ubuntu 18.04 
* Did you change the source code? yes : I changed the Cmakelists.txt inside Kimera-VIO replacing the line "find_package(OpenCV REQUIRED) with find_package(OpenCV 3.4.2 REQUIRED). Changing the line seems to have no effect on the output. Before and after result was the same.

ToniRV, 2020-11-02:
Can you add this line to the package.xml in https://github.com/MIT-SPARK/Kimera-VIO/blob/master/package.xml:
```
  <depend>opencv3_catkin</depend>
```
That might fix it, as you tell it what opencv to use.

ArghyaChatterjee, 2020-11-03:
Hi @ToniRV, Thanks for the reply. Actually I didn't need to modify the `package.xml` file.  The problem was solved after I deleted the previous workspace (where I initiated catkin with 'catkin init') and created a fresh workspace with a modification in the opencv version inside the CmakeLists.txt file in the Kimera-VIO directory ( `find_package(OpenCV 3.4.2 REQUIRED)`, in my case opencv version is 3.4.2) of Kimera-VIO-ROS package.

The problem I think I was having was 2 different versions of opencv (3.2 and 3.4.2) were installed in my system. I have initially installed opencv 3.4.2 in my system quite a back ago but during the installation of ROS, opencv version 3.2 was automatically installed as part of the full installation process. 

So, during initial failed attempt of kimera-VIO package building, some warnings were displayed regarding some libraries or directories are preceeded over other. During successful kimera-VIO package building, it was showing some sort of warning about conflicting version of libopencv_core.so.3.2 and libopencv_core.so.3.4 as well as libopencv_viz.so.3.2 and libopencv_viz.so.3.4 libraries are installed but in the end it was built successfully.   

Hope this answer helps someone who faced/will face similar issue. I am closing the issue.

ArghyaChatterjee, 2020-11-03:
The problem was solved. I am closing the issue.

closed 2020-11-03 by ArghyaChatterjee