MIT-SPARK/Kimera-VIO-ROS #115: kimera-vio-ros gtsam build fails on Ubuntu 20.04

berndpfrommer, 2020-11-05:
I've added support for Ubuntu 20.04 to Kimera and wanted to submit a PR, but hit a snag regarding GTSAM.

Kimera-vio-ros currently does not build for two reasons:

1) The image_undistort node does not compile with the c++-11 flag. I have filed a PR against the repo: https://github.com/ethz-asl/image_undistort/pull/64 so hopefully this will be fixed at some point.

2) The "master" branch of gtsam does not compile because it has a broken CMakeLists.txt file in the "wrap" subdirectory (some matlab wrappers): 

```
CMake Error at /home/pfrommer/Documents/kimera/src/gtsam/wrap/CMakeLists.txt:32 (target_link_libraries):
  The "debug" argument must be followed by a library.
```
This has been fixed with the latest gtsam release 4.0.3. Alas, gtsam v4.0.3 does not build on ubuntu 14.04 (trusty), because for that very same fix it requires cmake > 3.0.0, whereas trusty has only 2.8.12. In a pinch people can install cmake 3 on Ubuntu 14.04, so there is a workaround: https://askubuntu.com/questions/610291/how-to-install-cmake-3-2-on-ubuntu 

Question: how important is it for you to keep support for Ubuntu 14.04?

ToniRV, 2020-11-05:
Hi @berndpfrommer,

We only test for >=16.04, so 14.04 is not officially supported.

Thanks for the detailed issue!
We could add a build to our CI server with 20.04 and git checkout 4.0.3 in gtsam.
We could then just update the install readme.

Note though that I have no CI with the ROS wrapper: 
 - There is an open PR for it here: https://github.com/MIT-SPARK/Kimera-VIO-ROS/pull/32

Feel free to open a PR.

geothan, 2021-09-14:
Hi @berndpfrommer ,

i tired to install kimera-vio ros in Ubuntu 20. Your suggested modifications have been addressed now, but build fails. Have you managed to build this project in ubuntu 20?